CadQuery/cadquery #796: Error With Python 3.8 but Not With Python 3.9

jmwright, 2021-06-23:
## To Reproduce

A CadQuery environment created with the following line works:

```
conda install -c cadquery -c conda-forge cadquery=master python=3.9
```

But this does not create a working environment.

```
conda install -c cadquery -c conda-forge cadquery=master python=3.8
```

## Backtrace

This is what I get if I try to import cadquery in the Python 3.8 environment.

```
2021, 07:01:05) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cadquery
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jwright/anaconda3/envs/cadquery2/lib/python3.8/site-packages/cadquery/__init__.py", line 2, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
  File "/home/jwright/anaconda3/envs/cadquery2/lib/python3.8/site-packages/cadquery/occ_impl/geom.py", line 5, in <module>
    from OCP.gp import (
ImportError: /home/jwright/Downloads/cq-cli-Linux2/cq-cli/OCP.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN18BRepOffset_Inter2d14ConnexIntByIntERK11TopoDS_FaceR17BRepOffset_OffsetR19NCollection_DataMapI12TopoDS_ShapeS6_23TopTools_ShapeMapHasherERKS8_RKN11opencascade6handleI14BRepAlgo_AsDesEEddRK18BRepOffset_AnalyseR22NCollection_IndexedMapIS6_S7_ER26NCollection_IndexedDataMapIS6_16NCollection_ListIS6_ES7_E
```

## Environment

OS: Ubuntu 18.04
Conda Version: 4.8.5

adam-urbanczyk, 2021-06-24:
Is it fixed by #790?

jmwright, 2021-06-24:
Yes, sorry for the duplicate.

closed 2021-06-24 by jmwright