CadQuery/cadquery #798: BoundingBox Very Inaccurate

raphaelsetin, 2021-06-25:
I've came across this problem today, when I saw that the bounding box measurements of the models in which I imported (in `.STP` format) to CADQuery were off by a few millimeters -- depending on the model this could be higher. After some research and reading throughout the repo, I saw issues #7, #74, #79, and #167, in which **weren't successful at fixing this problem**.

**I tried to tweak the tolerance using all the examples provided in these discussions (to no avail), such as:**
- Changing the global tolerance:
`cadquery.occ_impl.geom.TOL = 0.00001`
- Passing the tolerance to the internal TopoDS method:
`cadquery.BoundBox._fromTopoDS(cube.val().wrapped, tol=0.00001)`
- Changing the tolerance on the default bounding box method:
`cube.val().BoundingBox(tolerance=0.00001)`

A good example file to use is this [cube_grid.step](https://raw.githubusercontent.com/cqparts/cqparts/master/tests/test-files/cube_grid.step). Also, for comparison, here are some values from CADQuery, PythonOCC, and FreeCAD:
1. CADQuery:
    - BoundingBox: 
        - X: 149.00000028 mm
        - Y: 149.00000028000002 mm
        - Z: 14.000000280000002 mm
    - Volume: 100000.00000000065 mm<sup>3</sup>
2. PythonOCC:
    - BoundingBox: 
        - X: 145.02000020000003 mm
        - Y: 145.02000020000003 mm
        - Z: 10.020000199999998 mm
    - Volume: 100000.00000000065 mm<sup>3</sup>
3. FreeCAD:
    - BoundingBox: 
        - X: 145.0 mm
        - Y: 145.00000000000003 mm
        - Z: 10.0 mm
    - Volume: 100000.00000000065 mm<sup>3</sup>

As you can see, the CADQuery implementation of the bounding box function is clearly flawed in precision. What is strange to me is how well the volume implementation is (super precise as the other libraries), but our bounding box implementation is not.

adam-urbanczyk, 2021-06-26:
I cannot reproduce on master:
```python
w = cq.importers.importStep('cube_grid.step')
c = w.val()
bb = c.BoundingBox()
bb.xlen, bb.ylen, bb.zlen
```
gives me:
```
 (145.0, 145.00000000000003, 10.0)
```

marcus7070, 2021-06-26:
> gives me:
> 
> ```
>  (145.0, 145.00000000000003, 10.0)
> ```

+1, installed through nix package manager instead of Conda. Both OCP/OCCT v7.5.1 and v7.4.

raphaelsetin, 2021-06-28:
Hey @adam-urbanczyk and @marcus7070, thank you for the quick feedback on this! This was definitely strange because as far as I can recall, CADQuery was always accurate for me in the past.

**So after some tests, I was able to find the bug, which seems to be non-user related.** Basically, whenever we get the CADQuery workplane and just get the bounding box, everything works as expected. But if we export the workplane to another format, such as `STL`, then getting the bounding box afterwards is less accurate. Volume seems to be unaffected as we can see from this test.

## Test Code

```
import cadquery as cq

wp = cq.importers.importStep('/tmp/cube_grid.step.stp')
wpVal = wp.val()
bBox = wpVal.BoundingBox()
volume = wpVal.Volume()
print('Before:', bBox.xlen, bBox.ylen, bBox.zlen)
print('Before:', volume)

cq.exporters.export(wp, '/tmp/test.stl')
bBox = wpVal.BoundingBox()
volume = wpVal.Volume()
print('After:', bBox.xlen, bBox.ylen, bBox.zlen)
print('After:', volume)
```

## Output

```
Before: 145.0 145.00000000000003 10.0
Before: 100000.00000000065

After: 149.00000028 149.00000028000002 14.000000280000002
After: 100000.00000000065
```

It seems that exporting the workplane messes up with the tolerance settings, though I haven't tested this for sure; but it's definitely something along these lines.

adam-urbanczyk, 2021-06-28:
Confirmed