CadQuery/cadquery #803: Revolve() seems to ignore the rotate() invoked prior to it

safdark, 2021-06-26:
I'm trying to create a curved bowl by constructing the cross-section in the XY plane using edges (lineTo, threePointArc, etc), rotating that closed set of edges into the XZ plane, and then revolving it 360-degrees to create the 3d solid (with normal in the +ve Z direction). Rotating seems to work fine, but the moment I try to revolve it, the edges collapse back into the XY plane, and instead of a 3d bowl, I get a 2d disk.

The code I have is:

```
flat_radius=33
taper_radius=4
height=3
thickness=1

plane = cq.Workplane("XY")
bowl = plane.line(xDist=flat_radius, yDist=0)\
        .tangentArcPoint(endpoint=(taper_radius, height), relative=True)\
        .hLine(distance=-thickness)\
        .tangentArcPoint(endpoint=(-(taper_radius-thickness), -(height-thickness)), relative=True)\
        .lineTo(x=0.0, y=thickness)\
        .close()\
        .rotate(axisStartPoint=(0, 0, 0), axisEndPoint=(1, 0, 0), angleDegrees=90)\
        .revolve(angleDegrees=360.0, axisStart=(0, 0, 0), axisEnd=(0, 0, 1), clean=False) <------- This isn't working as expected
```

Why doesn't revolve() respect the rotation done before it?

safdark, 2021-06-26:
I understand that there are multiple ways to get to most 3d solids, so my concern is not about _how_ to get to the bowl shape I want, but rather _why_ the given invocation sequence doesn't seem to be doing I'd expect it to, based on the cadquery documentation.

marcus7070, 2021-06-27:
If you are not aware of what the modelling context is, check out this gist first: https://gist.github.com/marcus7070/271e2ec67d1ff548341a4820868be0c2

The methods `line`, `tangentArcPoint`, `hLine`, `lineTo` and `close` push Edges or Wires to the modelling context. The `Workplane.rotate` method acts on the objects in `Workplane.objects`, and does not act on the edges/wires already pushed to the modelling context. 

`Workplane.revolve` sources the Wire objects to revolve from the modelling context.

Does that make sense now?

My advice would be to not use the rotate/translate methods in CQ (at least until you know exactly what they do). Position your workplane where you need to and create your wires in the correct place from the start, doing it the other way around is much more difficult.

marcus7070, 2021-06-27:
Also, the axis arguments to `Workplane.revolve` are in coordinates local to the current workplane, not global coordinates. It would be very unusual to revolve around the Z axis, you want the axis of revolution in the same plane as your 2d lines.

safdark, 2021-06-27:
Thanks for the explanation, and the link, @marcus7070! Such documentation is really helpful and I wish there was more of it. The link you sent me seems to be part of a larger documentation effort (based on the words: "To see a bit more about how the internals of CadQuery work ..."). Could you point me to the larger table-of-contents for that documentation/effort?

And, could you also suggest other such resources that I could peruse for an advanced understanding of cadquery? The documentation that is available on the readthedocs site is limited in this regard, and despite having gone through most of it, I feel like I've just scratched the surface, nowhere near what I'd need to implement advanced models.

marcus7070, 2021-06-27:
> Could you point me to the larger table-of-contents for that documentation/effort?

Sorry @safdark, that was just something I quickly wrote on it's own. I'm trying to tidy it up a bit and fit it into the main docs though.

> could you also suggest other such resources that I could peruse for an advanced understanding of cadquery? The documentation that is available on the readthedocs site is limited in this regard, and [...]

The absolute best place to go after the online docs is the source code itself. Once you understand a few of the internals like the modelling context, I find it all very readable (perhaps with the exception of the `Workplane.workplane` method, that needs some work).

adam-urbanczyk, 2021-06-27:
So to sum up - rotate works on `objects` and and revolve works on `pendingWires/Edges`. 

Regarding resources - it might be interesting to read the tests and take a look at the contrib repo.

closed 2021-06-27 by adam-urbanczyk

safdark, 2021-07-06:
> Sorry @safdark, that was just something I quickly wrote on it's own. I'm trying to tidy it up a bit and fit it into the main docs though.

That would be great, @marcus7070. Yes, the code is next on my list.
<br>

>So to sum up - rotate works on objects and revolve works on pendingWires/Edges.
>
> Regarding resources - it might be interesting to read the tests and take a look at the contrib repo.

Thx @adam-urbanczyk ... would be great to have precisely _such_ tidbits in a "Zen/Tidbits" section in the docs ;).
Cadquery-contrib looks interesting; hadn't checked it out before!