CadQuery/cadquery #809: Does OCP work in OCCT 7.5.2 ?

a09418849, 2021-06-30:
I used the general way to install cadquery  :
```
conda install -c conda-forge -c cadquery cadquery=master
```

and it showed the warning  :

```
type object 'OCP.BRepGProp.BRepGProp' has no attribute 'LinearProperties_s'
```

I noticed that the OCCT has been updated to version=7.5.2, but in cadquery is 7.5.1 
So I try to install OCCT 7.5.1 first  :
```
conda install -c conda-forge -c trelau pyocct=7.5.1.1
```
And then it works , but I don't know why.

Sorry for my bad English.

Envirements :
```
# Name                    Version                   Build  Channel
anyio                     3.2.1                    pypi_0    pypi
argon2-cffi               20.1.0           py38h2bbff1b_1
async_generator           1.10               pyhd3eb1b0_0
attrs                     21.2.0             pyhd3eb1b0_0
babel                     2.9.1                    pypi_0    pypi
backcall                  0.2.0              pyhd3eb1b0_0
bleach                    3.3.0              pyhd3eb1b0_0
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2021.5.25            haa95532_1
cadquery                  master                    py3.8    cadquery
cadquery-massembly        0.9.0                    pypi_0    pypi
certifi                   2021.5.30        py38haa95532_0
cffi                      1.14.5           py38hcd4344a_0
chardet                   4.0.0                    pypi_0    pypi
colorama                  0.4.4              pyhd3eb1b0_0
console_shortcut          0.1.1                         4
curl                      7.77.0               h789b8ee_0    conda-forge
cycler                    0.10.0                   pypi_0    pypi
decorator                 5.0.9              pyhd3eb1b0_0
defusedxml                0.7.1              pyhd3eb1b0_0
double-conversion         3.1.5                h0e60522_2    conda-forge
eigen                     3.3.9                h2d74725_1    conda-forge
entrypoints               0.3                      py38_0
expat                     2.4.1                h39d44d4_0    conda-forge
ezdxf                     0.16.4           py38hbd9d945_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.1            h1989441_1005    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeimage                 3.18.0               hfcb8cd7_7    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
gl2ps                     1.4.2                h0597ee9_0    conda-forge
glew                      2.1.0                h39d44d4_2    conda-forge
hdf4                      4.2.15               h0e5069d_3    conda-forge
hdf5                      1.10.6          nompi_h5268f04_1114    conda-forge
icu                       68.1                 h0e60522_0    conda-forge
idna                      2.10                     pypi_0    pypi
ilmbase                   2.5.5                h12d4b20_0    conda-forge
importlib-metadata        3.10.0           py38haa95532_0
importlib_metadata        3.10.0               hd3eb1b0_0
intel-openmp              2021.2.0           h57928b3_616    conda-forge
ipydatawidgets            4.2.0                    pypi_0    pypi
ipykernel                 5.3.4            py38h5ca1d4c_0
[... 109 lines cut ...]
traitlets                 5.0.5              pyhd3eb1b0_0
traittypes                0.2.1                    pypi_0    pypi
typing_extensions         3.10.0.0           pyha770c72_0    conda-forge
typish                    1.9.2              pyhd8ed1ab_0    conda-forge
urllib3                   1.26.6                   pypi_0    pypi
utfcpp                    3.2.1                h57928b3_0    conda-forge
vc                        14.2                 h21ff451_1
voila                     0.2.10                   pypi_0    pypi
vs2015_runtime            14.27.29016          h5e58377_2
vtk                       9.0.1           no_osmesa_py38h8c3c96c_109    conda-forge
wcwidth                   0.2.5                      py_0
webcolors                 1.11.1                   pypi_0    pypi
webencodings              0.5.1                    py38_1
websocket-client          1.1.0                    pypi_0    pypi
wheel                     0.36.2             pyhd3eb1b0_0
widgetsnbextension        3.5.1                    pypi_0    pypi
win32_setctime            1.0.3                      py_0    conda-forge
wincertstore              0.2                      py38_0
winpty                    0.4.3                         4
xz                        5.2.5                h62dcd97_1    conda-forge
zeromq                    4.3.3                ha925a31_3
zipp                      3.4.1              pyhd3eb1b0_0
zlib                      1.2.11            h62dcd97_1010    conda-forge
zstd                      1.5.0                h6255e5f_0    conda-forge
```

fedorkotov, 2021-06-30:
Maximum supported OCP version is 7.5.1. See https://github.com/CadQuery/cadquery/pull/792 and #790

jmwright, 2021-06-30:
For now, the following conda install line should produce a working environment.
```
conda install -c cadquery -c conda-forge cadquery=master ocp=7.5.1
```

fedorkotov, 2021-06-30:
@jmwright Maybe conda installation command should be updated in [README](https://github.com/CadQuery/cadquery#getting-started)?

jmwright, 2021-06-30:
It's my understanding that it should be fixed fairly soon so that the original line with just `cadquery=master` will work again. [Reference](https://github.com/CadQuery/cadquery/issues/790#issuecomment-869923900)

adam-urbanczyk, 2021-07-02:
Duplicate of #790

closed 2021-07-02 by adam-urbanczyk