CadQuery/cadquery #810: GEB-like intersection?

danbst, 2021-07-02:
I've made a GEB-like solid, but I have a feeling that it was done wrong. I'll attach source code in the end.

[image]

The problem was
a) create square letters (equal size both dimensions)
b) properly orient letters

I've resorted into number-mungling, which wasn't as fun as expected.

What were better options?

```python
import cadquery as cq
size = 3
letterArgs = dict(fontsize=size, distance=-size)
letter2 = dict(cut=False, combine=True, kind="bold", **letterArgs)

scaleT = lambda x, y, z: cq.Matrix([
        [x, 0, 0, 0],
        [0, y, 0, 0],
        [0, 0, z, 0],
        [0, 0, 0, 1]
    ])

G = (
    cq.Compound.makeText("G", kind="bold",size=1, height=0.8)
    .transformGeometry(scaleT(1.3,1,1))
    .rotate((0,0,0.01), (0.01,0,0), 180).translate((0.4,0,0))
    )

E = (
    cq.Compound.makeText("E", kind="bold",size=1, height=0.8)
    .transformGeometry(scaleT(1.5,1,1.1))
    .translate((-0.1,0,-0.48))
    )

B = (
    cq.Compound.makeText("B", kind="bold",size=1, height=0.8)
    .transformGeometry(scaleT(1.4,1.2,0.9))
    .rotate((-1,0,0), (1,0,0), 90).translate((-0.1,0.4,-0.05))
    )

GEB = (
        cq.Workplane("XY", origin=(0,0,size/2))
        .union(E)
        .intersect(B)
        .intersect(G)
        )

show_object(GEB)
```

fedorkotov, 2021-07-02:
> a) create square letters (equal size both dimensions)

That would be easier if you used a font with same width and height
Here is a stackexchange question with keywords to google for such a font
https://graphicdesign.stackexchange.com/questions/45260/name-for-type-that-has-the-same-width-and-height

danbst, 2021-07-02:
thanks! In mean time, I found a way to compute scale based on BoundingBox, and finally understood rotation, so both questions are resolved.

```python
import cadquery as cq

scaleT = lambda x, y, z: cq.Matrix([
        [x, 0, 0, 0],
        [0, y, 0, 0],
        [0, 0, z, 0],
        [0, 0, 0, 1]
    ])

def squarifyTo(obj, size=1):
    """ Autoscale to square size in XY """
    bbox = obj.BoundingBox()
    return obj.transformGeometry(
        scaleT(size / bbox.xlen, size / bbox.ylen, size / bbox.zlen))

def letter(lt):
    """ gen square letter and put it into center """
    result = squarifyTo(cq.Compound.makeText(
        lt, kind="bold", size=1, height=1))
    bbox = result.BoundingBox()
    return result.translate((
        -bbox.xmax + bbox.xlen/2,
        -bbox.ymax + bbox.ylen/2,
        -bbox.zmax + bbox.zlen/2))

E = (
     letter("E")
     # over X
     .rotate((-1,0,0), (1,0,0), 90)
     # over Z
     .rotate((0,0,-1), (0,0,1), 90)
     )

G = (
     letter("G")
     # over X
     .rotate((-1,0,0), (1,0,0), 90)
     )

B = (
     letter("B")
     )

GEB = (
        cq.Workplane("XY")
        .union(G)
        .intersect(E)
        .intersect(B)
        )

show_object(GEB)
```
[image]

closed 2021-07-02 by danbst