CadQuery/cadquery #811: nlopt solver with "PointInPlane" constraint can use too many iterations

marcus7070, 2021-07-03:
In the following code, while `base_pnt = (0, 0, 0)` the nlopt solver will use 2000 iterations. Change `base_pnt = (1, 0, 0)` and it will use just 4. The same behaviour occurs with just one constraint, but with three "PointInPlane" the assembly should be fully constrained and less iterations needed.

```python
assy = cq.Assembly(cq.Workplane().sphere(1), name="part0")
assy.add(cq.Workplane().box(1, 1, 1), name="part1")
base_pnt = cq.Vector(0, 0, 0)
for d in ((0, 0, 1), (0, 1, 0), (1, 0, 0)):
    assy.constrain(
        "part0",
        cq.Vertex.makeVertex(0, 0, 0),
        "part1",
        cq.Face.makePlane(width=1, length=1, basePnt=base_pnt, dir=cq.Vector(d)),
        "PointInPlane"
    )

assy.solve()
print(assy._solve_result)
```
```python
{'cost': 0.0, 'iters': 2000, 'status': 5}
```

On the old `scipy.optimise` (4c5816dd703afee9cd45283470675fbcee29f684) the same code produces:
```
         Current function value: 0.000000
         Iterations: 0
         Function evaluations: 113
         Gradient evaluations: 101
```

marcus7070, 2021-07-03:
I think the problem is that we have:
https://github.com/CadQuery/cadquery/blob/3cb72378b895398b4b5f6129f46b8866c2618067/cadquery/occ_impl/solver.py#L259

If there are no objections (@adam-urbanczyk?) I will set `xtol_abs` to `TOL` over in #797. Doing so fixes the above example, now it takes just 2 iterations.

adam-urbanczyk, 2021-07-03:
I'd propose to set xtol abs but to a much lower value than rel

marcus7070, 2021-07-03:
OK, I took a guess and set it to `TOL * 1e-3` in bc72db328dca8ce592fd06e259208f92001e8863, that value still solves this issue. You can change it again when you review #797.

closed 2021-07-06 by marcus7070