CadQuery/cadquery #817: cadquery 2:  GMSH and Finite Element Analysis

zxo102, 2021-07-07:
I am learning cadquery 2 and would like to use it to  create a 3D underground mine  model and display it on web with three.js,  and mesh it with GMSH for finite element analysis. Any suggestions? Thanks a lot.

bragostin, 2021-07-07:
This gmsh example may provide a good starting point: https://gitlab.onelab.info/gmsh/gmsh/-/blob/octreeSizeField/demos/api/terrain_stl.py

zxo102, 2021-07-08:
Thanks a lot. I will try this gmsh example.  Since freeCAD has included gmsh and FEM, I thought gmsh and FEM are available in cadquery 2 too.

fedorkotov, 2021-07-08:
CadQuery is a python library. You create CadQuery models by writing python scripts using classes and functions from the library. gmsh and FEM are not available in CadQuery itself but you can use any other python library in the same script as CadQuery including https://pypi.org/project/gmsh/ and https://github.com/nemocrys/pyelmer for example.

closed 2021-07-08 by adam-urbanczyk