CadQuery/cadquery #819: docs HTML API reference duplicates Parameter lists

lorenzncode, 2021-07-08:
See the HTML reference for Workplane.lineTo as an example:

https://cadquery.readthedocs.io/en/latest/classreference.html#cadquery.Workplane.lineTo

[image]

Notice there are two "Parameters:" lists where one list per method is expected.  Here is the portion of the lineTo docstring that documents parameters:

```
    :param float x: the x point, in workplane plane coordinates
    :param float y: the y point, in workplane plane coordinates
    :return: the Workplane object with the current point at the end of the new line
```

sphinx-build is currently configured to use the 3rd party extension "sphinx_autodoc_typehints".  
The docstring param definitions are expected to be of the form:

```
    :param x: the x point, in workplane plane coordinates
```

Modified docstring:

        """
        Make a line from the current point to the provided point

        :param x: the x point, in workplane plane coordinates
        :param y: the y point, in workplane plane coordinates
        :param forConstruction: whether or not the edge is used only for reference
        :return: the Workplane object with the current point at the end of the new line

        see :py:meth:`line` if you want to use relative dimensions to make a line instead.
        """

(I'm not sure if forConstruction should be documented or not - this is an example)
Re-running sphinx-build now yields:

[image]

The docs are still corrupted and contain a second Parameters: list (single item self).  I find this **second list is created due to sphinx_autodoc_typehints** option set in conf.py:
**```always_document_param_types = True```**

Commenting this out or setting to False yields:

[image]

Many of the API reference methods have this issue with multiple Parameters lists. Not all of them (such as intersect) because in those cases there is no other text description in the docstring after the first parameter list.  The extension creates the stub for self, and other args and they become part of the first list.

One option to clean up the API reference is to disable ```always_document_param_types``` and continue using the sphinx_autodoc_typehints extension along with making docstring updates.

The above was generated with sphinx 3.2.1 (pinned in environment.yml).  Sphinx has built-in autodoc_typehints and I ran a few experiments with Sphinx 4.0.3 (same modified docstring as above).  The following is no longer using the sphinx_autodoc_typehints extension.

First with: ```autodoc_typehints = 'none'```:

[image]

With ```autodoc_typehints = 'description'```:

[image]

Sphinx built-in typehint still adds "self" but handles it better than the other extension.  self is added to the Parameters list but at least we don't have two lists.  

Setting:
```
autodoc_typehints = 'description'
autodoc_typehints_description_target = 'documented'
```

[image]

marcus7070, 2021-07-08:
I think the following line in sphinx-autodoc-typehints should be dropping off the first parameter (`self`): https://github.com/agronholm/sphinx-autodoc-typehints/blob/master/sphinx_autodoc_typehints.py#L245
but it doesn't.

marcus7070, 2021-07-08:
> I think the following line in sphinx-autodoc-typehints should be dropping off the first parameter (`self`): https://github.com/agronholm/sphinx-autodoc-typehints/blob/master/sphinx_autodoc_typehints.py#L245
> but it doesn't for `Workplane.text` and the other examples.

I misread, that's the function that process the signature, not the docstring, which is [here](https://github.com/agronholm/sphinx-autodoc-typehints/blob/master/sphinx_autodoc_typehints.py#L407-L474).

The signature processing function does drop the first argument, the docstring function does not. :confused:

lorenzncode, 2021-07-09:
I find that it's dropped (the first parameter ```self```) only when there is no self-type.  

Testing with dummy input:

```
T = TypeVar("T", bound="MyClass1")
"""A type variable used to make the return type of a method the same as the
type of `self` or another argument.
"""

class MyClass1(object):

    """ My class 1 """

    def method1(self: T, arg1: bool = True) -> int:

        """
        This is method1

        :param arg1: this is arg1
        :return: 1 

        Example

        """

        return  1

class MyClass2(object):

    """ My class 2 """

    def method1(self, arg1: bool = True) -> int:

        """
        This is method1

        :param arg1: this is arg1
        :return: 1 

        Example

        """

        return  1

```

Extension sphinx_autodoc_typehints:

[image]

sphinx built-in:

[image]

lorenzncode, 2021-07-09:
The following will strip out the self param added by extension sphinx_autodoc_typehints.

```
from sphinx.ext.autodoc import between

def setup(app):
    app.connect('autodoc-process-docstring', between(marker=".*self:", what=['method'], exclude=True))
```

Note that while it does remove self, it does not solve the issue of duplicate Parameter lists.  For example, this docstring does not explicitly list arg2.

```
    def method1(self: T, arg1: bool = True, arg2: str = "X") -> int:

        """
        This is method1

        :param arg1: this is arg1
        :return: 1 

        Example

        """

        return  1
```

Result:

[image]

As mentioned, Sphinx built-in handles this case better (once we figure out how to exclude self):

[image]

Modifying the Sphinx built-in type hints to exclude self looks like it would be a different process.  "between" cannot be used as I find a different method is used than the extension (it does not change the docstring itself as sphinx_autodoc_typehints does).

marcus7070, 2021-07-09:
This might work with Sphinx's inbuilt autodoc:

```python
def drop_self(app, obj, bound_method):
    if hasattr(obj, "__annotations__"):
        obj.__annotations__.pop("self", None)

def setup(app):
    app.connect("autodoc-before-process-signature", drop_self)
```

I haven't checked what other type hinting this might wipe out.

marcus7070, 2021-07-09:
Before we go any further, can @adam-urbanczyk and @jmwright just let me know their opinions.

[image]

Regarding the HTML docs for `Workplane` methods:
1. Should `self` be in the signature?
2. Should `self: ~T` be in the parameters list?
3. Should `~T` be the return type?

I'd say 1: no, 2: no, 3: yes but we need a note at the top that `~T` is a TypeVar bound to Workplane.

adam-urbanczyk, 2021-07-12:
I'm for 2+3

marcus7070, 2021-07-19:
Ok @lorenzncode, I think the style to aim for is to have one autodoc list with:
```
Parameters:
  - self (~T)
 ...
Return type:
  - ~T
```

I don't have a strong opinion on how we get there, but I'd suggest we just correct any `:param float:` lines and move all `:param` and friends to the bottom of docstrings. I think that will fix all the issues you've identified, right?

lorenzncode, 2021-07-21:
We could substitute "Workplane" for "~T" by specifying the alias using autodoc_type_aliases.  Here is a screenshot with RTD version on left and local build on right. 

[image]

I like this solution (use built-in sphinx instead of additional extension) as it fixes the corrupt document structure (duplicate Parameter lists) without requiring any changes to the docstrings!

marcus7070, 2021-07-24:
Sounds good to me. I see @jmwright approves, @adam-urbanczyk do you mind if we do a PR to try replacing the sphinx-autodoc-typehints extension with Sphinx' inbuilt autodoc?

lorenzncode, 2021-10-30:
Another example comparing Workplane.polyline.  Current docs:

[image]

Updated:
[image]

As with each() the signature is more verbose than the existing docs.

lorenzncode, 2021-11-02:
I got side-tracked investigating the `autodoc_type_aliases` feature which is not necessary to address the problem of this issue.  I'd like to keep it simple and move forward to resolve this duplicate parameters in the HTML docs issue.

Proposed changes:

- Remove dependency on the current package used for doc typehints.  The Sphinx built-in extension can replace it to resolve this issue.  No other hooks or customization are required.  

- Update the sphinx version pin to current version

- Set the following sphinx autodoc config options: 

  `autodoc_typehints = "both"`

  This option is new in sphinx 4.1 and shows the typehints in the signature and as content of the function or method.

  `autodoc_typehints_description_target = "all"`

  This follows the current behavior but can be changed to `documented` if that is preferred.  See for example the current Workplane.line() doc.  The HTML shows `forConstruction` in the Parameter list even though it is not documented.
  

You can expect some presentation differences in the doc update compared to the old version.  For example, the built-in sphinx autodoc_typehints will show the qualified name `cq.occ_impl.shapes.Face` instead of `Face` and the typehints will now be hyperlinks.

See the following example for cutBlind. 

current docs:
[image]

with changes:

[image]

jmwright, 2021-11-04:
This seems like a reasonable path forward. +1

lorenzncode, 2021-11-05:
Great.  BTW, regarding `autodoc_typehints = "both"` - it could be reverted to "description" later if we want to match the current docs more closely.  I chose "both" because "signature" is the sphinx default and cadquery currently adds typehints to description.  

"both" may be useful for cases such as Workplane.polyline() where the optional type comment is shown in the description based on docstring: 

`:type listOfXYTuple: list of 2-tuples`

and then the type is shown in the signature as: 

`listOfXYTuple: Sequence[Union[Tuple[float, float], Tuple[float, float, float], cadquery.occ_impl.geom.Vector]]`

In the future perhaps this could also be simplified with type_aliases.

closed 2022-03-26 by lorenzncode