CadQuery/cadquery #820: Find "hidden" Volumes in a Model (undercuts)

raphaelsetin, 2021-07-08:
I am trying to find a way of obtaining the "obscured" volumes/solids of a part supposing we were to ray cast it from a specific direction, for instance Z axis. By ray casting on Z axis, I would want to get the computed volume of the cavities as solids, which is also known as `undercuts` in 3D modeling.

Explaining is more confusing than showing, so here is an example:
Download the STEP file here: [undercut example.zip](https://github.com/CadQuery/cadquery/files/6787688/undercut.example.zip)

### Original Model
[image]

### Volumes (in yellow) that I want to obtain (ray casting on Z axis)
[image]

I read the entire documentation, went through all examples, and even tried some functions to see if maybe I was misunderstanding their functionality. I still wasn't able to find a way in CADQuery to do this. I know you are here to share CADQuery answers, but if any of you know a way of doing this using another software, please let me know too, I am open to suggestions.

As of now, I am using `ThreeJS` to render the model on the webpage, and though their library is extensive and flexible, I didn't find a simple way of doing this.

Lastly, here is some documentation if the above is still not enough:
- [SolidWorks Undercut Analysis](https://help.solidworks.com/2020/English/SolidWorks/sldworks/c_mold_tools_undercut_analysis.htm?id=a9e7c8f922fa4c49ac0e0267687dc23a#Pg0)
- [SolidWorks Undercut Analysis PropertyManager](https://help.solidworks.com/2020/English/SolidWorks/sldworks/HIDD_DVE_UNDERCUT_ANALYSIS_2.htm?id=73de16386f414a12a443c44c18ce8dec#Pg0)

adam-urbanczyk, 2021-07-18:
Maybe you could find all faces that have a normal "downwards". You could use methods like `.Faces` and `normalAt`.

raphaelsetin, 2021-08-19:
Hey @adam-urbanczyk, thank you for the suggestion! I was able to solve this by developing an algorithm myself using the ThreeJS ray tracing API. However, I would like to do this within CADQuery itself, especially for performance and simplicity reasons.

I wasn't able to find any ray tracing functions in the CADQuery docs, so I would like to check with you if there are any ray tracing capabilities accessible within CADQuery.

jmwright, 2021-08-20:
> so I would like to check with you if there are any ray tracing capabilities accessible within CADQuery.

There are not raytracing capabilities within CadQuery. Since CadQuery is just Python, the closest you can get is to find a way to use a Python ray tracing package with CadQuery to do what you want. https://pypi.org/project/raytracing/

raphaelsetin, 2021-08-23:
@jmwright I didn't know about that package, but I was reading it now, and didn't find any relation with CADQuery, which makes me wonder how could I import a model in it and then do ray tracing.

Anyways, I know that CADQuery uses PythonOCC in the back end (correct me if I am wrong) as you guys moved away from the FreeCAD implementation some months ago, so would you know if PythonOCC has any ray tracing capabilities that maybe I could explore?

jmwright, 2021-08-23:
> but I was reading it now, and didn't find any relation with CADQuery

It's not associated with CadQuery but it's a Python package so you should be able to use it and CadQuery together in your own Python scripts.

>  I know that CADQuery uses PythonOCC in the back end (correct me if I am wrong)

CadQuery actually uses its own set of Python bindings called OCP instead of PythonOCC. I don't know if PythonOCC has raytracing capabilities within it, but OCP does not.

raphaelsetin, 2021-08-23:
That's good to know, thank you so much for the quick feedback 😃

closed 2021-08-23 by raphaelsetin