CadQuery/cadquery #823: vim / neovim - autocomplete stop working after seven repetitions

ardaij, 2021-07-12:
I use vim / nvim as editor. The code to completion working.
But if something happens seven times in a program, this stopped working.

For example:

import cadquery as cq
part = (cq.Workplane("XY")
        .circle(10)
        .extrude(10)
        .faces(">Z")
        .circle(10)
        .extrude(10)
        .faces(">Z")
        .circle(10)
        .extrude(10)
        .faces(">Z")
        .circle(10)
        .extrude(10)
        .faces(">Z")
        .circle(10)
        .extrude(10)
        .faces(">Z")
        .circle(10)
        .extrude(10)
        .faces(">Z")
        .circle(10)
        )

After the seventh ".circle()", (or seventh .lineTo(), etc. ) the code completion stop working.
I tried this with several add-ons, always the same result.

"davidhalter/jedi-vim",     "ncm2/ncm2-jedi"

fedorkotov, 2021-07-12:
This is not a CadQuery problem. You probably should report this to vim/nvim developers or to developers of whatever auto-completion plugin you are using with it.

adam-urbanczyk, 2021-07-18:
The code is valid Python, so closing.

closed 2021-07-18 by adam-urbanczyk