CadQuery/cadquery #825: Need some function from OCCT to get model's  name and by STEP

a09418849, 2021-07-16:
I need the following function to get model name : 
*  TCollection_ExtendedString.ToUTF8CString( )

But I think maybe it is my problem.
I tried to port the code from pythonocc to ocp. And the function `GetLabelName()` in `read_step_file_with_names_colors()` is not in OCCT originally, it is from pythonocc themself. I tried to generate `GetLabelName()` with python, so I think somewhat is wrong in it.
Besides this, I noticed that this can also get the model tree by the path of exploring the model.

Following code is copied and ported from Pythonocc 
@tpaviot `src/Extend/DataExchange.py` : def read_step_file_with_names_colors() : 
```python
import os
import OCP
from OCP.TDocStd import TDocStd_Document
from OCP.TCollection import TCollection_ExtendedString
from OCP.STEPCAFControl import STEPCAFControl_Reader
from OCP.IFSelect import IFSelect_RetDone, IFSelect_ItemsByEntity
from OCP.TDF import TDF_LabelSequence, TDF_Label
from OCP.TopLoc import TopLoc_Location
from OCP.Quantity import Quantity_Color, Quantity_TOC_RGB
from OCP.BRepBuilderAPI import BRepBuilderAPI_Transform

def read_step_file_with_names_colors(filename):
    """ Returns list of tuples (topods_shape, label, color)
    Use OCAF.
    """
    if not os.path.isfile(filename):
        raise FileNotFoundError("%s not found." % filename)
    # the list:
    output_shapes = {}

    # create an handle to a document
    doc = TDocStd_Document(TCollection_ExtendedString("pythonocc-doc"))

    # Get root assembly
    shape_tool = OCP.XCAFDoc.XCAFDoc_DocumentTool().ShapeTool_s(doc.Main())
    color_tool = OCP.XCAFDoc.XCAFDoc_DocumentTool().ColorTool_s(doc.Main())

    step_reader = STEPCAFControl_Reader()
    step_reader.SetColorMode(True)
    step_reader.SetLayerMode(True)
    step_reader.SetNameMode(True)
    step_reader.SetMatMode(True)
    step_reader.SetGDTMode(True)

    status = step_reader.ReadFile(filename)
    if status == IFSelect_RetDone:
        step_reader.Transfer(doc)

    locs = []
    
    def GetLabelName(self):
        txt = ''
        name = OCP.TDataStd.TDataStd_Name()
        if (not self.IsNull()) and self.FindAttribute(OCP.TDataStd.TDataStd_Name.GetID_s(),name):
            extstr = name.Get()
            txt = extstr.ToUTF8CString()
        return txt

    def _get_sub_shapes(lab, loc):
        l_subss = TDF_LabelSequence()
        shape_tool.GetSubShapes_s(lab, l_subss)
        l_comps = TDF_LabelSequence()
        shape_tool.GetComponents_s(lab, l_comps)
        name = GetLabelName(lab)
        print("Name :", name)

        if shape_tool.IsAssembly_s(lab):
            l_c = TDF_LabelSequence()
            shape_tool.GetComponents_s(lab, l_c)
            for i in range(l_c.Length()):
                label = l_c.Value(i+1)
                if shape_tool.IsReference_s(label):
                    label_reference = TDF_Label()
                    shape_tool.GetReferredShape_s(label, label_reference)
                    loc = shape_tool.GetLocation_s(label)

                    locs.append(loc)
                    _get_sub_shapes(label_reference, loc)
                    locs.pop()

        elif shape_tool.IsSimpleShape_s(lab):
            shape = shape_tool.GetShape_s(lab)

            loc = TopLoc_Location()
            for l in locs:
                loc = loc.Multiplied(l)
            c = Quantity_Color(0.5, 0.5, 0.5, Quantity_TOC_RGB)  # default color
            colorSet = False
            if (color_tool.GetInstanceColor(shape, XCAFDoc_ColorType(0), c) or
                    color_tool.GetInstanceColor(shape, XCAFDoc_ColorType(1), c) or
                    color_tool.GetInstanceColor(shape, XCAFDoc_ColorType(2), c)):
                color_tool.SetInstanceColor(shape, XCAFDoc_ColorType(0), c)
                color_tool.SetInstanceColor(shape, XCAFDoc_ColorType(1), c)
                color_tool.SetInstanceColor(shape, XCAFDoc_ColorType(2), c)
                colorSet = True
                n = c.Name()
                print('    instance color Name & RGB: ', c, n, c.Red(), c.Green(), c.Blue())

            if not colorSet:
                if (color_tool.GetColor(lab, XCAFDoc_ColorType(0), c) or
                        color_tool.GetColor(lab, XCAFDoc_ColorType(1), c) or
                        color_tool.GetColor(lab, XCAFDoc_ColorType(2), c)):

                    color_tool.SetInstanceColor(shape, XCAFDoc_ColorType(0), c)
                    color_tool.SetInstanceColor(shape, XCAFDoc_ColorType(1), c)
                    color_tool.SetInstanceColor(shape, XCAFDoc_ColorType(2), c)

                    n = c.Name()
                    print('    shape color Name & RGB: ', c, n, c.Red(), c.Green(), c.Blue())

            shape_disp = BRepBuilderAPI_Transform(shape, loc.Transformation()).Shape()
            if not shape_disp in output_shapes:
                output_shapes[shape_disp] = [GetLabelName(lab), c]
            for i in range(l_subss.Length()):
                lab_subs = l_subss.Value(i+1)
                shape_sub = shape_tool.GetShape_s(lab_subs)

                c = Quantity_Color(0.5, 0.5, 0.5, Quantity_TOC_RGB)  # default color
                colorSet = False
                if (color_tool.GetInstanceColor(shape_sub, XCAFDoc_ColorType(0), c) or
                        color_tool.GetInstanceColor(shape_sub, XCAFDoc_ColorType(1), c) or
                        color_tool.GetInstanceColor(shape_sub, XCAFDoc_ColorType(2), c)):
                    color_tool.SetInstanceColor(shape_sub, XCAFDoc_ColorType(0), c)
                    color_tool.SetInstanceColor(shape_sub, XCAFDoc_ColorType(1), c)
                    color_tool.SetInstanceColor(shape_sub, XCAFDoc_ColorType(2), c)
                    colorSet = True
                    n = c.Name()
                    print('    instance color Name & RGB: ', c, n, c.Red(), c.Green(), c.Blue())

                if not colorSet:
                    if (color_tool.GetColor(lab_subs, XCAFDoc_ColorType(0), c) or
                            color_tool.GetColor(lab_subs, XCAFDoc_ColorType(1), c) or
                            color_tool.GetColor(lab_subs, XCAFDoc_ColorType(2), c)):
                        color_tool.SetInstanceColor(shape, XCAFDoc_ColorType(0), c)
                        color_tool.SetInstanceColor(shape, XCAFDoc_ColorType(1), c)
                        color_tool.SetInstanceColor(shape, XCAFDoc_ColorType(2), c)

                        n = c.Name()
                        print('    shape color Name & RGB: ', c, n, c.Red(), c.Green(), c.Blue())
                shape_to_disp = BRepBuilderAPI_Transform(shape_sub, loc.Transformation()).Shape()
                if not shape_to_disp in output_shapes:
                    output_shapes[shape_to_disp] = [GetLabelName(lab_subs), c]

    def _get_shapes():
        labels = TDF_LabelSequence()
        shape_tool.GetFreeShapes(labels)
        #global cnt
        #cnt += 1

        print()
        print("Number of shapes at root :", labels.Length())
        print()
        for i in range(labels.Length()):
            root_item = labels.Value(i+1)
            _get_sub_shapes(root_item, None)
    _get_shapes()
    return output_shapes

read_step_file_with_names_colors('./temp.stp')
```

adam-urbanczyk, 2021-07-17:
There is no assembly import from step implemented in CQ. Is this what you are requesting?

Regarding OCCT/OCP this is a wrong tracker. To convert from `TCollection_ExtendedString` to python object you can use `TCollection_ExtendedString.ToExtString()`

a09418849, 2021-07-17:
Thanks for your reply.

I wanted to get the name from STEP first. I understand that getting the model tree from STEP is difficult.
I have tried `TCollection_ExtendedString.ToExtString()` to get string, but not thing I got.

```python
''
```

I thought maybe my funtion `GetLabelNam` has problem.
```python
    def GetLabelName(self):
        txt = ''
        name = OCP.TDataStd.TDataStd_Name()
        if (not self.IsNull()) and self.FindAttribute(OCP.TDataStd.TDataStd_Name.GetID_s(),name):
            extstr = name.Get()
            txt = extstr.ToUTF8CString()
        return txt
```
It was ported from the function GetLabelName @tpaviot by C++.
```c++
        %feature("autodoc", "Returns the label name") GetLabelName;
		%extend{
			std::string GetLabelName() {
			std::string txt;
			Handle(TDataStd_Name) name;
			if (!self->IsNull() && self->FindAttribute(TDataStd_Name::GetID(),name)) {
			TCollection_ExtendedString extstr = name->Get();
			char* str = new char[extstr.LengthOfCString()+1];
			extstr.ToUTF8CString(str);
			txt = str;
			delete[] str;}
			return txt;}
		};
```

I wanted to get name and RGB color in STEP, so I can use these as labels for machine learning.
For the part of searching color, it is work from my function.

Thanks for your reply.

adam-urbanczyk, 2021-07-17:
AFAICT `TCollection_ExtendedString.ToExtString()` is the correct method.  You can double check if your name is empty by using the `Length()` or `LengthOfCString()` methods. I have no experience with reading STEP, take a look at the OCCT docs, FreeCAD source or this program: https://github.com/fougue/mayo.

a09418849, 2021-07-17:
check this：

https://github.com/tpaviot/pythonocc-core/issues/602
https://github.com/tpaviot/pythonocc-core/commit/fb362a5b2e91e9f1550d2364b442ed2c3b7e470c

`GetLabelName` cannot ported from C++ to Python, simply.

adam-urbanczyk, 2021-07-18:
Can you explain which Python API do you need instead of posting links?  Did you try the sanity check above?

a09418849, 2021-07-18:
> Can you explain which Python API do you need instead of posting links? Did you try the sanity check above?

Sorry, I just need the way to get the name in STEP. 

In the issuse : 
tpaviot/pythonocc-core#602
They found that it can not convert the name into string by only using python. It always get empty.

adam-urbanczyk, 2021-07-18:
It is not about the name, but about how `FindAttribute` is implemented on the OCCT side and how it is wrapped by OCP. This is not a CQ issue.

closed 2021-07-18 by adam-urbanczyk

a09418849, 2021-07-20:
Thanks. 
It works.

```python
read_step_file_with_names_colors('./screw_insert_boss_asm.stp')
```

```python
import os
import OCP
from OCP.TDataStd import TDataStd_Name
from OCP.Standard import Standard_GUID
from OCP.TDocStd import TDocStd_Document
from OCP.TCollection import TCollection_ExtendedString
from OCP.STEPCAFControl import STEPCAFControl_Reader
from OCP.IFSelect import IFSelect_RetDone, IFSelect_ItemsByEntity
from OCP.TDF import TDF_LabelSequence, TDF_Label, TDF_AttributeIterator
from OCP.TopLoc import TopLoc_Location
from OCP.Quantity import Quantity_Color, Quantity_TOC_RGB
from OCP.BRepBuilderAPI import BRepBuilderAPI_Transform
from OCP.XCAFDoc import XCAFDoc_DocumentTool, XCAFDoc_ColorType
from OCP import TopAbs

TopAbs_LUT = {
        'FORWARD':TopAbs.TopAbs_FORWARD,
        'REVERSED':TopAbs.TopAbs_REVERSED,
        'INTERNAL':TopAbs.TopAbs_INTERNAL,
        'EXTERNAL':TopAbs.TopAbs_EXTERNAL,
        'COMPOUND':TopAbs.TopAbs_COMPOUND,
        'COMPSOLID':TopAbs.TopAbs_COMPSOLID,
        'SOLID':TopAbs.TopAbs_SOLID,
        'SHELL':TopAbs.TopAbs_SHELL,
        'FACE':TopAbs.TopAbs_FACE,
        'WIRE':TopAbs.TopAbs_WIRE,
        'EDGE':TopAbs.TopAbs_EDGE,
        'VERTEX':TopAbs.TopAbs_VERTEX,
        'SHAPE':TopAbs.TopAbs_SHAPE,
        'IN':TopAbs.TopAbs_IN,
        'OUT':TopAbs.TopAbs_OUT,
        'ON':TopAbs.TopAbs_ON,
        'UNKNOWN':TopAbs.TopAbs_UNKNOWN,
        'ORIENTATION':TopAbs.TopAbs_Orientation,
        'SHAPEENUM':TopAbs.TopAbs_ShapeEnum,
        'STATE':TopAbs.TopAbs_State
    }

##########################
# Step import #
##########################

def read_step_file_with_names_colors(filename : str):
    """ Returns list of tuples (topods_shape, label, color)
    Use OCAF.
    """
    if not os.path.isfile(filename):
        raise FileNotFoundError("%s not found." % filename)
    # the list:
    output_shapes = {}

    # create an handle to a document
    doc = TDocStd_Document(TCollection_ExtendedString("pythonocc-doc"))

    # Get root assembly
    shape_tool = OCP.XCAFDoc.XCAFDoc_DocumentTool().ShapeTool_s(doc.Main())
    color_tool = OCP.XCAFDoc.XCAFDoc_DocumentTool().ColorTool_s(doc.Main())

    step_reader = STEPCAFControl_Reader()
    step_reader.SetColorMode(True)
    step_reader.SetLayerMode(True)
    step_reader.SetNameMode(True)
[... 120 lines cut ...]
                    output_shapes[shape_to_disp] = [GetLabelName(lab_subs), tree_temp, (c.Red(), c.Green(), c.Blue())]

    def _get_shapes():
        labels = TDF_LabelSequence()
        shape_tool.GetFreeShapes(labels)
        #global cnt
        #cnt += 1

        print()
        print("Number of shapes at root :", labels.Length())
        print()
        for i in range(labels.Length()):
            root_item = labels.Value(i+1)
            _get_sub_shapes(root_item, None, '.'+str(i))
    _get_shapes()
    data = []
    for item,property_ in zip( list(output_shapes.keys()),list(output_shapes.values()) ):
        data.append({
            'CQ_OCP_TopDS': item,
            'Tree': property_[1],
            'Name': property_[0],
            'RGB': property_[2]
        })
    return data
```

tpaviot, 2021-07-20:
@a09418849 This code snippet looks like a port of the related pythonocc-core ``read_step_file_with_names_colors`` function (https://github.com/tpaviot/pythonocc-core/blob/a244920ada69c337c6049e61d3e2809dbb489eb8/src/Extend/DataExchange.py#L127), mostly by removing comments and replacing ``from OCC.* import [..]`` with ``from OCP.* import [...]``. I'm happy you consider this as useful work, don't forget to credit the original authors along with the LGPL V3 header if ever you plan to redistribute the code.

a09418849, 2021-07-20:
@tpaviot I apologize for that.
If I offend you by deleting and modifing your code with forgetting to inform you, I am terribly sorry about that.
I will not forget to credit you along with any public license  if I ever plan to redistribute the code in any place.

In my modifing, I add the attribute of Tree to get model's path.
The Tree like this:
* 0.0.1.0.2(0) <--would be SOLID
* 0.0.1.0.2(2) <--would be other FACE belong to SOLID
I haven't tried if the STEP including edge what may happen.

```python
##The function in this file of read_step_file_with_names_colors_tree is copied, ported and modified from Thomas Paviot (tpaviot@gmail.com).
##The original Copyright :
"""
##Copyright 2018 Thomas Paviot (tpaviot@gmail.com)
##
##This file is part of pythonOCC.
##
##pythonOCC is free software: you can redistribute it and/or modify
##it under the terms of the GNU Lesser General Public License as published by
##the Free Software Foundation, either version 3 of the License, or
##(at your option) any later version.
##
##pythonOCC is distributed in the hope that it will be useful,
##but WITHOUT ANY WARRANTY; without even the implied warranty of
##MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
##GNU Lesser General Public License for more details.
##
##You should have received a copy of the GNU Lesser General Public License
##along with pythonOCC.  If not, see <http://www.gnu.org/licenses/>.
"""
## You can get original code from below address.
## https://github.com/tpaviot/pythonocc-core/blob/a244920ada69c337c6049e61d3e2809dbb489eb8/src/Extend/DataExchange.py#L127

import os
import OCP
from OCP.TDataStd import TDataStd_Name
from OCP.Standard import Standard_GUID
from OCP.TDocStd import TDocStd_Document
from OCP.TCollection import TCollection_ExtendedString
from OCP.STEPCAFControl import STEPCAFControl_Reader
from OCP.IFSelect import IFSelect_RetDone, IFSelect_ItemsByEntity
from OCP.TDF import TDF_LabelSequence, TDF_Label, TDF_AttributeIterator
from OCP.TopLoc import TopLoc_Location
from OCP.Quantity import Quantity_Color, Quantity_TOC_RGB
from OCP.BRepBuilderAPI import BRepBuilderAPI_Transform
from OCP.XCAFDoc import XCAFDoc_DocumentTool, XCAFDoc_ColorType
from OCP import TopAbs

TopAbs_LUT = {
        'FORWARD':TopAbs.TopAbs_FORWARD,
        'REVERSED':TopAbs.TopAbs_REVERSED,
        'INTERNAL':TopAbs.TopAbs_INTERNAL,
        'EXTERNAL':TopAbs.TopAbs_EXTERNAL,
        'COMPOUND':TopAbs.TopAbs_COMPOUND,
        'COMPSOLID':TopAbs.TopAbs_COMPSOLID,
        'SOLID':TopAbs.TopAbs_SOLID,
        'SHELL':TopAbs.TopAbs_SHELL,
        'FACE':TopAbs.TopAbs_FACE,
        'WIRE':TopAbs.TopAbs_WIRE,
        'EDGE':TopAbs.TopAbs_EDGE,
        'VERTEX':TopAbs.TopAbs_VERTEX,
        'SHAPE':TopAbs.TopAbs_SHAPE,
        'IN':TopAbs.TopAbs_IN,
        'OUT':TopAbs.TopAbs_OUT,
        'ON':TopAbs.TopAbs_ON,
        'UNKNOWN':TopAbs.TopAbs_UNKNOWN,
        'ORIENTATION':TopAbs.TopAbs_Orientation,
        'SHAPEENUM':TopAbs.TopAbs_ShapeEnum,
        'STATE':TopAbs.TopAbs_State
[... 155 lines cut ...]
                    output_shapes[shape_to_disp] = [GetLabelName(lab_subs), tree_temp, (c.Red(), c.Green(), c.Blue())]

    def _get_shapes():
        labels = TDF_LabelSequence()
        shape_tool.GetFreeShapes(labels)
        #global cnt
        #cnt += 1

        print()
        print("Number of shapes at root :", labels.Length())
        print()
        for i in range(labels.Length()):
            root_item = labels.Value(i+1)
            _get_sub_shapes(root_item, None, '.'+str(i))
    _get_shapes()
    data = []
    for item,property_ in zip( list(output_shapes.keys()),list(output_shapes.values()) ):
        data.append({
            'CQ_OCP_TopDS': item,
            'Tree': property_[1],
            'Name': property_[0],
            'RGB': property_[2]
        })
    return data
```