CadQuery/cadquery #826: STEP import as an assembly

adam-urbanczyk, 2021-07-17:


a09418849, 2021-07-17:
Try this:
https://drive.google.com/file/d/1uNoSakAb8fRtFri1-0YIOQsNMPdAX5u1/view?usp=sharing

I used pythonOCC to get name and RGB. 
Just open `Untitled.ipynb`. And there are many empty item  in my model.

In `items_occ` : LIST[( <class 'TopoDS_Solid'>,  str(model name), tuple( int(R), int(G), int(B) ), str(model tree path) )]
In `items_cq_ocp` : LIST[( <OCP.TopoDS.TopoDS_Shape>, str(model tree path) )]

I used `str(model tree path)` to know which items in occ and ocp are equal.
 `str(model tree path)`  like this `0.0.1.0.1(0)`, and it is interesting that if `()` is more than 0 like `(2)`, it would be the face type.

For the problem of `TCollection_ExtendedString`, try this : 
tpaviot/pythonocc-core#602
tpaviot/pythonocc-core@fb362a5

But I have other problem. 
In this line : 
```python
assembly.save('./assembly_by_cq.step')
```
I tried to open `assembly_by_cq.step` by CREO 6.0. It has GFX Glitch like this : 
[image]

If posible, please fix.

a09418849, 2021-07-17:
> Try this:
> https://drive.google.com/file/d/1uNoSakAb8fRtFri1-0YIOQsNMPdAX5u1/view?usp=sharing
>
> I used pythonOCC to get name and RGB.
> Just open `Untitled.ipynb`. And there are many empty item in my [...]

It would have same problem when I save STEP first and then reload STEP in cadquery.
[image]
The expect:
[image]

The problem only in my code

adam-urbanczyk, 2021-07-18:
@a09418849 please don't pollute this issue with unrelated comments - it is meant to track the future implementation of STEP import. If you found a bug, you can open a new issue and provide a reproducible example there.

closed 2025-10-01 by adam-urbanczyk