CadQuery/cadquery #827: Add import/export IGES and add filename extension .stp ( for STEP )

a09418849, 2021-07-17:
The filename extension of STEP has `.step` and `.stp`. In Creo 6.0, it used `.stp`.
Like this : 
```python
cq.assembly.save('./model.stp')
```
For exporting IGES, try `export_iges_file` @michaelgale
https://github.com/michaelgale/cq-kit/blob/master/cqkit/cq_files.py
```python
def export_iges_file(shape, filename, author=None, organization=None):
    """ Exports a shape to an IGES file.  """
    # initialize iges writer in BRep mode
    writer = IGESControl_Writer("MM", 1)
    Interface_Static_SetIVal("write.iges.brep.mode", 1)
    # write surfaces with iges 5.3 entities
    Interface_Static_SetIVal("write.convertsurface.mode", 1)
    Interface_Static_SetIVal("write.precision.mode", 1)
    if author is not None:
        Interface_Static_SetCVal("write.iges.header.author", author)
    if organization is not None:
        Interface_Static_SetCVal("write.iges.header.company", organization)
    writer.AddShape(shape.val().wrapped)
    writer.ComputeModel()
    writer.Write(filename)
``` 
The IGES cannot support assembly and multisolid and so on, only solid or part called in Creo is the best for IGES.
P.S. : The filename extension of IGES also has two name, `.iges` and `igs`.

a09418849, 2021-07-18:
For importing IGES, try `read_iges_file` @tpaviot 
https://github.com/tpaviot/pythonocc-core/blob/a244920ada69c337c6049e61d3e2809dbb489eb8/src/Extend/DataExchange.py
```python
def read_iges_file(filename, return_as_shapes=False, verbosity=False, visible_only=False):
    """ read the IGES file and returns a compound
    filename: the file path
    return_as_shapes: optional, False by default. If True returns a list of shapes,
                      else returns a single compound
    verbosity: optionl, False by default.
    """
    if not os.path.isfile(filename):
        raise FileNotFoundError("%s not found." % filename)

    iges_reader = IGESControl_Reader()
    iges_reader.SetReadVisible(visible_only)
    status = iges_reader.ReadFile(filename)

    _shapes = []

    if status == IFSelect_RetDone:  # check status
        if verbosity:
            failsonly = False
            iges_reader.PrintCheckLoad(failsonly, IFSelect_ItemsByEntity)
            iges_reader.PrintCheckTransfer(failsonly, IFSelect_ItemsByEntity)
        iges_reader.TransferRoots()
        nbr = iges_reader.NbRootsForTransfer()
        for _ in range(1, nbr+1):
            nbs = iges_reader.NbShapes()
            if nbs == 0:
                print("At least one shape in IGES cannot be transfered")
            elif nbr == 1 and nbs == 1:
                a_res_shape = iges_reader.Shape(1)
                if a_res_shape.IsNull():
                    print("At least one shape in IGES cannot be transferred")
                else:
                    _shapes.append(a_res_shape)
            else:
                for i in range(1, nbs+1):
                    a_shape = iges_reader.Shape(i)
                    if a_shape.IsNull():
                        print("At least one shape in STEP cannot be transferred")
                    else:
                        _shapes.append(a_shape)
    # if not return as shapes
    # create a compound and store all shapes
    if not return_as_shapes:
        builder = BRep_Builder()
        compound = TopoDS_Compound()
        builder.MakeCompound(compound)
        for s in _shapes:
            builder.Add(compound, s)
        _shapes = compound
    return _shapes
```

adam-urbanczyk, 2021-07-18:
-1 on this - IGES is really inferior wrt STEP. As you mentioned yourself there are many things not supported by IGES. If you really need it you can always use OCP directly.

jmwright, 2021-07-18:
@adam-urbanczyk What's wrong with supporting IGES import as long as OCCT enables it? Maybe someone will have a large amount of IGES files that they want to import, transform, and export to STEP (or another format).

adam-urbanczyk, 2021-07-19:
There is nothing "wrong", but it supports much fewer features and is to my understanding obsolete. A user in such a need can always use OCCT/OCP directly.

a09418849, 2021-07-20:
I think why the commercial softwares of CAD retain exporting `IGES` is that `IGES` don't have assembly class so it can more stably export the model comparing with STEP. 
Because of relative coordinates, if you import `STEP` in `.prt` file, the solid may explode.

adam-urbanczyk, 2021-07-20:
If Creo "explodes" something that is a valid STEP, you need to contact their support. If you are using CQ assemblies, you can convert the to compound and export a STEP without any hierarchy. 

I really think that we need to draw a line somewhere and not try to support everything but focus on what is essential (I'd say that e.g. STEP import is more important than IGES support and so is a zyllion of modeling feature on this tracker). @jmwright @marcus7070 if you think it makes sense to add this and want to spend time, feel free to reopen,

closed 2021-07-20 by adam-urbanczyk