CadQuery/cadquery #832: Adding revolveCut and sweepCut

Jojain, 2021-07-21:
Hello,

As far as I know currently we can't use `revolve` and `sweep` to remove material with the fluent API.
I think it would be nice to have it, and to have in general for each matter creation operation the opposite matter removal operation.

This could be implemented either as a param in the discussed methods as `additive = False` for example.
Or to implement additionals methods as it has been done with `extrude` and `cutBlind`, let's say `revolveCut` and `sweepCut`.

PS : Personal opinion here : I think it would be nice to have coherent way to have material addition/removal operations. So either having a standard method with a param to remove material or 2 methods but not mix these two ways in order to keep consistency.

adam-urbanczyk, 2021-07-21:
There is already an issue to track this: https://github.com/CadQuery/cadquery/issues/30. The proposal would be to extend combine with other options.

Jojain, 2021-07-23:
Sorry, I'm closing this and continuing discussion on #30

closed 2021-07-23 by Jojain