CadQuery/cadquery #833: 2D shapes are not consistently union'ed when extruded

jdthorpe, 2021-07-23:
I noticed that when I create multiple overlapping 2D shapes and extrude them
they are generally `union()`'ed together into a single object:

```py
single_object = (
    cq.Workplane("XY")
    .moveTo(1.5,1)
    .rect(3,2)
    .moveTo(2,2.1)
    .polygon(5,0.5)
    .extrude(1)
)
```

and this works nicely when all of the shapes are well behaved shapes such as
`.rect()`, `.circle()`, or `.ellipse()`.  However when any two of them behave
like poly-lines the result of `.extrude()` is a set of separate but overlapping
objects.

```py
multiple_overlapping_objects = (
    cq.Workplane("XY")
    # equivalent rectangle from individual lines
    .line(0,2)
    .line(3,0)
    .line(0,-2)
    .close()
    # same polygon as above
    .moveTo(2,2.1)
    .polygon(5,0.5)
    .extrude(1)
)
```

as evidenced by:

```py
single_object.faces(">Z").wires().size()
# > 1
multiple_overlapping_objects.faces(">Z").wires().size()
# > 2
```

Note that if I tag the plane  (`cq.Workplane("XY").tag("xy")`) and extrude the
intermediate shapes (`.extrude(1).workplaneFromTagged("xy")`), I get the expected result -- i.e. a single object that is the union of all the extruded shapes:

```py
( 
    cq.Workplane("XY").tag("xy")
    # equivalent rectangle from individual lines
    .line(0,2)
    .line(3,0)
    .line(0,-2)
    .close()
    .extrude(1).workplaneFromTagged("xy")
    # define the outline
    .moveTo(2,2.1)
    .polygon(5,0.5)
    .extrude(1)
    .faces(">Z").wires().size()
)
# > 1
```

Possibly related to [this issue](https://github.com/CadQuery/cadquery/issues/759)

Jojain, 2021-07-23:
I cannot reproduce your exemple on Cadquery 2.1. Which version do you have?

jdthorpe, 2021-07-23:
I'm using Cadquery 2.1 and I'm pretty sure I used [cadquery=master](https://cadquery.readthedocs.io/en/latest/installation.html#command-line-installation), which I installed in to a new Conda environment about a week ago.   Here's my `pip freeze` output: 

```
(cadquery-test) jasonthorpe@hi kb-cq % pip freeze
alabaster==0.7.12
appdirs @ file:///home/conda/feedstock_root/build_artifacts/appdirs_1603108395799/work
applaunchservices==0.2.1
appnope @ file:///Users/runner/miniforge3/conda-bld/appnope_1610094673755/work
argh @ file:///home/conda/feedstock_root/build_artifacts/argh_1595627874344/work
astroid @ file:///Users/runner/miniforge3/conda-bld/astroid_1625057737111/work
async-generator==1.10
atomicwrites @ file:///home/conda/feedstock_root/build_artifacts/atomicwrites_1588182545583/work
attrs @ file:///home/conda/feedstock_root/build_artifacts/attrs_1620387926260/work
autopep8 @ file:///home/conda/feedstock_root/build_artifacts/autopep8_1615918605177/work
Babel @ file:///home/conda/feedstock_root/build_artifacts/babel_1619719576210/work
backcall @ file:///home/conda/feedstock_root/build_artifacts/backcall_1592338393461/work
backports.functools-lru-cache @ file:///home/conda/feedstock_root/build_artifacts/backports.functools_lru_cache_1618230623929/work
black @ file:///home/conda/feedstock_root/build_artifacts/black-recipe_1626468239465/work
bleach @ file:///home/conda/feedstock_root/build_artifacts/bleach_1626371669429/work
brotlipy==0.7.0
cadquery==2.1
certifi==2021.5.30
cffi @ file:///Users/runner/miniforge3/conda-bld/cffi_1625835329396/work
chardet @ file:///Users/runner/miniforge3/conda-bld/chardet_1610093613037/work
charset-normalizer @ file:///home/conda/feedstock_root/build_artifacts/charset-normalizer_1626371162869/work
click @ file:///Users/runner/miniforge3/conda-bld/click_1621503946569/work
cloudpickle @ file:///home/conda/feedstock_root/build_artifacts/cloudpickle_1598400192773/work
colorama @ file:///home/conda/feedstock_root/build_artifacts/colorama_1602866480661/work
CQ-editor==0.3.0.dev0
cryptography @ file:///Users/runner/miniforge3/conda-bld/cryptography_1616851523607/work
debugpy @ file:///Users/runner/miniforge3/conda-bld/debugpy_1620178588063/work
decorator @ file:///home/conda/feedstock_root/build_artifacts/decorator_1621187651333/work
defusedxml @ file:///home/conda/feedstock_root/build_artifacts/defusedxml_1615232257335/work
diff-match-patch @ file:///home/conda/feedstock_root/build_artifacts/diff-match-patch_1594679019945/work
docutils @ file:///Users/runner/miniforge3/conda-bld/docutils_1618676291311/work
entrypoints @ file:///home/conda/feedstock_root/build_artifacts/entrypoints_1605121927639/work/dist/entrypoints-0.3-py2.py3-none-any.whl
ezdxf @ file:///Users/runner/miniforge3/conda-bld/ezdxf_1624268799212/work
flake8 @ file:///home/conda/feedstock_root/build_artifacts/flake8_1601874335748/work
future @ file:///Users/runner/miniforge3/conda-bld/future_1610147271444/work
idna @ file:///home/conda/feedstock_root/build_artifacts/idna_1609836280497/work
imagesize==1.2.0
importlib-metadata @ file:///Users/runner/miniforge3/conda-bld/importlib-metadata_1625463744025/work
intervaltree==3.0.2
ipykernel @ file:///Users/runner/miniforge3/conda-bld/ipykernel_1626354780449/work/dist/ipykernel-6.0.2-py3-none-any.whl
ipython @ file:///Users/runner/miniforge3/conda-bld/ipython_1625027511604/work
ipython-genutils==0.2.0
isort @ file:///home/conda/feedstock_root/build_artifacts/isort_1625737910378/work
jedi @ file:///Users/runner/miniforge3/conda-bld/jedi_1605054534285/work
Jinja2 @ file:///home/conda/feedstock_root/build_artifacts/jinja2_1621419064915/work
jsonschema @ file:///home/conda/feedstock_root/build_artifacts/jsonschema_1614815863336/work
jupyter-client @ file:///home/conda/feedstock_root/build_artifacts/jupyter_client_1615693636836/work
jupyter-core @ file:///Users/runner/miniforge3/conda-bld/jupyter_core_1612125283939/work
jupyterlab-pygments @ file:///home/conda/feedstock_root/build_artifacts/jupyterlab_pygments_1601375948261/work
keyring @ file:///Users/runner/miniforge3/conda-bld/keyring_1616689837979/work
lazy-object-proxy @ file:///Users/runner/miniforge3/conda-bld/lazy-object-proxy_1616507436989/work
Logbook @ file:///Users/runner/miniforge3/conda-bld/logbook_1610342923857/work
loguru @ file:///Users/runner/miniforge3/conda-bld/loguru_1618589909889/work
MarkupSafe @ file:///Users/runner/miniforge3/conda-bld/markupsafe_1621455730541/work
matplotlib-inline @ file:///home/conda/feedstock_root/build_artifacts/matplotlib-inline_1618935594181/work
mccabe==0.6.1
mistune @ file:///Users/runner/miniforge3/conda-bld/mistune_1624941339085/work
mypy-extensions @ file:///Users/runner/miniforge3/conda-bld/mypy_extensions_1610127199348/work
nbclient @ file:///home/conda/feedstock_root/build_artifacts/nbclient_1614336084111/work
nbconvert @ file:///Users/runner/miniforge3/conda-bld/nbconvert_1624472871251/work
nbformat @ file:///home/conda/feedstock_root/build_artifacts/nbformat_1617383142101/work
nest-asyncio @ file:///home/conda/feedstock_root/build_artifacts/nest-asyncio_1617163391303/work
nptyping @ file:///home/conda/feedstock_root/build_artifacts/nptyping_1620496790955/work
numpy @ file:///Users/runner/miniforge3/conda-bld/numpy_1624399924833/work
numpydoc @ file:///home/conda/feedstock_root/build_artifacts/numpydoc_1601580905698/work
packaging @ file:///home/conda/feedstock_root/build_artifacts/packaging_1625323647219/work
[... 50 lines cut ...]
[... cut ...]
watchdog @ file:///Users/runner/miniforge3/conda-bld/watchdog_1624720256750/work
wcwidth @ file:///home/conda/feedstock_root/build_artifacts/wcwidth_1600965781394/work
webencodings==0.5.1
wrapt @ file:///Users/runner/miniforge3/conda-bld/wrapt_1624971809674/work
wurlitzer @ file:///Users/runner/miniforge3/conda-bld/wurlitzer_1617142534010/work
yapf @ file:///home/conda/feedstock_root/build_artifacts/yapf_1618790300295/work
zipp @ file:///home/conda/feedstock_root/build_artifacts/zipp_1625284368454/work
(cadquery-test) jasonthorpe@hi kb-cq %```

jdthorpe, 2021-07-23:
I reinstalled again in a new conda environment just now and got the same result as in the OP.  I noticed that during the conda install , that the original solution / installation fails and conda has to fall back to a second repodata source.  Also FWIW I'm on OSX.

Here's the output from the conda install in the new conda environment: 

```sh
% conda install -c cadquery -c conda-forge cadquery=master

Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: done

==> WARNING: A newer version of conda exists. <==
  current version: 4.7.12
  latest version: 4.10.3

Please update conda by running

    $ conda update -n base -c defaults conda

## Package Plan ##

  environment location: /usr/local/anaconda3/envs/cqtest2

  added / updated specs:
    - cadquery=master

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    cadquery-master            |            py3.9         377 KB  cadquery
    libffi-3.3                 |       h046ec9c_2          45 KB  conda-forge
    ncurses-6.2                |       h2e338ed_4         881 KB  conda-forge
    nlopt-2.7.0                |   py39h783de76_1         389 KB  conda-forge
    numpy-1.21.1               |   py39h7eed0ac_0         5.8 MB  conda-forge
    pip-21.1.3                 |     pyhd8ed1ab_0         1.1 MB  conda-forge
    python-3.9.6               |hd187cdc_1_cpython        12.4 MB  conda-forge
    readline-8.1               |       h05e3726_0         266 KB  conda-forge
    setuptools-49.6.0          |   py39h6e9494a_3         950 KB  conda-forge
    sqlite-3.36.0              |       h23a322b_0         1.7 MB  conda-forge
    tk-8.6.10                  |       h0419947_1         3.3 MB  conda-forge
    tzdata-2021a               |       he74cb21_1         121 KB  conda-forge
    wheel-0.36.2               |     pyhd3deb0d_0          31 KB  conda-forge
    xz-5.2.5                   |       haf1e3a3_1         228 KB  conda-forge
    zlib-1.2.11                |    h7795811_1010         102 KB  conda-forge
    ------------------------------------------------------------
                                           Total:        27.6 MB

The following NEW packages will be INSTALLED:

  appnope            conda-forge/osx-64::appnope-0.1.2-py39h6e9494a_1
  backcall           conda-forge/noarch::backcall-0.2.0-pyh9f0ad1d_0
  backports          conda-forge/noarch::backports-1.0-py_2
  backports.functoo~ conda-forge/noarch::backports.functools_lru_cache-1.6.4-pyhd8ed1ab_0
  bzip2              conda-forge/osx-64::bzip2-1.0.8-h0d85af4_4
  c-ares             conda-forge/osx-64::c-ares-1.17.1-h0d85af4_1
  ca-certificates    conda-forge/osx-64::ca-certificates-2021.5.30-h033912b_0
  cadquery           cadquery/osx-64::cadquery-master-py3.9
  certifi            conda-forge/osx-64::certifi-2021.5.30-py39h6e9494a_0
  curl               conda-forge/osx-64::curl-7.77.0-hb861fe1_0
  decorator          conda-forge/noarch::decorator-5.0.9-pyhd8ed1ab_0
  double-conversion  conda-forge/osx-64::double-conversion-3.1.5-he49afe7_2
  eigen              conda-forge/osx-64::eigen-3.3.9-h926bf3e_1
  expat              conda-forge/osx-64::expat-2.4.1-he49afe7_0
  ezdxf              conda-forge/osx-64::ezdxf-0.16.4-py39hf018cea_0
  ffmpeg             conda-forge/osx-64::ffmpeg-4.3.1-hb787a4c_2
  font-ttf-dejavu-s~ conda-forge/noarch::font-ttf-dejavu-sans-mono-2.37-hab24e00_0
  font-ttf-inconsol~ conda-forge/noarch::font-ttf-inconsolata-3.000-h77eed37_0
[... 74 lines cut ...]
  pugixml            conda-forge/osx-64::pugixml-1.11.4-he49afe7_0
  pygments           conda-forge/noarch::pygments-2.9.0-pyhd8ed1ab_0
  pyparsing          conda-forge/noarch::pyparsing-2.4.7-pyh9f0ad1d_0
  python             conda-forge/osx-64::python-3.9.6-hd187cdc_1_cpython
  python_abi         conda-forge/osx-64::python_abi-3.9-2_cp39
  rapidjson          conda-forge/osx-64::rapidjson-1.1.0-hb1e8313_1002
  readline           conda-forge/osx-64::readline-8.1-h05e3726_0
  setuptools         conda-forge/osx-64::setuptools-49.6.0-py39h6e9494a_3
  sqlite             conda-forge/osx-64::sqlite-3.36.0-h23a322b_0
  tbb                conda-forge/osx-64::tbb-2020.2-h940c156_4
  tbb-devel          conda-forge/osx-64::tbb-devel-2020.2-h940c156_4
  tk                 conda-forge/osx-64::tk-8.6.10-h0419947_1
  traitlets          conda-forge/noarch::traitlets-5.0.5-py_0
  typing_extensions  conda-forge/noarch::typing_extensions-3.10.0.0-pyha770c72_0
  typish             conda-forge/noarch::typish-1.9.2-pyhd8ed1ab_0
  tzdata             conda-forge/noarch::tzdata-2021a-he74cb21_1
  utfcpp             conda-forge/osx-64::utfcpp-3.2.1-h694c41f_0
  vtk                conda-forge/osx-64::vtk-9.0.1-no_osmesa_py39h519f44b_107
  wcwidth            conda-forge/noarch::wcwidth-0.2.5-pyh9f0ad1d_2
  wheel              conda-forge/noarch::wheel-0.36.2-pyhd3deb0d_0
  x264               conda-forge/osx-64::x264-1!161.3030-h0d85af4_1
  xz                 conda-forge/osx-64::xz-5.2.5-haf1e3a3_1
  zlib               conda-forge/osx-64::zlib-1.2.11-h7795811_1010
  zstd               conda-forge/osx-64::zstd-1.5.0-h582d3a0_0
```

adam-urbanczyk, 2021-07-25:
You cannot extrude intersecting wires in a reproducible way. There is actually no union operation involved in this issue.