CadQuery/cadquery #834: docs HTML API reference improve usability by splitting into multiple pages

lorenzncode, 2021-07-24:
Currently, the class reference is flat. It is a *huge* page and in my opinion this hinders usability.  While the Python source splits up Workplane, Assembley, Selector, lower level Geometry, and Topological classes, the HTML docs dump everything into a single large page.  The Python source has its structure for a reason and the docs would benefit from a similar structure.

We can search the HTML docs in two ways

- The built-in Sphinx "Search docs" feature 

   This is very useful but requires extra navigation.

- web browser page search "Ctrl+F"

   This type of search can potentially provide quick navigation of the class reference.  With the flat page it is difficult to navigate.  I think it would make more sense to have say a page for Workplane separate from other classes.  Then you would bookmark say the Workplane page when working with Workplane and can quickly navigate the class methods.  When working with the say the geom.py classes the user could specifically navigate to a page dedicated to these classes.

As a workaround, yes we can open the source files.   Even with the source files open in tabs in a text editor it would be nice to augment this with the HTML reference.

adam-urbanczyk, 2021-07-25:
In what sense usability is hindered? What can you achieve with split pages that you cannot achieve with a single one?

lorenzncode, 2021-07-25:
I mean in the sense of quickly navigating the class reference.  By navigation I'm referring to both browsing through the list of class methods and plain text search within the browser page.  

Searching common terms such as plane, angle, rotate will generate many hits and take more time to go through.  Say a new user searches for "circle" in the HTML docs.  They will first find makeCircle() which should not be of interest to them at this time.

I'm not necessarily suggesting to split every class to a separate page but I think some structure similar to the source itself could improve navigation through the docs.  A new user could stick with Workplane when getting started.  Someone working with the lower level APIs might open another page dedicated to those classes.

Improving the speed at which I can navigate and find what I'm looking for is what I think I can achieve by the split pages.

revolter, 2021-08-13:
I stumbled upon this issue too. It takes a lot of time, and a lot of "find next" to find what I need. Maybe have a sidebar with a list of all the available functions, so you can easily scroll to them, find the one you need, and then click to jump to its full documentation.