CadQuery/cadquery #836: Help with my cadquery plugin

zxo102, 2021-07-29:
Hi, there. I want to make a tunnel plugin. Here is the original code for  a tunnel model:
```
width =4.0
height=2.0
result = cq.Workplane("XY").lineTo(width, 0).lineTo(width,height).threePointArc((width/2.0, height*1.5),(0.0, height)).close().extrude(10)
show_object(result, options = {"color": "yellow", "alpha": 0.7})
```
But the following tunnel plugin does not work with AttributeError: 'Workplane' object has no attribute 'wrapped'.     how do I make it right?
```
def makeArchTunnel(self,height,width,length):
    ss = cq.Workplane().lineTo(width, 0).lineTo(width,height).threePointArc((width/2.0, height*1.5),(0.0, height)).close().extrude(length)
    mySolid = cq.Solid.makeSolid(ss)
    return mySolid

cq.Workplane.makeArchTunnel = makeArchTunnel
result = (cq.Workplane("XY").makeArchTunnel(2.0,4.0,10))
show_object(result, options = {"color": "green", "alpha": 0.7})
```

Thanks a lot.

ouyang

jmwright, 2021-07-29:
If you wanted to pass the `Workplane` object to `makeSolid` you would have to call `val()` on it first to get the lower level CadQuery object, which is what `makeSolid` expects. You don't need to do that though since you already have access to the solid you want. You can just use the `val()` of your workplane directly. I've modified the code below to do that and to also follow the `eachpoint` method from the [plugin example](https://cadquery.readthedocs.io/en/latest/extending.html#plugin-example).
```python
import cadquery as cq

def makeArchTunnel(self,height,width,length):
    def _makeArchTunnel(loc):
        ss = cq.Workplane().lineTo(width, 0).lineTo(width,height).threePointArc((width/2.0, height*1.5),(0.0, height)).close().extrude(length)
        mySolid = ss.val().locate(loc)
        return mySolid

    return self.eachpoint(_makeArchTunnel)

cq.Workplane.makeArchTunnel = makeArchTunnel
result = (cq.Workplane("XY").makeArchTunnel(2.0,4.0,10))
show_object(result, options = {"color": "green", "alpha": 0.7})
```

zxo102, 2021-07-30:
Thanks, jmwright. It works now.

closed 2021-07-30 by zxo102