CadQuery/cadquery #838: Where is the parent object?

zxo102, 2021-08-02:
Hi there, 

I use [the Introspective Example](https://cadquery.readthedocs.io/en/latest/primer.html?highlight=tidy_repr#an-introspective-example) to run the following cadquery script:
```
part = cq.Workplane()
print("1. Wp==> ",part)
part = part.box(1, 1, 1)
print("2. box==> ",part)
part = part.tag("base")
print("3. tag==> ",part)
part = part.faces(">>Z")
print("4. faces==> ",part)
part = part.wires()
print("5. wires==> ",part)
part = part.toPending()
print("6. toPending==> ",part)
part = part.translate((0.1, 0.1, 1.0))
print("7. translate==> ",part)
part = part.toPending()
print("8. toPending==> ",part)
part = part.loft()
print("9. loft==> ",part)
part = part.faces(">>X", tag="base")
print("10. faces-tag-bases==> ",part)
part = part.workplane(centerOption="CenterOfMass")
print("12. wp==> ",part)
part = part.circle(0.2)
print("13. circle==> ",part)
part = part.extrude(1)
print("14. extrude==> ",part)
```
Everything is understandable except the parent object "0x000002C40F0D5CA0" at step 14 is from "nowhere" of the above steps, as shown in the following log:
```
...
14. extrude==>  Workplane object at 0x000002C40F0E2BE0:
  parent: Workplane object at 0x000002C40F0D5CA0
  plane: Plane object at 0x000002C40F0E5D60:
...
```

Where is the parent object (parent: Workplane object at 0x000002C40F0D5CA0)?

lorenzncode, 2021-08-03:
If you specify extrude with the clean=False argument you should find that the parent object after extrude is from the previous step.  By default clean=True, so extrude includes an additional call to newObject internally.

zxo102, 2021-08-03:
Thanks,  lorenzncode.  Found the parent object with the clean=Flase.

marcus7070, 2021-08-03:
It's great to see someone following all the way through to the end of that example and still paying attention, thanks @zxo102! And thanks for the quick explanation @lorenzncode. I'll add a note to the example explaining what's going on.