CadQuery/cadquery #840: The script in Assemblies Doc

zxo102, 2021-08-04:
With latest version of cadquery and cq-editor on my Windows 10, the script in [Assemblies](https://cadquery.readthedocs.io/en/latest/primer.html?highlight=tidy_repr#assemblies)
```
from cadquery import *
w = 10
d = 10
h = 10
part1 = Workplane().box(2*w,2*d,h)
part2 = Workplane().box(w,d,2*h)
part3 = Workplane().box(w,d,3*h)
assy = (
    Assembly(part1, loc=Location(Vector(-w,0,h/2)))
    .add(part2, loc=Location(Vector(1.5*w,-.5*d,h/2)), color=Color(0,0,1,0.5))
    .add(part3, loc=Location(Vector(-.5*w,-.5*d,2*h)), color=Color("red"))
)
```
results in 

[image]

which is different from the output in [Assemblies](https://cadquery.readthedocs.io/en/latest/primer.html?highlight=tidy_repr#assemblies)
[image]

zxo102, 2021-08-04:
Adding  "show_object(assy)" at end returns the normal one :))
[image]

jmwright, 2021-08-06:
@zxo102 Yes, you need `show_object()` to display an assembly correctly. Otherwise CQ-editor does its magic to infer which objects you want to display and ignores their assembly attributes.

You have to use `toCompound()` when exporting an assembly to STL because the STL format does not support assemblies like the STEP format does.

```python
from cadquery import *

w = 10
d = 10
h = 10

part1 = Workplane().box(2*w,2*d,h)
part2 = Workplane().box(w,d,2*h)
part3 = Workplane().box(w,d,3*h)
assy = (
    Assembly(part1, loc=Location(Vector(-w,0,h/2)))
    .add(part2, loc=Location(Vector(1.5*w,-.5*d,h/2)), color=Color(0,0,1,0.5))
    .add(part3, loc=Location(Vector(-.5*w,-.5*d,2*h)), color=Color("red"))
)

exporters.export(assy.toCompound(), 'assy.stl')
```

[image]

zxo102, 2021-08-06:
@jmwright  Thanks a lot for your help.

zxo102, 2021-08-06:
@jmwright  How do you draw the grid on the XY plane with coordinates in the above output?

jmwright, 2021-08-06:
@zxo102 That's a screenshot from PrusaSlicer, which is not a CadQuery project. I just wanted to show that the exported STL file comes out correctly.

closed 2021-08-30 by adam-urbanczyk