CadQuery/cadquery #842: Create new object based off of selections of loft

r4dr3fr4d, 2021-08-14:
Trying to figure out more cadquery behavior - I have a loft, from which I'd like to add a box at its front-left corner.  I've tried multiple ways to do it but no luck - either the resulting box will be rotated a la a previous object (and therefore not parallel to the loft), or in the one case where I can get it to be parallel, it's positioned at a previous workplane.

I've tried creating the box off of different selections of the front-left corner, with and without calling workplane() - please see the comments.

I appreciate any help!

```
import cadquery as cq
from cadquery import CQ
from cadquery import exporters

s = cq.selectors.StringSyntaxSelector

box_length = box_width = 18
box_spacing = 2
box_thickness = 5

# helper functions to get the boxes' front/back faces and wires
cq.Workplane.front_face  = lambda self: self.faces(cq.NearestToPointSelector(self.plane.toWorldCoords((0,box_length/2,0)).toTuple()))

cq.Workplane.back_face   = lambda self: self.faces(cq.NearestToPointSelector(self.plane.toWorldCoords((0,-box_length/2,0)).toTuple()))

cq.Workplane.front_wires     = lambda self: self.front_face().wires().vals()[0]
cq.Workplane.back_wires      = lambda self: self.back_face().wires().vals()[0]

# create a box
box_wp = cq.Workplane("front")
box_1 = box_wp.box(box_length,box_width,box_thickness)

# create another box offset and tilted from the first box
box_wp = box_wp.transformed(offset=(0,box_length/2 + box_spacing, box_thickness/2), rotate=(5,0,0)) \
                .transformed(offset=(0,box_length/2, -box_thickness/2))
box_2 = box_wp.box(box_length,box_width,box_thickness)

# loft the boxes together
box_2.ctx.pendingWires = [box_1.front_wires(), box_2.back_wires()]
box_loft = box_2.workplane().loft(combine=False)

# flush the >Y face of the second box to the global Y direction
face_verts = box_2.faces('>Y').vertices().vals()
tups = sorted([v.toTuple() for v in face_verts])
tups[0:2] = tups[0:2][::-1]
tups = tups + [tups[0]]
face_wire = cq.Wire.makePolygon([cq.Vector(t) for t in tups])
new_tup = max([t[1] for t in tups]) + 1
flushed_tups = [(t[0], new_tup, t[2]) for t in tups]
vects = [cq.Vector(t) for t in flushed_tups]
flushed_wire = cq.Wire.makePolygon(vects)

box_2.ctx.pendingWires = [face_wire, flushed_wire]
flushed_face = box_2.workplane().loft(combine=False)

## and finally add another box to the >Y end of the this flushed face
#end_box = flush_face('>Y', extend=max_stagger-col_staggers[i]))

# show the intial stuff
show_object(box_1)
show_object(box_2)
show_object(box_loft)
show_object(flushed_face)

# now the problem/question:
# I want to add a box, parallel to the origin X/Y/Z planes, to the >Y<X (front-left) tip of this
# object
# issue is, I can't seem to make cadquery both select something at the front-left of the flushed_face
# *and* give me a box parallel to the origin

# I try making boxes off of the following selections by the front-left:
test_face = flushed_face.faces('<X')
test_edge = flushed_face.edges(s('|Z')-s('>X')-s('<Y'))
test_vert = test_edge.vertices('<Z')

# this one has the right position, but is rotated
test_box1 = test_face.rect(5,5).extrude(5, combine=False)

# this one isn't rotated, but it's back by box_2 for some reason?
test_box2 = test_face.workplane().rect(5,5).extrude(5, combine=False)

# *all* of these are rotated
test_box3 = test_edge.rect(5,5).extrude(5, combine=False)
test_box4 = test_edge.workplane().rect(5,5).extrude(5, combine=False)
test_box5 = test_vert.rect(5,5).extrude(5, combine=False)
test_box6 = test_vert.workplane().rect(5,5).extrude(5, combine=False)

GREEN = (100,200,50)
WHITE = (200,200,200)
GREY  = (50,50,50)

# show the test objects, moving some for visual ease
show_object(test_box1.translate((-3.5,0,0)), options={'color':GREEN})
show_object(test_box2.translate((-1,0,0)), options={'color':WHITE})
show_object(test_box6.translate((0,5,0)), options={'color':GREY})

```

adam-urbanczyk, 2021-08-14:
TBH I don't understand the question. Can you make a drawing indicating the target geometry?

r4dr3fr4d, 2021-08-14:
Sure, if I run this, I get the following:

[image]

[image]

The only line I added was to display the test_vert (moved upwards so you can see it):
`show_object(test_vert.translate((0,0,20)))`

I would like to create a box that originates at the test_vert, parallel to the face of the flushed_face (loft) that I created.  In other words, I want to work off of one part of that loft.  If you look at the test boxes I created, they are either rotated or positioned to a previous object in the stack (for example, the white one seems to be aligned with the box2 from which I created the loft).

Does that make sense?

lorenzncode, 2021-08-15:
Perhaps try the following change:
`
test_box1b = test_face.workplane(centerOption="CenterOfMass").rect(5,5).extrude(5, combine=False)
show_object(test_box1b, options={"color": "red"})
`
[image]

r4dr3fr4d, 2021-08-16:
@lorenzncode I appreciate the reply and showing me an option I haven't seen before, but I need the top of box to line up with the top of the flushed_face, and preferably, to have the same height/Z-width as the flushed_face.  While your suggestion looks closer than what I've been able to come up with so far, it seems to be positioning based off of some center-of-mass calculation instead of a part of another object (face/edge/vert I select).  You can see that in your picture, but I've zoomed in from the left side here to highlight it:

[image]

lorenzncode, 2021-08-16:
I think I follow now and BTW I'm still learning CadQuery myself.  We'll still use "CenterOfMass" - to locate the workplane at the vertex rather than face.  See the docs example "Locating a Workplane on a vertex".

> top of the box to line up with the top of the flushed_face

OK here is the flushed_face part in blue with two of the vertices in red.

[image]

```
vert1 = flushed_face.faces("<X").vertices(">Y").vertices(">Z")
vert2 = flushed_face.faces("<X").vertices(">Y").vertices("<Z")
vert1v = vert1.val().Center()
vert2v = vert2.val().Center()
dz = vert1v.z-vert2v.z
print(f"vert1 Center = {vert1v}")
print(f"vert2 Center = {vert2v}")
#dz is not exactly 5
print(f"dz = {dz}")
```

vert1 Center = Vector: (-9.0, 30.36728327938971, 4.068803369457847)
vert2 Center = Vector: (-9.0, 30.36728327938971, -0.9121701210008806)
dz = 4.980973490458728

> give me a box parallel to the origin

Let's then create the workplane and modify the zDir.

```
wp1 = vert1.workplane(centerOption="CenterOfMass")
print(wp1.plane.xDir)
print(wp1.plane.zDir)
```

Vector: (1.0, 0.0, 0.0)
Vector: (0.0, -0.08715574274765817, 0.9961946980917455)

```
wp1.plane.zDir = cq.Vector((0,0,1)).normalized()
test_box1c = wp1.workplane(centerOption="CenterOfMass").center(-2.5,0).rect(5,5).extrude(-dz, combine=False)
show_object(test_box1c, options={"color": "red"})
```

[image]

lorenzncode, 2021-08-16:
Of course, normalized() wasn't necessary there.

```vert1.plane.zDir = cq.Vector(0,0,1)```

r4dr3fr4d, 2021-08-16:
Nice!  So I guess the main thing I needed to do was manually modify the workplane normal? I'll definitely remember that for future stuff.  

Thank you @lorenzncode and @adam-urbanczyk !

closed 2021-08-16 by r4dr3fr4d

lorenzncode, 2021-08-16:
Yes I think so and perhaps another way to do it is to create a new Workplane on the named XY plane.

```
wp2rect = cq.Workplane("XY", vert1v).center(-2.5,0).rect(5,5)
show_object(wp2rect, options={"color": "red"})
```
[image]