CadQuery/cadquery #849: [Discussion] Fuzzy Boolean operations

Jojain, 2021-08-17:
Hello,
I randomly found this article on OCCT :
https://dev.opencascade.org/content/fuzzy-boolean-operations
That talks about fuzzy boolean operations. It says : 
> This option allows operators to handle robustly cases of touching and near-coincident, misaligned entities of the arguments.

While hanging out on the discord channel I have seen several times people (including me) having trouble with cut or fuse operations between shapes.

Last example in mind was the fusing of two tore of same radius at different angle. While it should fuse correctly it is possible that the standard fuse operation failed due to numeric rounding error of the two tore radius resulting in non-perfect overlapping of the radii.

I this case I think Fuzzy bool ops could come in handy. I cannot try it right now but I'm pretty sure it would solve the problem.

I have found one use in the code of `SetFuzzyValue` for fuse/union operation with the setting of `tol` parameter. But I think it could/should be exposed for cutting operations aswell.

In anycase if this type of operation improves greatly the rendering time + the quality of the results I think it could be interesting to give better explanation on what there are and how one should use them.
I feel that the `tol` argument could have better doc since it is not very clear for the newcomer of what is a Fuzzy operation.

``` python
def union(
        self: T,
        toUnion: Optional[Union["Workplane", Solid, Compound]] = None,
        clean: bool = True,
        glue: bool = False,
        tol: Optional[float] = None,
    ) -> T:
        """
        Unions all of the items on the stack of toUnion with the current solid.
        If there is no current solid, the items in toUnion are unioned together.

        :param toUnion:
        :type toUnion: a solid object, or a CQ object having a solid,
        :param boolean clean: call :py:meth:`clean` afterwards to have a clean shape (default True)
        :param boolean glue: use a faster gluing mode for non-overlapping shapes (default False)
        :param float tol: tolerance value for fuzzy bool operation mode (default None)
        :raises: ValueError if there is no solid to add to in the chain
        :return: a CQ object with the resulting object selected
        """
```

If the fuzzy operation always result in speed and quality gain when using rightfully it could be interesting to think of way of using it by default without need user input ?

adam-urbanczyk, 2021-08-17:
The `tol` should be added to cut too. I'm afraid, I have not seen yet this dramatic speed and quality improvements (in my issue cases at least). For the time being I'd consider this an "advanced" option.

aaroncaba, 2023-05-05:
Adam,

I have a use case where the `.cut()` method does not produce correct output without the fuzzy operation `tol=` parameter.  Adding the fuzzy `tol=` parameter creates correct results.

I have an analysis code that generates shape primitives with the union/cut operations in an XML format.  My current visualization method is to reading the XML in a C# program and driving SolidWorks to create sketches, extrudes, etc. based on the XML.  But ... interfacing with SolidWorks through the API is really slow and scales as O(N^3) with the number of operations.  

I can instead auto-generate a python script to drive CadQuery to generate the b-rep and export a STEP.  This is 2 or 3 orders of magnitude faster than the SolidWorks solution.  But after washing the geometry through all the different formats (XML -> C# -> python -> CadQuery -> OCCT) numerical inaccuracies are creeping in and preventing good unioning & cutting of very nearly tangent faces.

First I tried scaling my geometry up or down by 1,000X, but that didn't fix the tolerance issues.  Either the cut failed completely, the kernel crashed, or the result was twisted, nasty, incorrect geometry.

Then, I added the `tol` parameter to `.cut(tol=)` and was able to get good results.  Here's an example file with a cylinder tangent to a prism and `cut()` results with various tolerances.  The leftmost block shows the cylinder cut not doing anything, with the right most two blocks showing the expected result.  The middle blocks are different varieties of bad.

[image]

Here is minimal example script showing the output from my python generator.  Tons of round-off is visible in the output.  

```import cadquery as cq
from cadquery import (Vector, Workplane)

# Convenience wrappers for primitives that accept the properties we get from the other code

def myCylinder(p1: Vector, p2: Vector, r: float) -> Workplane:
    length = (p2 - p1).Length
    axis = p2 - p1
    return (cq.Workplane()
            .cylinder(length, r, centered=[True, True, False], direct=axis)
            .translate(p1))

# Points of a triangle in 3-D space, all on the same plane.  pt[0] === pt[3]
prismPoints = [Vector(11097.756407278355, 27264.360438303873, 390397.99999999994),
               Vector(5806.690426293661, 4777.330019118937, 390397.99999999994),
               Vector(-807.1420499372014, 7422.86300961128, 390397.99999999994),
               Vector(11097.756407278355, 27264.360438303873, 390397.99999999994)]

# workplane roated to same plane as the prismPoints
wp0 = cq.Workplane("XY").workplane() \
    .transformed(offset=Vector(0, 0, 0),
                 rotate=Vector(-179.99999645813307, -3.541866914610772E-06, -90.00000170754738))
localPts = []
for p in prismPoints:
    localPts.append(wp0.plane.toLocalCoords(Vector(p)))

# Extrude triangle into a prism
prism = (wp0.polyline(localPts).wire().extrude(446963.8, both=True))

# trim off one corner of the prism
cuttingFace = cq.Face.makePlane(
    None, None,
    Vector(5222.807074484849, 12797.653765894665, 390397.99999999994),
    Vector(0.37625884003871934, 0.9265145898973839, 0)
)
prism = prism \
    .split(cuttingFace, keepBottom=True) \
    .solids(cq.DirectionMinMaxSelector(Vector(0.37625884003871934, 0.9265145898973839, 0), False))

# draw a cylinder to cap the cut surface
cylinder = myCylinder(
    p1=Vector(5080.000000000001, 12446.000000000002, -56565.80000000004),
    p2=Vector(5080.000000000001, 12446.000000000002, 837361.7999999999),
    r=2463.7999999999997
)

try1 = prism.cut(cylinder)
try2 = prism.cut(cylinder,tol=.0001).translate(Vector(10000,0,0))
try3 = prism.cut(cylinder,tol=.001).translate(Vector(20000,0,0))
try4 = prism.cut(cylinder,tol=.01).translate(Vector(30000,0,0))
try5 = prism.cut(cylinder,tol=.1).translate(Vector(40000,0,0))
try6 = prism.cut(cylinder,tol=1).translate(Vector(50000,0,0))

all = try1.union(try2).union(try3).union(try4).union(try5).union(try6)
cq.exporters.export(all, 'cutcylinder.STEP')
```

aaroncaba, 2023-05-08:
Thanks Adam!  I had stared to setup a development environment and write a test for this, but your patch is more complete than mine would have been.

closed 2023-05-10 by adam-urbanczyk