CadQuery/cadquery #853: Hang in extrude call current master ocp 7.5.2

lorenzncode, 2021-08-20:
### To reproduce 

The process hangs (at least several minutes until manually kill process) when running the following:

```
import cadquery as cq
import math

radius1 = 45
radius2 = 95
total_width = 350
extrusion_depth = 20

# the center for both arcs, for improved readability
center_x_arc1 = radius1
center_x_arc2 = total_width - radius2

distance_centers = center_x_arc2 - center_x_arc1

# angle between the X axis and the tangential connector
alpha = math.asin((radius2 - radius1) / distance_centers)

# coordinates of the points on both arcs where the tangential line starts/ends
x1 = center_x_arc1 - radius1 * math.sin(alpha)
y1 = radius1 * math.cos(alpha)

x2 = center_x_arc2 - radius2 * math.sin(alpha)
y2 = radius2 * math.cos(alpha)

result = cq.Workplane("XY").radiusArc((x1, y1), radius1).lineTo(x2, y2).tangentArcPoint((total_width, 0), relative=False).mirrorX().extrude(extrusion_depth)
```

The above was posted to Google groups by Andreas Aug 17.  It must work for Andreas, and Roger, perhaps due to different versions.

@Jojain confirmed, with a fresh CadQuery install, is also able to reproduce the same behavior.

The issue seems to be with the final extrude() call in the chain.  Removing extrude() I can see the 2D shape in CQ-Editor.

### Environment

cadquery master
pytest output: 388 passed, 14 warnings

``` 
$ conda list | grep cadquery
cadquery                  2.1                       dev_0    <develop>
cq-editor                 master                    py3.9    cadquery
ocp                       7.5.2beta               1_py3.9    cadquery
sphinxcadquery            1.3.3                    pypi_0    pypi
```

$ uname -a
Linux fedora 5.13.9-200.fc34.x86_64 #1 SMP Sun Aug 8 14:34:00 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

### pstack

```
Thread 1 (Thread 0x7fb69d5c6740 (LWP 975585) "python"):
#0  0x00007fb686830e7b in Geom2d_TrimmedCurve::SetTrim(double, double, bool, bool) () from /home/lorenzn/tools/miniconda3/envs/cqdev/lib/python3.9/site-packages/../../libTKG2d.so.7
#1  0x00007fb6868311ec in Geom2d_TrimmedCurve::Geom2d_TrimmedCurve(opencascade::handle<Geom2d_Curve> const&, double, double, bool, bool) () from /home/lorenzn/tools/miniconda3/envs/cqdev/lib/python3.9/site-packages/../../libTKG2d.so.7
#2  0x00007fb6855e842c in ?? () from /home/lorenzn/tools/miniconda3/envs/cqdev/lib/python3.9/site-packages/../../libTKShHealing.so.7
#3  0x00007fb6855f4f4b in ShapeUpgrade_UnifySameDomain::IntUnifyFaces(TopoDS_Shape const&, NCollection_IndexedDataMap<TopoDS_Shape, NCollection_List<TopoDS_Shape>, TopTools_ShapeMapHasher>&) () from /home/lorenzn/tools/miniconda3/envs/cqdev/lib/python3.9/site-packages/../../libTKShHealing.so.7
#4  0x00007fb6855f7d68 in ShapeUpgrade_UnifySameDomain::UnifyFaces() () from /home/lorenzn/tools/miniconda3/envs/cqdev/lib/python3.9/site-packages/../../libTKShHealing.so.7
#5  0x00007fb6855fd711 in ShapeUpgrade_UnifySameDomain::Build() () from /home/lorenzn/tools/miniconda3/envs/cqdev/lib/python3.9/site-packages/../../libTKShHealing.so.7
#6  0x00007fb68b7753a3 in pybind11::cpp_function::initialize<pybind11::cpp_function::initialize<void, ShapeUpgrade_UnifySameDomain, , pybind11::name, pybind11::is_method, pybind11::sibling, char [53]>(void (ShapeUpgrade_UnifySameDomain::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [53])::{lambda(ShapeUpgrade_UnifySameDomain*)#1}, void, ShapeUpgrade_UnifySameDomain*, pybind11::name, pybind11::is_method, pybind11::sibling, char [53]>(pybind11::cpp_function::initialize<void, ShapeUpgrade_UnifySameDomain, , pybind11::name, pybind11::is_method, pybind11::sibling, char [53]>(void (ShapeUpgrade_UnifySameDomain::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [53])::{lambda(ShapeUpgrade_UnifySameDomain*)#1}&&, void (*)(ShapeUpgrade_UnifySameDomain*), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [53])::{lambda(pybind11::detail::function_call&)#3}::_FUN(pybind11::detail::function_call&) () from /home/lorenzn/tools/miniconda3/envs/cqdev/lib/python3.9/site-packages/OCP.cpython-39-x86_64-linux-gnu.so
#7  0x00007fb687ae8b95 in pybind11::cpp_function::dispatcher(_object*, _object*, _object*) () from /home/lorenzn/tools/miniconda3/envs/cqdev/lib/python3.9/site-packages/OCP.cpython-39-x86_64-linux-gnu.so
#8  0x00005645a5b9df85 in cfunction_call (func=0x7fb67cc970e0, args=<optimized out>, kwargs=<optimized out>) at /home/conda/feedstock_root/build_artifacts/python-split_1625973859697/work/Objects/methodobject.c:539
#9  0x00005645a5b84986 in _PyObject_MakeTpCall (tstate=0x5645a7dd37c0, callable=0x7fb67cc970e0, args=<optimized out>, nargs=<optimized out>, keywords=<optimized out>) at /home/conda/feedstock_root/build_artifacts/python-split_1625973859697/work/Objects/call.c:191
#10 0x00005645a5be025a in _PyObject_VectorcallTstate (kwnames=<optimized out>, nargsf=<optimized out>, args=0x7fb5efd893a0, callable=0x7fb67cc970e0, tstate=0x5645a7dd37c0) at /home/conda/feedstock_root/build_artifacts/python-split_1625973859697/work/Include/cpython/abstract.h:116
#11 _PyObject_VectorcallTstate (kwnames=<optimized out>, nargsf=<optimized out>, args=0x7fb5efd893a0, callable=0x7fb67cc970e0, tstate=0x5645a7dd37c0) at /home/conda/feedstock_root/build_artifacts/python-split_1625973859697/work/Include/cpython/abstract.h:103
[... 25 lines cut ...]
#37 _PyEval_EvalFrameDefault (tstate=<optimized out>, f=<optimized out>, throwflag=<optimized out>) at /home/conda/feedstock_root/build_artifacts/python-split_1625973859697/work/Python/ceval.c:3504
#38 0x00005645a5b77430 in _PyEval_EvalFrame (throwflag=0, f=0x7fb6726c8be0, tstate=0x5645a7dd37c0) at /home/conda/feedstock_root/build_artifacts/python-split_1625973859697/work/Include/internal/pycore_ceval.h:40
#39 _PyEval_EvalCode (tstate=0x5645a7dd37c0, _co=<optimized out>, globals=<optimized out>, locals=<optimized out>, args=<optimized out>, argcount=<optimized out>, kwnames=0x0, kwargs=0x5645a7e31900, kwcount=<optimized out>, kwstep=1, defs=0x7fb6730298c8, defcount=4, kwdefs=0x0, closure=0x0, name=0x7fb68fe05b70, qualname=0x7fb673024300) at /home/conda/feedstock_root/build_artifacts/python-split_1625973859697/work/Python/ceval.c:4327
#40 0x00005645a5bdf596 in _PyFunction_Vectorcall (func=<optimized out>, stack=0x5645a7e318f0, nargsf=<optimized out>, kwnames=<optimized out>) at /home/conda/feedstock_root/build_artifacts/python-split_1625973859697/work/Objects/call.c:396
#41 0x00005645a5c1e569 in _PyObject_VectorcallTstate (kwnames=0x0, nargsf=<optimized out>, args=0x5645a7e318f0, callable=0x7fb672df3430, tstate=0x5645a7dd37c0) at /home/conda/feedstock_root/build_artifacts/python-split_1625973859697/work/Include/cpython/abstract.h:118
#42 PyObject_Vectorcall (kwnames=0x0, nargsf=<optimized out>, args=0x5645a7e318f0, callable=0x7fb672df3430) at /home/conda/feedstock_root/build_artifacts/python-split_1625973859697/work/Include/cpython/abstract.h:127
#43 call_function (kwnames=0x0, oparg=<optimized out>, pp_stack=<synthetic pointer>, tstate=0x5645a7dd37c0) at /home/conda/feedstock_root/build_artifacts/python-split_1625973859697/work/Python/ceval.c:5072
#44 _PyEval_EvalFrameDefault (tstate=<optimized out>, f=<optimized out>, throwflag=<optimized out>) at /home/conda/feedstock_root/build_artifacts/python-split_1625973859697/work/Python/ceval.c:3504
#45 0x00005645a5b77430 in _PyEval_EvalFrame (throwflag=0, f=0x5645a7e31780, tstate=0x5645a7dd37c0) at /home/conda/feedstock_root/build_artifacts/python-split_1625973859697/work/Include/internal/pycore_ceval.h:40
#46 _PyEval_EvalCode (tstate=0x5645a7dd37c0, _co=<optimized out>, globals=<optimized out>, locals=<optimized out>, args=<optimized out>, argcount=<optimized out>, kwnames=0x0, kwargs=0x0, kwcount=<optimized out>, kwstep=2, defs=0x0, defcount=0, kwdefs=0x0, closure=0x0, name=0x0, qualname=0x0) at /home/conda/feedstock_root/build_artifacts/python-split_1625973859697/work/Python/ceval.c:4327
#47 0x00005645a5c5d517 in _PyEval_EvalCodeWithName (_co=<optimized out>, globals=<optimized out>, locals=<optimized out>, args=<optimized out>, argcount=<optimized out>, kwnames=<optimized out>, kwargs=0x0, kwcount=0, kwstep=2, defs=0x0, defcount=0, kwdefs=0x0, closure=0x0, name=0x0, qualname=0x0) at /home/conda/feedstock_root/build_artifacts/python-split_1625973859697/work/Python/ceval.c:4359
[... cut ...]
#57 Py_RunMain () at /home/conda/feedstock_root/build_artifacts/python-split_1625973859697/work/Modules/main.c:683
#58 0x00005645a5cd1599 in Py_BytesMain (argc=<optimized out>, argv=<optimized out>) at /home/conda/feedstock_root/build_artifacts/python-split_1625973859697/work/Modules/main.c:1129
#59 0x00007fb69d5f2b75 in __libc_start_main () from /lib64/libc.so.6
#60 0x00005645a5c4a6b1 in _start () at /home/conda/feedstock_root/build_artifacts/python-split_1625973859697/work/Objects/typeobject.c:2334
```

jmwright, 2021-08-22:
@lorenzncode We've had problems in the past with the mirror operations creating duplicate or extra edges that mess up subsequent 3D operations like extrude. It would be interesting to see if:

1. You have the expected number of edges before the extrude (and same number as with earlier versions of OCP/OCCT that work).
2. If it works without the `mirrorX()` call. That means you'll need to draw the other 2D side by reversing the operations of the original side. This will eliminate the mirror as the issue and will let us know if there's something strange about the geometry that extrude can't handle. For a very basic initial test you could just call `close()` and extrude the 2D half that you have already.

lorenzncode, 2021-08-22:
Thanks for the hints @jmwright!  Findings so far:

1. Yes I have the expected number of edges (3 before mirrorX).
2. Yes I confirm it works without the mirrorX() call.

   a. Initial test with `close()` to extrude the top half

    ```
    result = (
        cq.Workplane("XY")
        .radiusArc((x1, y1), radius1)
        .lineTo(x2, y2)
        .tangentArcPoint((total_width, 0), relative=False)
        .close()
        .extrude(extrusion_depth)
    )  # half, no hang
    ```

    b. Without mirrorX()

    ```
    result = (
        cq.Workplane("XY")
        .radiusArc((x1, y1), radius1)
        .lineTo(x2, y2)
        .tangentArcPoint((total_width, 0), relative=False)
    )  # top half
    result = (
        result.radiusArc((x2, -y2), radius2)
        .lineTo(x1, -y1)
        .tangentArcPoint((0, 0), relative=False)
    )  # bottom half
    result = result.consolidateWires().extrude(extrusion_depth) # no hang
    ```

lorenzncode, 2021-08-22:
Update - narrowed this down to the `extrude` call with `clean=True` (the default).  There is no hang when specifying `clean=False`.

```
result = (
    cq.Workplane("XY")
    .radiusArc((x1, y1), radius1)
    .lineTo(x2, y2)
    .tangentArcPoint((total_width, 0), relative=False)
    .mirrorX()
    .extrude(extrusion_depth, clean=False)
)  # no hang
```

lorenzncode, 2021-08-22:
I see there are existing open issues that are similar including #698.

jmwright, 2021-08-22:
Ah yes, I should have asked about that first. So clean=True in this situation fails on 7.5.2 but not 7.5.1?

jmwright, 2021-08-22:
I wonder if `mirrorX` is resulting in a profile that is not closed. Trying to extrude an unclosed profile can result in strange kernel behavior.

jmwright, 2021-08-22:
It might be worth checking the wire after `mirrorX` with `IsClosed`. https://github.com/CadQuery/cadquery/blob/4c45eb20e3576a2cff74f30c76bc96bf0f58561a/cadquery/occ_impl/shapes.py#L1347

lorenzncode, 2021-08-22:
> So clean=True in this situation fails on 7.5.2 but not 7.5.1?

I checked now and clean=True hangs with both 7.5.2 and 7.5.1.  New install with master and 7.5.1:

```
$ conda list | grep cadquery
cadquery                  2.1                       dev_0    <develop>
ocp                       7.5.1beta               1_py3.9    cadquery
sphinxcadquery            1.3.3                    pypi_0    pypi
```

> It might be worth checking the wire after mirrorX with IsClosed

I confirmed `IsClosed()` returns `True` here.

jmwright, 2021-08-22:
Ok, thanks for the investigation.

adam-urbanczyk, 2021-09-19:
It looks that it hangs in `ShapeUpgrade_UnifySameDomain` so for now you could try to using `clean=False`