CadQuery/cadquery #854: Distance between two shapes

safdark, 2021-08-21:
I see the following method stubs in *occ_impl/geom.py*:
```
    def distanceToLine(self):
        raise NotImplementedError("Have not needed this yet, but OCCT supports it!")

    ...

    def distanceToPlane(self):
        raise NotImplementedError("Have not needed this yet, but OCCT supports it!")
```

Any idea when these might be implemented?

Better yet, any chance we could have a generic method like below, abstracted on the Shape class, in **occ_impl/shapes.py**?
```
    def distanceToShape(self, other_shape: Shape) -> float:
       """
            Straight-line distance to other shape. Interpretation of the 'start' and 'end' points based on:
                * Edge/Wire: center
                * Face: center (or CoM if not planar)
                * Shell: CoM
                * Solid: CoM
                * Compound: CoM
       """
       ...
```

jmwright, 2021-08-23:
I think there is support for implementing these, but we have limited resources like most other open source projects. We do accept PRs from the community.

willhansen, 2022-06-05:
For reference, I think these are the relevant OCCT functions:

- [Distance from plane to point](https://old.opencascade.com/doc/occt-7.5.0/refman/html/classgp___pln.html#af043365c76bee31706cc3625de3d84a3)
- [Distance from line to point](https://old.opencascade.com/doc/occt-7.5.0/refman/html/classgp___lin.html#a4f0e6e4002ed7b364acd9b876bdebc2f)