CadQuery/cadquery #855: Get wire/edge position by absolute distance. (Convert Face to GeoJson)

a09418849, 2021-08-22:
I found that if I export faces to dxf, the end points wouldn't be conneted. 
Such like following :
[image]

So I try to convert the face element to GeoJson.
```PYTHON
import shapely
from shapely.geometry import *

import cadquery as cq
from cadquery.selectors import*

def wire_to_Polygon(wire, interval_distance):
    wire_length = wire.Length()
    if interval_distance<10e-6:
        interval_distance=10e-6
    elif interval_distance > wire_length/100:
        interval_distance = wire_length/100
    div = wire_length/interval_distance
    linspace = [1]+[ i/div for i in range(int(div)-1,0,-1)]+[0]
    vector_list = wire.positions(linspace,mode='length')
    linestring = LineString([ (vector.x,vector.y) for vector in vector_list])
    linestring_boundarys = linestring.buffer(interval_distance).buffer(-1*interval_distance*(0.99)).boundary
    if isinstance( linestring_boundarys,type(LineString()) ):
        if interval_distance == 10e-6:
            print('In wire_to_Polygon : Have the aspect ratio is too large. '+'Change interval_distance = '+str(interval_distance))
            linestring = linestring_boundarys
        else:
            print('In wire_to_Polygon : Have the aspect ratio is too large. '+'Change interval_distance = '+str(interval_distance/2))
            linestring = wire_to_Polygon(wire, interval_distance/2).boundary
    else:
        linestring = linestring_boundarys[0]
    return Polygon(linestring)

def face_to_polygon(face, interval_distance):
    # Only read X and Y coordinate.
    outerWire = face.outerWire()
    innerWires = face.innerWires()
    polygon = Polygon(wire_to_Polygon(outerWire, interval_distance))
    for innerWire in innerWires:
        polygon = polygon.difference(wire_to_Polygon(innerWire, interval_distance))
    return polygon

def cut_section_to_geojson(result, Plane, interval_distance):
    result_plane = cq.Workplane(Plane)
    result_plane.objects = result.vals()
    result_section = result_plane.section()
    section_faces = [result_section.plane.toLocalCoords(face_)  for object_ in result_section.objects for face_ in object_.Faces()]
    polygons = [ face_to_polygon(face,interval_distance) for face in section_faces]
    return MultiPolygon(polygons)
```
But I have a problem about the function `wire.positions()`. It use `wire.paramAt()` to get parameter value, and the `param d` in `wire.paramAt()` is normalized distance `[0, 1]` not absolute distance.
I think it should add absolute distance to get position. Like this:
```python
Mixin1D.positions([0, 0.2, 0.9, 1.2], mode = "absolute distance")
```
Besides this, you can try my function to get the closed and connected polygon with GeoJson.
```python
result0 = cq.Workplane("XY")
result0 = result0.moveTo(3,4).circle(1).extrude(10,both=True)
result0 = result0.spline([(0,0),(1,0),(1,1),(0,1)]).close().extrude(1,both=True).translate((-3,-4,0))
result0 = result0.moveTo(x,y).sphere(0.5)
result0 = result0.moveTo(4,3).circle(1).extrude(1,both=True)
result0 = result0.union( cq.Workplane("XY").
                        spline([(0,0),(-1,0),(-0.1,-0.5),(-1,-1),(0,-1)]).
                        close().
                        extrude(1,both=True).
                        translate((3,-4,0.5)).
                        rotate((0,0,0),(1,0,0),(5))
                       )
result0.plane = cq.Plane(origin=(0,0.1,0.5), normal=(0.05,0.05,1))
result0 = result0.section()

cut_section_to_geojson(result0,result0.plane,0.1)
```
[image]

And it would be easy to use ezdxf to convet GeoJson to dxf.

adam-urbanczyk, 2021-08-30:
Regarding DXF not being exported, reproducible example please? I don't see how GeoJson is related to CQ. Regarding position:
```
def positionAt(
        self, d: float, mode: Literal["length", "parameter"] = "length",
    ) -> Vector:
        ...
```

you can use `mode=length`.

a09418849, 2021-08-30:
Export section by dxf example : 
```python
import cadquery as cq

result0 = cq.Workplane("XY")
result0 = result0.moveTo(3,4).circle(1).extrude(10,both=True)
result0 = result0.spline([(0,0),(1,0),(1,1),(0,1)]).close().extrude(1,both=True).translate((-3,-4,0))
result0 = result0.moveTo(4,3).circle(1).extrude(1,both=True)
result0 = result0.union( cq.Workplane("XY").
                        spline([(0,0),(-1,0),(-0.1,-0.5),(-1,-1),(0,-1)]).
                        close().
                        extrude(1,both=True).
                        translate((3,-4,0.5)).
                        rotate((0,0,0),(1,0,0),(5))
                       )
result0.plane = cq.Plane(origin=(0,0.1,0.5), normal=(0.05,0.05,1))
result0 = result0.section()
cq.exporters.export(result0,"test.dxf", cq.exporters.ExportTypes.DXF)
```
[image]

I need to export technical drawing to manufacturer. The manufacturer usually use DXF to see the section or three-view drawing.
In my example, the exporter of dxf can not get the closed wire, so I used other way to get closed wire first. GeoJson can be converted to DXF by ezdxf easily.

In positionAt,  `mode == "length"` use paramAt, but paramAt noted that `Compute parameter value at the specified normalized distance`. I think maybe the `normalized distance` just a kind of simple linear scaling, so I can change `absolute distance` to `normalized distance`.
```python
    def positionAt(
        self: Mixin1DProtocol,
        d: float,
        mode: Literal["length", "parameter"] = "length",
    ) -> Vector:
        """Generate a position along the underlying curve.
        :param d: distance or parameter value
        :param mode: position calculation mode (default: length)
        :return: A Vector on the underlying curve located at the specified d value.
        """

        curve = self._geomAdaptor()

        if mode == "length":
            param = self.paramAt(d)
        else:
            param = d

        return Vector(curve.Value(param))
```
```python
    def paramAt(self: Mixin1DProtocol, d: float) -> float:
        """
        Compute parameter value at the specified normalized distance.

        :param d: normalized distance [0, 1]
        :return: parameter value
        """

        curve = self._geomAdaptor()

        l = GCPnts_AbscissaPoint.Length_s(curve)
        return GCPnts_AbscissaPoint(curve, l * d, 0).Parameter()
```