CadQuery/cadquery #873: Loft oddity (very specific offsets required between construction workplanes)

NothanUmber, 2021-08-31:
Ran into a quite odd behavior with loft.

What I do:
Building a longer shape by lofting a number of sketches (with ruled=True, want minimal lofts). One speciality is that every two workplanes usually have offset = 0, because the shape should jump from one size to another.
Usually this works well. E.g. here an example of what I mean:
[image]
which is the output of this program:
obj = cq.Workplane("XZ") \
        .rect(10, 10) \
        .workplane(offset=20) \
        .rect(10, 10) \
        .workplane() \
        .rect(20, 20) \
        .workplane(offset=20) \
        .rect(20, 20) \
        .workplane() \
        .circle(30, ) \
        .workplane(offset=20) \
        .circle(30) \
        .loft(ruled=True)

What happens:
Now for more complex problems offset 0 apparently isn't accepted anymore.
Have added three examples with increasing complexity and degree of odditiy:
[loft_anomaly.zip](https://github.com/CadQuery/cadquery/files/7083510/loft_anomaly.zip)
reproduce_loft_anomaly.py: First an example where all offsets can still be 0 - as expected

reproduce_loft_anomaly2.py: Here I add two new sketches. Now the offset between two previous sketches that could happily be 0 before has to be 0.0001
reproduce_loft_anomaly3.py: And when adding a whole bunch of other sketches, now one of the offsets has to be exactly 0.5 (neither 0.49 nor 0.51 work) and the one that previously had to be >= 0.0001 now has to be >= 0.001

Sorry that I couldn't simplify the examples more - then the effect disappeared.

NothanUmber, 2021-08-31:
P.S.: My initial solution was to loft the sketches pair-wise and then to union them (didn't lead to the desired result). I was able to add them all to the same stack via add and then return that - this worked. But then when I export it to STEP I get separate solids, not one. So putting all sketches into one big loft appeared the most natural next approach to me.

adam-urbanczyk, 2021-09-19:
TBH I don't think that the loft operation was designed to handle zero offsets between sketches.