CadQuery/cadquery #874: Thread example

mantielero, 2021-08-31:
I have used cadquery before, but  don't have much experience with it. I do it through FreeCAD workbench.

I would like to see an example about how to make threads. 

I have seen [this](https://github.com/dcowden/cadquery/issues/195) and [this](https://github.com/CadQuery/cadquery/issues/407). 

I think it would be useful to have an example within the CadQuery examples.

Thanks

dcowden, 2021-08-31:
Check out Roger's excellent cq_warehouse project, which contains fasteners:

https://github.com/gumyr/cq_warehouse/blob/main/src/cq_warehouse/fastener.py

I'm sure that would provide a very well tested example of threads.

adam-urbanczyk, 2021-08-31:
https://github.com/CadQuery/cadquery-contrib/blob/master/examples/Thread.py

mantielero, 2021-08-31:
The **fastener.py** example is pretty complex for me. Regarding the **Thread.py** example, it fails for me with:
```
Error executing CQGI-compliant script. 'Workplane' object has no attribute 'parametricCurve'
```

adam-urbanczyk, 2021-08-31:
Then try with the latest version of CQ:
https://github.com/CadQuery/cadquery/blob/6a94606b7f911192fae375d9dcaecd5b80e8d610/cadquery/cq.py#L1944

mantielero, 2021-09-01:
That solve the issue. In any case, I think it would be great to have that example in the documentation.

Thanks all for making CadQuery,

closed 2021-09-01 by mantielero

josyb, 2024-07-26:
> https://github.com/CadQuery/cadquery-contrib/blob/master/examples/Thread.py

If we change the pitch to 1.0 to get a more dense thread we get this result:
[image]

adam-urbanczyk, 2024-07-26:
It does work with some tweaking. What is your use case/what are your trying to get exactly?

[image]

josyb, 2024-07-26:
> It does work with some tweaking. What is your use case/what are your trying to get exactly?

Yes, that's true - tweaking is what we do all the time  when threads are involved, fiddle with the length, vary a radius slightly, move it up or down, ....
I have a list of metric and Whitworth threads in my library and just simply want to select out of these. 
I often have to tweak in other cases too, sometimes swapping operations, adding or subtracting  EPSILON (as in OpenSCad)
But most of the times, if not always, I get things working.

adam-urbanczyk, 2024-07-26:
I was hoping to see some code examples or repos. Anyhow, try this:

```python
from cadquery import *
from math import *

def helix(r0,r_eps,p,h,d=0,frac=1e-1):
    
    def func(t):
        
        if t>frac and t<1-frac:
            z = h*t + d
            r = r0+r_eps
        elif t<=frac:
            z = h*t + d*sin(pi/2 *t/frac)
            r = r0 + r_eps*sin(pi/2 *t/frac)
        else:
            z = h*t - d*sin(2*pi - pi/2*(1-t)/frac)
            r = r0 - r_eps*sin(2*pi - pi/2*(1-t)/frac)
            
        x = r*sin(-2*pi/(p/h)*t)
        y = r*cos(2*pi/(p/h)*t)
        
        return x,y,z
    
    return func

def thread(radius, pitch, height, d, radius_eps, aspect= 10,N = 100, deg =3, smooth =None):

    e1_bottom = (cq.Workplane("XY")
        .parametricCurve(helix(radius,0,pitch,height,-d), N=N, maxDeg=deg, smoothing=smooth).val()
    )
    e1_top = (cq.Workplane("XY")
        .parametricCurve(helix(radius,0,pitch,height,d), N=N, maxDeg=deg, smoothing=smooth).val()
    )
    
    e2_bottom = (cq.Workplane("XY")
        .parametricCurve(helix(radius,radius_eps,pitch,height,-d/aspect), N=N, maxDeg=deg, smoothing=smooth).val()
    )
    e2_top = (cq.Workplane("XY")
        .parametricCurve(helix(radius,radius_eps,pitch,height,d/aspect),N=N, maxDeg=deg, smoothing=smooth).val()
    )
    
    f1 = Face.makeRuledSurface(e1_bottom, e1_top)
    f2 = Face.makeRuledSurface(e2_bottom, e2_top)
    f3 = Face.makeRuledSurface(e1_bottom, e2_bottom)
    f4 = Face.makeRuledSurface(e1_top, e2_top)
    
    sh = Shell.makeShell([f1,f2,f3,f4])
    rv = Solid.makeSolid(sh)
    
    return rv

radius = 4
pitch = 0.25
height = 4
d = pitch/3
radius_eps = 0.15
eps=1e-3

core = cq.Workplane("XY",origin=(0,0,-d)).circle(4).circle(3).extrude(height+1.75*d)
th1 = thread(radius-eps,pitch,height,d,radius_eps)
th2  =thread(radius-1+eps,pitch,height,d,-radius_eps)

res = core.union(Compound.makeCompound([th1,th2]))

show_object(res)
```

josyb, 2024-07-27:
Hi Adam,

No repo - my code needs a lot of cleaning up first. Plus I don't think we need yet another's confusing code. I tend to do as I did in OpenScad; not sure that would be a good example.

I did use the code pointed to in your first reply above (at the top) to generate that faulty design.

I now plugged in the code you just supplied. That works fine and I can change the inputs to alter the thread's appearance.
Some observations: it runs so slow that at first  I suspected a hang-up. The result looks rather coarse.
[image]

Reverting some of the defaults back to, e.g. `N=400` and `deg=6`, sped things up again; with a plausible thread (plausible not perfect because the generated thread is not ISO or Whitworth). Resulting in a nicer smooth surface too.
[image]
Undoing the last change, `smoothing=None`, brings the *remarkable* result back. So all I have to do is to keep `smooth=None`
What I have to do now is to calculate your  arguments starting with the ISO and Whitworth specification.

Let's end with well deserved _praise_: I really like the smooth start and end of the thread.

Regards,
Josy

adam-urbanczyk, 2024-07-28:
I cannot reproduce your issue with `maxDeg=3` (see below). Are you using CQ or CQ-editor for visualization (based on the screenshot I don't think so)? If not, your are likely observing deltas in tessellation and not in the geometry.  NB: it is not really helpful to flag issue without a MRE.

[image]
[image]

adam-urbanczyk, 2024-07-28:
My bad, I can reproduce the issue after all. It seems that the approximation algorithm does not like to many points if the degree is specified as range.
[image]

AFAICT this seems to be the most robust set of parameters, one might need to adjust N based on the height.

```python
def thread(radius, pitch, height, d, radius_eps, aspect= 10,N = 100, deg =3, smooth =None, tol=1e-6):
    minDeg = deg
    e1_bottom = (cq.Workplane("XY")
        .parametricCurve(helix(radius,0,pitch,height,-d), N=N, minDeg=minDeg, maxDeg=deg, smoothing=smooth, tol=tol).val()
    )
    e1_top = (cq.Workplane("XY")
        .parametricCurve(helix(radius,0,pitch,height,d), N=N,  minDeg=minDeg, maxDeg=deg, smoothing=smooth, tol=tol).val()
    )
    
    e2_bottom = (cq.Workplane("XY")
        .parametricCurve(helix(radius,radius_eps,pitch,height,-d/aspect), N=N,  minDeg=minDeg, maxDeg=deg, smoothing=smooth, tol=tol).val()
    )
    e2_top = (cq.Workplane("XY")
        .parametricCurve(helix(radius,radius_eps,pitch,height,d/aspect),N=N,  minDeg=minDeg, maxDeg=deg, smoothing=smooth,tol=tol).val()
    )
    
    f1 = Face.makeRuledSurface(e1_bottom, e1_top)
    f2 = Face.makeRuledSurface(e2_bottom, e2_top)
    f3 = Face.makeRuledSurface(e1_bottom, e2_bottom)
    f4 = Face.makeRuledSurface(e1_top, e2_top)
    
    sh = Shell.makeShell([f1,f2,f3,f4])
    rv = Solid.makeSolid(sh)
    
    return rv
```

josyb, 2024-07-28:
> NB: it is not really helpful to flag issue without a MRE.

The MRE was your code, verbatim, I only changed parameters.

I use [CAD Assistant](https://www.opencascade.com/products/cad-assistant/) to view the generated `.step` output. CQ-Editor sometimes had/has issues displaying the design. As I need the `.step` for the 3D printer or the CNC machining I then can see what I get. I  also use another viewer [3D Tool](https://www.3d-tool.com) which has a few extra functions like cross-section  and to view assemblies.

Thanks for your work!
Regards,
Josy

lorenzncode, 2024-07-29:
Here is a metric M12 fine 1.5mm pitch screw example.  This is based on @adam-urbanczyk's thread function.

```py
from math import *

import cadquery as cq
from cadquery.occ_impl.shapes import *

def helix(r0, r_eps, p, h, d=0, frac=1e-1):
    def func(t):
        if t > frac and t < 1 - frac:
            z = h * t + d
            r = r0 + r_eps
        elif t <= frac:
            z = h * t + d * sin(pi / 2 * t / frac)
            r = r0 + r_eps * sin(pi / 2 * t / frac)
        else:
            z = h * t - d * sin(2 * pi - pi / 2 * (1 - t) / frac)
            r = r0 - r_eps * sin(2 * pi - pi / 2 * (1 - t) / frac)

        x = r * sin(-2 * pi / (p / h) * t)
        y = r * cos(2 * pi / (p / h) * t)

        return x, y, z

    return func

def thread(
    dmajor,
    pitch,
    height,
    radius_eps,
    flat_major,
    flat_minor,
    N_per_turn=40,
    deg=3,
    smooth=None,
    tol=1e-6,
):
    radius = dmajor / 2
    d1 = flat_major / 2
    d2 = pitch / 2 - flat_minor / 2

    taper = 1 / 2  # end taper at this fraction of turn approximately
    turns = height / pitch

    N = int(N_per_turn * turns)
    frac = (taper * (N / turns - 1) + 1) / N

    minDeg = deg
    e1_bottom = (
        cq.Workplane("XY")
        .parametricCurve(
            helix(radius, 0, pitch, height, -d1, frac),
            N=N,
            minDeg=minDeg,
            maxDeg=deg,
            smoothing=smooth,
            tol=tol,
        )
        .val()
    )
    e1_top = (
        cq.Workplane("XY")
        .parametricCurve(
            helix(radius, 0, pitch, height, d1, frac),
            N=N,
            minDeg=minDeg,
            maxDeg=deg,
            smoothing=smooth,
[... 158 lines cut ...]

core = (
    cq.Workplane("XY", origin=(0, 0, -eps))
    .circle(dminor / 2)
    .extrude(len_thread + 2 * eps)
)

th_with_core = core.union(th, clean=False, tol=eps)

th_with_chamfer = chamfer_thread(th_with_core, dmajor, pitch)

hex_head = make_screw_head(len_unthreaded)

screw = hex_head.translate((0, 0, len_screw - eps)).union(th_with_chamfer)

# cq.exporters.export(screw, "screw.stl")
show_object(screw)

```

My run time for screw length of 16mm was ~12s.  Screw length of 30mm took ~30s and 60mm ~2m10s.

CQ-editor render time is insignificant with appropriate tolerance settings.  I would recommend the values shown here or no smaller.

[image]

josyb, 2024-07-30:
And this is a 3D-printer M12 thread (0.2mm ColorFabb NGen)
[image]