CadQuery/cadquery #876: Possibility for more verbose error messages?

NothanUmber, 2021-09-01:
Frequently just getting "StdFail_NotDone: BRep_API: Command not done"
Well, I see that it didn't get done. Interesting question would be - why?
In many cases I can guess what might be wrong and fix it. But it is often a tedious process to find that out without any hints.
I guess this error message is coming out of OCCT? Is there a way that CadQuery could pass on more infos out of the CAD kernel what it doesn't like? Or is this all one can get out of the kernel?

jmwright, 2021-09-01:
> Or is this all one can get out of the kernel?

I believe that's all the info OCCT provides with that class of error. One of the other devs who knows OCCT better may know more.

parnold-x, 2021-09-01:
The why is difficult. But cadquery could report the context of the occ error. Like in fillet, extrude or something

NothanUmber, 2021-09-01:
Might be better than nothing. In most cases the function can be pinned down quite fast (e.g. fillet or extrude etc.) by taking in the functions one by one. But e.g. the fillet engine seems to be quite picky, in many cases it refuses to do anything at all.
Think better error messages (and more corner cases where e.g. fillets "just work") will be critical when people start to use cadquery for more complex things. One can of course e.g. leave all critical elements that easily throw errors (like fillets) out, export to STEP and then add those in a separate CAD program. But that would have to be redone after every change, then, depending on problem and iterations of such a process,  it might be more efficient to do the entire model in a GUI-CAD in the first place.

adam-urbanczyk, 2021-09-01:
I don't know how to get more error info from OCCT. As to locating the error, python trace-back tells exactly in which operation the error occurred.

NothanUmber, 2021-09-01:
Good point, yepp, no need to take things in one by one.
E.g. here one of the examples:
When taking my example from this thread: https://github.com/CadQuery/cadquery/issues/877
then this works for num_pipes from 1 to 4 (num_pipes in line 15). But changing the number of pipes from 4 to 5 then one gets the famous error:
Traceback (most recent call last):
  File "C:\polymoe\experiments\fillet_performance.py", line 285, in <module>
    result = main()
  File "C:\polymoe\experiments\fillet_performance.py", line 282, in main
    return body.build()
  File "C:\polymoe\experiments\fillet_performance.py", line 246, in build
    flute_body = flute_body.edges(cq.selectors.BoxSelector(
  File "C:\Users\strix\anaconda3\envs\cadquery\lib\site-packages\cadquery\cq.py", line 1277, in fillet
    s = solid.fillet(radius, edgeList)
  File "C:\Users\strix\anaconda3\envs\cadquery\lib\site-packages\cadquery\occ_impl\shapes.py", line 2232, in fillet
    return self.__class__(fillet_builder.Shape())
OCP.StdFail.StdFail_NotDone: BRep_API: command not done

The callstack tells exactly where this is, it's these lines:
# body front fillets
flute_body = flute_body.edges(cq.selectors.BoxSelector(
    (self.config.pipe_length - 2,
     -self.config.body_width_at_position(self.config.pipe_length - 2)/2 - 2,
     2),
    (self.config.pipe_length + 2,
     self.config.body_width_at_position(self.config.pipe_length - 2)/2 + 2,
     -self.config.body_height_at_position(self.config.pipe_length - 2) - 2)
)).fillet(2.5)
            
When taking those out it works. Or when changing the fillet radius from 2.5 to e.g. 1.5 it works, too.
Which I don't understand at all, because none of the widths or heights of the to-fillet elements (nor the faces in between) should have changed, when changing the num_pipes from 4 to 5 then there is simply one more repetition of a pattern that could be filletet before with radius 2.5 which doesn't work now anymore.
In many cases I made a mistake and finally found it when looking hard enough (even without a hint). Also wouldn't rule out that I made a mistake in this case (in doubt it's me is probably a good default assumption ;) ). But it's another case where I wonder whether I just don't get it or whether it's as random as it appears. Of course such situations can be "solved" via trial and error, but that doesn't lead to trustworthy and stable designs, I would prefer to understand why things fail when they do.

Here a picture for that might help to visualize why I think that the maximally possible fillet radius shouldn't depend on the number of pipes:
[image]

NothanUmber, 2021-09-03:
Played around with FreeCad. As it should use the same fillet engine under the hood, hope was that if it gives any explanations when trying the same fillet as in CadQuery, then those would likely also apply there. But FreeCad just writes an exclamation mark in front of the fillet and writes "Recompute failed! Please check report view" (in the report view...). So not better.
Then tried the same game in Solid Edge Community Edition. There it meant "Function couldn't be applied with this value". Well, slightly better but it also doesn't explain why this fillet cannot be computed. (As a human I could say how such a fillet would have to look like with a bigger radius than OCCT and Parasolid allow...)
So long story short - when even Parasolid doesn't get that done any better and also doesn't give any rationales why and there is no way to get more information out of OCCT, then I think one has to live it for now.
Think the issue can be closed.

closed 2021-09-03 by NothanUmber