CadQuery/cadquery #877: Fillets (or selection?) really slow (x20 compared to no fillets)

NothanUmber, 2021-09-01:
Here a part that takes ~2 seconds to render without fillets and ~40 seconds with.
Not sure whether it's the fillets or the selectors. (Beside the fillets I don't need any selectors at all).
Some "fillets" are created by tangentArcPoint in the sketches, these are super fast though (unnoticable difference between having these on or off)
In the attached file one can switch off the fillets by changing line 3 (ENABLE_FILLETS = False).
[fillet_performance.zip](https://github.com/CadQuery/cadquery/files/7094022/fillet_performance.zip)

Might be worth profiling what is going on, takes really a long time for such a simple model.

adam-urbanczyk, 2021-09-01:
So could you share the profiling results?

NothanUmber, 2021-09-01:
Ok, the majority seems to be in CQ-editor. When pressing Ctrl+S it takes 38 seconds until the model is visible on screen with fillets and ~2s without fillets.
But when I execute the program inside PyCharm (with cProfile.run) , then it takes ~6s with fillets and ~2s without fillets.

Profile text with fillets:
         1019547 function calls (964112 primitive calls) in 6.136 seconds

   Ordered by: internal time

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
        6    3.247    0.541    3.247    0.541 shapes.py:2218(fillet)
        6    1.242    0.207    1.242    0.207 shapes.py:946(_bool_op)
    56155    0.322    0.000    0.363    0.000 geom.py:67(__init__)
       12    0.268    0.022    0.268    0.022 shapes.py:375(clean)
        7    0.134    0.019    0.135    0.019 shapes.py:2573(makeLoft)
     5538    0.114    0.000    0.213    0.000 geom.py:706(_calcTransforms)
  31624/2    0.072    0.000    0.255    0.127 pyparsing.py:1647(_parseNoCache)
    16586    0.045    0.000    0.172    0.000 geom.py:157(normalized)
        6    0.039    0.006    1.443    0.240 cq.py:3195(union)
   206043    0.035    0.000    0.035    0.000 {built-in method builtins.isinstance}
     3424    0.031    0.000    0.086    0.000 shapes.py:638(centerOfMass)
     5524    0.029    0.000    0.664    0.000 geom.py:514(__init__)
    20542    0.024    0.000    0.027    0.000 geom.py:94(x)
      136    0.022    0.000    0.022    0.000 shapes.py:682(_entities)
    11076    0.019    0.000    0.019    0.000 geom.py:260(__init__)
   211076    0.019    0.000    0.019    0.000 {built-in method builtins.len}
    15742    0.019    0.000    0.039    0.000 pyparsing.py:554(__init__)
    17352    0.019    0.000    0.021    0.000 geom.py:102(y)
  8497/11    0.018    0.000    0.255    0.023 pyparsing.py:4049(parseImpl)
    16968    0.018    0.000    0.021    0.000 geom.py:110(z)
     5524    0.017    0.000    0.060    0.000 geom.py:129(cross)
     9009    0.017    0.000    0.025    0.000 pyparsing.py:602(__setitem__)
     3424    0.017    0.000    0.017    0.000 {built-in method OCP.BRepGProp.LinearProperties_s}
    16622    0.016    0.000    0.079    0.000 geom.py:126(toTuple)
   110255    0.016    0.000    0.016    0.000 geom.py:122(wrapped)
   5988/1    0.016    0.000    0.255    0.255 pyparsing.py:4249(parseImpl)
     5524    0.016    0.000    0.227    0.000 geom.py:700(_setPlaneDir)
     6485    0.014    0.000    0.014    0.000 shapes.py:331(shapetype)
    11415    0.014    0.000    0.016    0.000 geom.py:118(Length)
     3048    0.014    0.000    0.025    0.000 shapes.py:339(downcast)
     2472    0.012    0.000    0.039    0.000 pyparsing.py:852(__iadd__)
     5538    0.011    0.000    0.262    0.000 geom.py:565(origin)
      472    0.011    0.000    0.680    0.001 cq.py:177(__init__)
     4465    0.011    0.000    0.026    0.000 pyparsing.py:3339(parseImpl)
    13063    0.011    0.000    0.012    0.000 pyparsing.py:1628(preParse)
     1980    0.009    0.000    0.014    0.000 pyparsing.py:857(<listcomp>)
      840    0.009    0.000    0.019    0.000 geom.py:219(transform)
      458    0.009    0.000    0.669    0.001 geom.py:391(named)
     5564    0.008    0.000    0.008    0.000 {built-in method OCP.TopoDS.Edge_s}
     4464    0.007    0.000    0.011    0.000 pyparsing.py:2896(parseImpl)
    15742    0.006    0.000    0.010    0.000 pyparsing.py:545(__new__)
     6992    0.006    0.000    0.006    0.000 pyparsing.py:304(__init__)
     1624    0.006    0.000    0.006    0.000 geom.py:211(toPnt)
        6    0.005    0.001    3.402    0.567 cq.py:1251(fillet)
      130    0.005    0.000    0.032    0.000 shapes.py:731(<listcomp>)
   4249/1    0.005    0.000    0.255    0.255 pyparsing.py:4460(parseImpl)
      336    0.005    0.000    0.011    0.000 shapes.py:1641(makeLine)
    19916    0.004    0.000    0.004    0.000 pyparsing.py:495(__getitem__)
     4537    0.004    0.000    0.004    0.000 {method 'match' of 're.Pattern' objects}
     9009    0.004    0.000    0.004    0.000 pyparsing.py:493(__init__)
      770    0.003    0.000    0.012    0.000 shapes.py:1176(endPoint)
  1761/19    0.003    0.000    0.005    0.000 cq.py:738(_findType)
     1416    0.003    0.000    0.017    0.000 pyparsing.py:4847(parseImpl)
     3424    0.003    0.000    0.089    0.000 shapes.py:590(Center)
      896    0.003    0.000    0.003    0.000 shapes.py:1384(_geomAdaptor)
     2617    0.003    0.000    0.003    0.000 shapes.py:495(hashCode)
        4    0.003    0.001    0.059    0.015 selectors.py:136(filter)
     3036    0.003    0.000    0.027    0.000 shapes.py:368(__init__)
      457    0.002    0.000    0.687    0.002 cq.py:1346(newObject)
     4040    0.002    0.000    0.003    0.000 pyparsing.py:593(__getitem__)
       56    0.002    0.000    0.003    0.000 shapes.py:1626(makeTangentArc)
[... 321 lines cut ...]
       14    0.000    0.000    0.000    0.000 cq.py:2295(_addPendingWire)
        6    0.000    0.000    0.000    0.000 shapes.py:495(hashCode)
        6    0.000    0.000    0.000    0.000 shapes.py:770(<listcomp>)
        4    0.000    0.000    0.173    0.043 fillet_performance.py:130(_build_pipe_slow_airchamber_sketch)
       14    0.000    0.000    0.000    0.000 geom.py:151(__mul__)
        6    0.000    0.000    0.010    0.002 cq.py:972(solids)
       14    0.000    0.000    0.000    0.000 cq.py:2378(<listcomp>)
        6    0.000    0.000    0.000    0.000 fillet_performance.py:72(mouthpiece_width)
[... cut ...]
        6    0.000    0.000    0.000    0.000 cq.py:414(vals)
        2    0.000    0.000    0.000    0.000 fillet_performance.py:53(slow_airchamber_length)
        7    0.000    0.000    0.000    0.000 typing.py:1333(cast)
        1    0.000    0.000    0.000    0.000 {method 'disable' of '_lsprof.Profiler' objects}
        1    0.000    0.000    0.000    0.000 {method 'get' of 'mappingproxy' objects}
        1    0.000    0.000    0.000    0.000 fillet_performance.py:83(__init__)
        1    0.000    0.000    0.000    0.000 geom.py:161(Center)

NothanUmber, 2021-09-02:
Have repeated the experiment with cq-editor. The results aren't super precise, because I started cq-editor with profiler and the right file already loaded, clicked on render as soon as the UI was there and clicked on exit as soon as the rendering was finished.
I guess everything below viewer.py is mostly identical with cadquery.

Here the results:

total time in function (tottime, first few entries):

         3604193 function calls (3464228 primitive calls) in 43.824 seconds

   Ordered by: internal time

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
        2   27.875   13.937   27.876   13.938 viewer.py:161(display_many)
        6    3.255    0.542    3.255    0.542 shapes.py:2218(fillet)
        1    1.773    1.773   36.778   36.778 {built-in method exec_}
        6    1.273    0.212    1.274    0.212 shapes.py:946(_bool_op)
  119/114    1.244    0.010    1.251    0.011 {built-in method _imp.create_dynamic}
     43/1    0.994    0.023    1.000    1.000 diff_match_patch.py:257(diff_bisect)
    10113    0.619    0.000    0.619    0.000 {built-in method nt.stat}
     1810    0.508    0.000    0.508    0.000 {built-in method io.open_code}
        1    0.351    0.351    0.351    0.351 occt_widget.py:145(_initialize)
    56300    0.340    0.000    0.381    0.000 geom.py:67(__init__)
        1    0.325    0.325    0.325    0.325 {built-in method setDefaultTextOption}
       12    0.276    0.023    0.277    0.023 shapes.py:375(clean)
     1806    0.184    0.000    0.184    0.000 {built-in method marshal.loads}
       17    0.158    0.009    0.158    0.009 {function CodeEditor.paintEvent at 0x000002129C464820}
        9    0.157    0.017    0.168    0.019 {built-in method show}
 7799/716    0.151    0.000    0.407    0.001 sre_parse.py:493(_parse)
        7    0.133    0.019    0.133    0.019 shapes.py:2573(makeLoft)

cummulative time (first entries):

   2150/1    0.007    0.000   44.284   44.284 {built-in method builtins.exec}
        1    0.000    0.000   44.284   44.284 cq-editor-script.py:3(<module>)
        1    0.000    0.000   39.492   39.492 __main__.py:14(main)
        1    2.800    2.800   37.233   37.233 {built-in method exec_}
        1    0.000    0.000   33.674   33.674 debugger.py:222(render)
  347/224    0.002    0.000   27.490    0.123 {method 'emit' of 'PyQt5.QtCore.pyqtBoundSignal' objects}
        2   27.372   13.686   27.373   13.687 viewer.py:161(display_many)
        1    0.000    0.000   27.366   27.366 object_tree.py:228(addObjects)
        1    0.000    0.000    6.302    6.302 debugger.py:182(_exec)
        1    0.000    0.000    6.301    6.301 <string>:1(<module>)
        1    0.000    0.000    6.301    6.301 <string>:389(main)
        1    0.001    0.001    6.301    6.301 <string>:181(build)
   2057/6    0.016    0.000    4.826    0.804 <frozen importlib._bootstrap>:1002(_find_and_load)
   2055/6    0.010    0.000    4.826    0.804 <frozen importlib._bootstrap>:967(_find_and_load_unlocked)
   1956/6    0.011    0.000    4.821    0.804 <frozen importlib._bootstrap>:659(_load_unlocked)
   1806/6    0.007    0.000    4.821    0.803 <frozen importlib._bootstrap_external>:784(exec_module)
   2454/6    0.002    0.000    4.819    0.803 <frozen importlib._bootstrap>:220(_call_with_frames_removed)
        1    0.021    0.021    4.788    4.788 __main__.py:1(<module>)
        1    0.000    0.000    4.714    4.714 main_window.py:1(<module>)
      180    0.007    0.000    4.594    0.026 __init__.py:1(<module>)
        6    0.006    0.001    3.477    0.580 cq.py:1251(fillet)
       16    0.001    0.000    3.391    0.212 __init__.py:2(<module>)
        6    3.309    0.552    3.309    0.552 shapes.py:2218(fillet)
  744/187    0.002    0.000    3.197    0.017 {built-in method builtins.__import__}
        1    0.000    0.000    2.619    2.619 editor.py:1(<module>)
        1    0.000    0.000    2.319    2.319 codeeditor.py:7(<module>)
        1    0.000    0.000    2.089    2.089 main_window.py:28(__init__)
        6    0.038    0.006    1.492    0.249 cq.py:3195(union)
1799/1023    0.004    0.000    1.395    0.001 <frozen importlib._bootstrap>:1033(_handle_fromlist)
        6    0.000    0.000    1.285    0.214 shapes.py:3102(fuse)
        6    1.283    0.214    1.284    0.214 shapes.py:946(_bool_op)
1946/1937    0.005    0.000    1.283    0.001 <frozen importlib._bootstrap>:558(module_from_spec)
  119/114    0.001    0.000    1.214    0.011 <frozen importlib._bootstrap_external>:1106(create_module)
  119/114    1.205    0.010    1.213    0.011 {built-in method _imp.create_dynamic}
        2    0.000    0.000    1.012    0.506 codeeditor.py:2616(set_text)
        1    0.000    0.000    1.010    1.010 mixins.py:77(restoreComponentState)
        1    0.000    0.000    1.010    1.010 editor.py:229(restoreComponentState)
        1    0.000    0.000    1.010    1.010 editor.py:148(load_from_file)
        1    0.000    0.000    1.010    1.010 codeeditor.py:2626(set_text_from_file)
     1806    0.025    0.000    1.009    0.001 <frozen importlib._bootstrap_external>:856(get_code)
        4    0.000    0.000    1.002    0.250 decorators.py:18(wrapper)
        3    0.000    0.000    1.002    0.334 codeeditor.py:1241(document_did_change)
        2    0.000    0.000    1.001    0.501 mixins.py:678(set_eol_chars)
        3    0.000    0.000    1.001    0.334 diff_match_patch.py:1488(patch_make)
     87/3    0.000    0.000    1.000    0.333 diff_match_patch.py:78(diff_main)
     86/2    0.001    0.000    1.000    0.500 diff_match_patch.py:138(diff_compute)
     43/1    0.994    0.023    1.000    1.000 diff_match_patch.py:257(diff_bisect)
        1    0.000    0.000    0.968    0.968 main_window.py:76(prepare_panes)

NothanUmber, 2021-09-02:
And finally here the cq-editor times with fillets disabled (also with starting overhead etc.):

3082926 function calls (2996902 primitive calls) in 10.928 seconds

   Ordered by: internal time

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
        1    1.343    1.343    4.012    4.012 {built-in method exec_}
  119/114    1.212    0.010    1.220    0.011 {built-in method _imp.create_dynamic}
     40/1    0.994    0.025    1.000    1.000 diff_match_patch.py:257(diff_bisect)
        6    0.923    0.154    0.923    0.154 shapes.py:946(_bool_op)
    10107    0.604    0.000    0.604    0.000 {built-in method nt.stat}
     1810    0.488    0.000    0.488    0.000 {built-in method io.open_code}
        1    0.351    0.351    0.351    0.351 occt_widget.py:145(_initialize)
        1    0.318    0.318    0.318    0.318 {built-in method setDefaultTextOption}
    43771    0.256    0.000    0.286    0.000 geom.py:67(__init__)
     1806    0.170    0.000    0.170    0.000 {built-in method marshal.loads}
        9    0.169    0.019    0.180    0.020 {built-in method show}
 7799/716    0.146    0.000    0.399    0.001 sre_parse.py:493(_parse)
        6    0.125    0.021    0.125    0.021 shapes.py:375(clean)
        7    0.116    0.017    0.124    0.018 utils.py:54(dock)
        7    0.107    0.015    0.107    0.015 shapes.py:2573(makeLoft)
     4662    0.097    0.000    0.183    0.000 geom.py:706(_calcTransforms)
        1    0.096    0.096    0.099    0.099 base.py:49(__init__)
     1861    0.092    0.000    0.092    0.000 {method 'read' of '_io.BufferedReader' objects}
4262/4128    0.077    0.000    0.728    0.000 {built-in method builtins.__build_class__}
10224/533    0.075    0.000    0.178    0.000 sre_compile.py:71(_compile)
        2    0.073    0.036    0.081    0.040 viewer.py:161(display_many)
     1210    0.069    0.000    0.108    0.000 version.py:48(parse)
      574    0.068    0.000    0.068    0.000 {built-in method nt.listdir}
        7    0.068    0.010    0.068    0.010 {function CodeEditor.paintEvent at 0x00000188A9CA5820}
    89851    0.063    0.000    0.063    0.000 sre_parse.py:233(__next)
386626/385045    0.062    0.000    0.071    0.000 {built-in method builtins.isinstance}
     5555    0.058    0.000    0.058    0.000 {method 'search' of 're.Pattern' objects}
     1102    0.050    0.000    0.050    0.000 {built-in method builtins.compile}
   114857    0.043    0.000    0.062    0.000 sre_parse.py:164(__getitem__)
     3408    0.042    0.000    0.538    0.000 <frozen importlib._bootstrap_external>:1438(find_spec)
     1902    0.042    0.000    0.052    0.000 sre_compile.py:276(_optimize_charset)
    13960    0.039    0.000    0.150    0.000 geom.py:157(normalized)
      101    0.037    0.000    0.038    0.000 {built-in method io.open}

adam-urbanczyk, 2021-09-02:
So long story short, decrease the tessellation quality in CQ-editor. FYI: next time you can use `%prun` in the embedded console in CQ-editor.

closed 2021-09-02 by adam-urbanczyk

NothanUmber, 2021-09-02:
How do I change the tesselation quality? Can't find that in settings now manual?

jmwright, 2021-09-02:
> How do I change the tesselation quality? Can't find that in settings nor manual?

Edit->Preferences->3D Viewer

Check _Deviation_ and _Angular deviation_ settings. Try decreasing the _Deviation_ setting first.

NothanUmber, 2021-09-02:
Thanks! With higher Deviation values the rendering time goes down, but now the fillets aren't very round anymore.
So it might be best to disable fillets by default during modelling and only enable them ocassionally to see whether they still work - and of course for export.
And if fillets have to be tested/judged, one option could be to export to STEP from the CadQuery script and then to use e.g. MoI3d as viewer. There the CadQuery + visualization run still take 12s for this particular part with fillets enabled - still not "live-coding" realms. But probably ok when used just for verification from time to time.