CadQuery/cadquery #880: Segmentation fault during _bool_op in multiprocessing environment

ReubenVandezande, 2021-09-03:
While using cadquery in multiprocessing environments (ie.  multiprocessing and MPI), I've noted a significant number of segfaults during the `_bool_op` routine. I generate a number of geometries in my script and see segfaults by around 1,000-10,000 generated geometries. Looking at the backtrace, I believe this issue is caused by the hard-coded `op.SetRunParallel(True)` in the `_bool_op` method. 

I've been able to successfully generate ~1,000,000 geometries without segfault after manually setting this to `op.SetRunParallel(False)`.

## To Reproduce

I cannot reproduce the issue on command; this seems to be environment/geometry dependent and I can't post the full script here as it's quite large (modeling some complex systems). I can include a simplified version that communicates the idea of the offending script, but this example script doesn't segfault.

```
import cadquery as cq
from multiprocessing import Pool

def main():
    # Run the below function outside the pool to ensure it works correctly
    # generate()

    with Pool(5) as pool:
        while True:
            attempts = []
            for _ in range(1000):
                attempts.append(pool.apply_async(generate))
            
            [attempt.get() for attempt in attempts]

def generate():
    obj = (
        cq.Workplane('XZ')
        .rect(10, 10)
        .extrude(10))
    
    split = (
        cq.Workplane('YZ', obj.val().Center(), obj.val())
        .workplane()
        .split(keepBottom=True)
        .solids('<Y')
        .val())
    

if __name__ == '__main__':
    main()

```

## Backtrace

This is a (scrubbed) backtrace from my program.

```
Fatal Python error: Segmentation fault

Thread 0x00002ae45476fa00 (most recent call first):
  File ".conda/envs/example/lib/python3.8/site-packages/cadquery/occ_impl/shapes.py", line 892 in _bool_op
  File ".conda/envs/example/lib/python3.8/site-packages/cadquery/occ_impl/shapes.py", line 2788 in cut
  File ".conda/envs/example/lib/python3.8/site-packages/cadquery/cq.py", line 262 in split
  ...

```

## Environment

OS: Scientific Linux 7.8

```
# packages in environment at ******:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
alsa-lib                  1.2.3                h516909a_0    conda-forge
attrs                     21.2.0             pyhd3eb1b0_0  
[... 178 lines cut ...]
typish                    1.9.2              pyhd8ed1ab_0    conda-forge
urllib3                   1.26.6             pyhd8ed1ab_0    conda-forge
utfcpp                    3.2.1                ha770c72_0    conda-forge
vtk                       9.0.1           no_osmesa_py38h3850a3d_109    conda-forge
wcwidth                   0.2.5              pyh9f0ad1d_2    conda-forge
wheel                     0.36.2             pyhd3deb0d_0    conda-forge
x264                      1!161.3030           h7f98852_1    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libice               1.0.10               h7f98852_0    conda-forge
xorg-libsm                1.2.3             hd9c2040_1000    conda-forge
xorg-libx11               1.7.2                h7f98852_0    conda-forge
xorg-libxau               1.0.9                h7f98852_0    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h7f98852_1    conda-forge
xorg-libxt                1.2.1                h7f98852_2    conda-forge
xorg-xextproto            7.3.0             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.5                h516909a_1    conda-forge
yaml                      0.2.5                h516909a_0    conda-forge
zlib                      1.2.11            h516909a_1010    conda-forge
zstd                      1.5.0                ha95c52a_0    conda-forge

```

Using:

adam-urbanczyk, 2021-09-03:
Can you provide a backtrace (not python traceback) e.g. using `catchsegv` or your tooling of choice?

ReubenVandezande, 2021-09-03:
catcgsegv output:

```
*** stack smashing detected ***: python terminated
======= Backtrace: =========
/lib64/libc.so.6(__fortify_fail+0x37)[0x2adb308c7577]
/lib64/libc.so.6(+0x118532)[0x2adb308c7532]
.../.conda/envs/example/lib/python3.8/site-packages/mpi4py/MPI.cpython-38-x86_64-linux-gnu.so(+0xd3830)[0x2adb37327830]
.../.conda/envs/example/lib/python3.8/site-packages/mpi4py/MPI.cpython-38-x86_64-linux-gnu.so(+0xcc000)[0x2adb37320000]
[0x2adb37221f40]
======= Memory map: ========
2adb2f859000-2adb2f87b000 r-xp 00000000 fd:00 596140                     /usr/lib64/ld-2.17.so
2adb2f87b000-2adb2f87d000 rw-p 00000000 00:00 0 
2adb2f87d000-2adb2f884000 r--s 00000000 fd:00 789082                     /usr/lib64/gconv/gconv-modules.cache
2adb2f884000-2adb2f8c4000 rw-p 00000000 00:00 0 
2adb2f8c4000-2adb2f8c5000 r--p 00000000 00:2b 2244536000171595538        .../.conda/envs/example/lib/python3.8/lib-dynload/_heapq.cpython-38-x86_64-linux-gnu.so
2adb2f8c5000-2adb2f8c8000 r-xp 00001000 00:2b 2244536000171595538        .../.conda/envs/example/lib/python3.8/lib-dynload/_heapq.cpython-38-x86_64-linux-gnu.so
2adb2f8c8000-2adb2f8cb000 r--p 00004000 00:2b 2244536000171595538        .../.conda/envs/example/lib/python3.8/lib-dynload/_heapq.cpython-38-x86_64-linux-gnu.so
2adb2f8cb000-2adb2f8cc000 r--p 00006000 00:2b 2244536000171595538        .../.conda/envs/example/lib/python3.8/lib-dynload/_heapq.cpython-38-x86_64-linux-gnu.so
2adb2f8cc000-2adb2f8cd000 rw-p 00007000 00:2b 2244536000171595538        .../.conda/envs/example/lib/python3.8/lib-dynload/_heapq.cpython-38-x86_64-linux-gnu.so
2adb2f8cd000-2adb2f8ce000 r-xs 00000000 00:27 2414479323                 /tmp/ffi6zhDZJ (deleted)
2adb2f8ce000-2adb2f8cf000 rw-s 00000000 00:27 2414479323                 /tmp/ffi6zhDZJ (deleted)
2adb2f8cf000-2adb2f8d0000 rw-s 00000000 00:13 2414534686                 /dev/shm/sem.vTYlgf (deleted)
2adb2f8d0000-2adb2fa55000 rw-p 00000000 00:00 0 
2adb2fa55000-2adb2fa57000 r--p 00000000 00:2b 14643689548722307344       .../.conda/envs/example/lib/python3.8/lib-dynload/zlib.cpython-38-x86_64-linux-gnu.so
2adb2fa57000-2adb2fa5c000 r-xp 00002000 00:2b 14643689548722307344       .../.conda/envs/example/lib/python3.8/lib-dynload/zlib.cpython-38-x86_64-linux-gnu.so
2adb2fa5c000-2adb2fa5f000 r--p 00007000 00:2b 14643689548722307344       .../.conda/envs/example/lib/python3.8/lib-dynload/zlib.cpython-38-x86_64-linux-gnu.so
2adb2fa5f000-2adb2fa60000 r--p 00009000 00:2b 14643689548722307344       .../.conda/envs/example/lib/python3.8/lib-dynload/zlib.cpython-38-x86_64-linux-gnu.so
2adb2fa60000-2adb2fa61000 rw-p 0000a000 00:2b 14643689548722307344       .../.conda/envs/example/lib/python3.8/lib-dynload/zlib.cpython-38-x86_64-linux-gnu.so
2adb2fa61000-2adb2fa64000 r--p 00000000 00:2b 4528523718003062083        .../.conda/envs/example/lib/python3.8/lib-dynload/_bz2.cpython-38-x86_64-linux-gnu.so
2adb2fa64000-2adb2fa74000 r-xp 00003000 00:2b 4528523718003062083        .../.conda/envs/example/lib/python3.8/lib-dynload/_bz2.cpython-38-x86_64-linux-gnu.so
2adb2fa74000-2adb2fa77000 r--p 00013000 00:2b 4528523718003062083        .../.conda/envs/example/lib/python3.8/lib-dynload/_bz2.cpython-38-x86_64-linux-gnu.so
2adb2fa77000-2adb2fa78000 r--p 00015000 00:2b 4528523718003062083        .../.conda/envs/example/lib/python3.8/lib-dynload/_bz2.cpython-38-x86_64-linux-gnu.so
2adb2fa78000-2adb2fa7a000 rw-p 00016000 00:2b 4528523718003062083        .../.conda/envs/example/lib/python3.8/lib-dynload/_bz2.cpython-38-x86_64-linux-gnu.so
2adb2fa7a000-2adb2fa7b000 r--p 00021000 fd:00 596140                     /usr/lib64/ld-2.17.so
2adb2fa7b000-2adb2fa7c000 rw-p 00022000 fd:00 596140                     /usr/lib64/ld-2.17.so
2adb2fa7c000-2adb2fa7d000 rw-p 00000000 00:00 0 
2adb2fa7d000-2adb2fa81000 r-xp 00000000 fd:00 596144                     /usr/lib64/libSegFault.so
2adb2fa81000-2adb2fc80000 ---p 00004000 fd:00 596144                     /usr/lib64/libSegFault.so
2adb2fc80000-2adb2fc81000 r--p 00003000 fd:00 596144                     /usr/lib64/libSegFault.so
2adb2fc81000-2adb2fc82000 rw-p 00004000 fd:00 596144                     /usr/lib64/libSegFault.so
2adb2fc82000-2adb2fc99000 r-xp 00000000 fd:00 525338                     /usr/lib64/libpthread-2.17.so
2adb2fc99000-2adb2fe98000 ---p 00017000 fd:00 525338                     /usr/lib64/libpthread-2.17.so
2adb2fe98000-2adb2fe99000 r--p 00016000 fd:00 525338                     /usr/lib64/libpthread-2.17.so
2adb2fe99000-2adb2fe9a000 rw-p 00017000 fd:00 525338                     /usr/lib64/libpthread-2.17.so
2adb2fe9a000-2adb2fe9e000 rw-p 00000000 00:00 0 
2adb2fe9e000-2adb2fea0000 r-xp 00000000 fd:00 596150                     /usr/lib64/libdl-2.17.so
2adb2fea0000-2adb300a0000 ---p 00002000 fd:00 596150                     /usr/lib64/libdl-2.17.so
2adb300a0000-2adb300a1000 r--p 00002000 fd:00 596150                     /usr/lib64/libdl-2.17.so
2adb300a1000-2adb300a2000 rw-p 00003000 fd:00 596150                     /usr/lib64/libdl-2.17.so
2adb300a2000-2adb300a4000 r-xp 00000000 fd:00 596181                     /usr/lib64/libutil-2.17.so
2adb300a4000-2adb302a3000 ---p 00002000 fd:00 596181                     /usr/lib64/libutil-2.17.so
2adb302a3000-2adb302a4000 r--p 00001000 fd:00 596181                     /usr/lib64/libutil-2.17.so
2adb302a4000-2adb302a5000 rw-p 00002000 fd:00 596181                     /usr/lib64/libutil-2.17.so
[... cut ...]

Current thread 0x00002adb2f8d2f80 (most recent call first):
  File ".../.conda/envs/example/lib/python3.8/traceback.py", line 155 in _some_str
  File ".../.conda/envs/example/lib/python3.8/traceback.py", line 514 in __init__
  File ".../.conda/envs/example/lib/python3.8/traceback.py", line 103 in print_exception
  File ".../git_repos/PSO/exec/swarm.py", line 85 in runMPI
  File ".../git_repos/PSO/exec/swarm.py", line 68 in main
  File ".../git_repos/PSO/exec/swarm.py", line 94 in <module>

```

adam-urbanczyk, 2021-09-26:
Looking at the backtrace, the crash happens in the mpi4py module. Are you able to debug what is happening there? Is this problem occuring only with MPI (and actually what is the point of using MPI)?