CadQuery/cadquery #881: Accessing and modifying prior objects in the parent chain without losing the child context(s)

safdark, 2021-09-03:
Let's say I have an original 3D object called 'o'. I add/cut some 3D structures onto it (e.g, extrusions, holes etc) -- called 'e1', 'e2', etc -- to produce, in sequence, cumulative objects 'o1', 'o2', respectively.

Now, from 'o1', I want to be able to access 'e1' and 'o', and add modifications to them (to produce 'o2', 'o2' etc), without losing 'o1' in the process. In other words, I want to cumulatively add stuff to the model, with the freedom to navigate back to any previously added object or previously generated compound/solid.

To take a concrete example, say I create a box (workplane 'o') and then add 'e1' to extend it in the +Z direction to produce a workplane ('o1'). Let's also assume, without loss of generality, that the ">X" faces of these cumulative objects are all co-planar (i.e, overlapping). So, o1's face would be a vertical combination of o's face and e1's face.

Now, I'm looking for a way to be able to selectively locate a workplane() onto the center (i.e, "CenterOfBoundingBox/CenterOfMass/ProjectedOrigin") of the ">X" face of _each version_ of this object, **without needing to use the center()/translate() methods**. To illustrate this further, here's some simple code.

```
import cadquery as cq

o_l = 10
o_w = 20
o_h = 30

e_h = 40

body = (
    # o
    cq.Workplane("XY")
    .box(length=o_l, width=o_w, height=o_h)
    .tag("o")

    # +e1 ==> o1
    .faces(">Z")
    .workplane()
    .rect(xLen=o_l, yLen=o_w)
    .extrude(distance=e_h)

    # -------- Problem is here --------
    .faces(">X")
    .workplane(centerOption="CenterOfBoundBox")           # <------------- origin is at the center of the ">X" face of the cumulative solid 'o1'
    .sphere(radius=3)
)
show(body)
```

[image]

The centers ("CenterOfBoundingBox") for the ">X" face for the following objects would be at the following "heights" relative to the base:
- **o :**   *15*            ----- [15 = _o_h / 2_]
- **e1:** *50*            ---- [50 = _o_h + (e_h / 2)_]
- **o1:** *35*            ---- [35 = _(o_h + e_h) / 2_]

The call to workplane(centerOption="CenterOfBoundingBox") on o1.faces(">X") selects the origin at height 35 on that face, which is expected, but what API would I use to position the origin at the other centers [height = 15 or height = 35] starting from the 'o1' workplane?

I also tried using the tagged 'o', as follows:
```
    ...
    ...

    # -------- Problem is here --------
    .solids(tag='o')                                  # <-------------------- attempt to retrieve the context of the original object 'o'
    .faces(">X")
    .workplane(centerOption="CenterOfBoundBox")           # <------------- ... but I lose all the later context (of 'e1' and 'o1')
    .sphere(radius=3)
```

... But that takes me back to 'o' and I seem to lose 'e1' and 'o1'.
[image]

Alternatively, manually shifting the origin of faces(">X").workplane() by using the center() API seems more like a hack, since all the relevant information is already present in the stack and I shouldn't need to recalculate all that explicitly. Essentially, I'm looking for a way to access a previous parent's context while performing operations on the latest child in the hierarchy (cumulatively). How would I do this?

lorenzncode, 2021-09-04:
One method is to extrude with combine=False and combine only later once you are finished referencing the solids.

```
import cadquery as cq

o_l = 10
o_w = 20
o_h = 30
e1_h = 40
e2_r = 3

b0 = cq.Workplane("XY").box(length=o_l, width=o_w, height=o_h, centered=False)
b1 = (
    b0.faces(">Z")
    .workplane(centerOption="CenterOfMass")
    .rect(xLen=o_l, yLen=o_w)
    .extrude(distance=e1_h, combine=False)
)

s0 = b0.faces(">X").workplane(centerOption="CenterOfBoundBox").sphere(radius=e2_r)
s1 = b1.faces(">X").workplane(centerOption="CenterOfBoundBox").sphere(radius=e2_r)

body = s0.union(s1)

```

(added box arg centered=False to more closely match the height values mentioned - spheres at z 15, and 50)

safdark, 2021-09-05:
Many thanks for the working example, @lorenzncode .

Would have been great if there was a way to switch around the hierarchy more fluently (i.e, without forking), but I can't complain...This is the most convenient programmatic library I've found for my needs.

jmwright, 2021-09-07:
@safdark There might be some ideas you can pull from this assembly example. https://cadquery.readthedocs.io/en/latest/assy.html?highlight=tag#defining-reusable-components

safdark, 2021-09-26:
Thanks @jmwright .

closed 2021-09-26 by safdark