CadQuery/cadquery #883: Workplane.extrude side effect - Wire changed to Vector

gumyr, 2021-09-06:
**To Reproduce**

The Workplane.extrude method seems to have a side effect of changing the wire its operating on. See the following code:
```python
import cadquery as cq

outside_profile = (
    cq.Workplane("XY")
    .lineTo(0, 47)
    .spline(
        [(17, 65), (90, 20), (130, 5)],
        tangents=[(0, 1), (0, -1)],
        includeCurrent=True,
        scale=False,
    )
    .lineTo(130, 0)
    .close()
)
print(f"outside_profile is type {type(outside_profile.val())}")
print(f"outside_profile isValid: {outside_profile.val().isValid()}")
inside_profile = outside_profile.offset2D(-2)
print(f"inside_profile is type {type(inside_profile.val())}")
outside_core = outside_profile.extrude(25)
inside_profile = outside_profile.offset2D(-2)
print(f"inside_profile is type {type(inside_profile.val())}")
```
This outputs:
```
outside_profile is type <class 'cadquery.occ_impl.shapes.Wire'>
outside_profile isValid: True
inside_profile is type <class 'cadquery.occ_impl.shapes.Wire'>
inside_profile is type <class 'cadquery.occ_impl.geom.Vector'>
```
Note that the type of `inside_profile` changes from `Wire` prior to `.extrude` to `Vector` post `.extrude`.

**Environment**

OS: Ubuntu 21.04
cadquery @ master
Output of `conda list` from your active Conda environment:
```
# packages in environment at /home/roger/anaconda3/envs/cadquery-dev:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
alabaster                 0.7.12                     py_0    conda-forge
alsa-lib                  1.2.3                h516909a_0    conda-forge
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
argh                      0.26.2          pyh9f0ad1d_1002    conda-forge
astroid                   2.6.2            py39hf3d152e_0    conda-forge
async_generator           1.10                       py_0    conda-forge
atomicwrites              1.4.0              pyh9f0ad1d_0    conda-forge
attrs                     21.2.0             pyhd8ed1ab_0    conda-forge
autopep8                  1.5.6              pyhd8ed1ab_0    conda-forge
babel                     2.9.1              pyh44b312d_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
black                     19.10b0                    py_4    conda-forge
bleach                    3.3.0              pyh44b312d_0    conda-forge
brotlipy                  0.7.0           py39h3811e60_1001    conda-forge
build                     0.5.1                    pypi_0    pypi
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.17.1               h7f98852_1    conda-forge
ca-certificates           2021.5.30            ha878542_0    conda-forge
cadquery                  master                    py3.9    cadquery
certifi                   2021.5.30        py39hf3d152e_0    conda-forge
cffi                      1.14.5           py39he32792d_0    conda-forge
chardet                   4.0.0            py39hf3d152e_1    conda-forge
[... 245 lines cut ...]
watchdog                  2.1.3            py39hf3d152e_0    conda-forge
wcwidth                   0.2.5              pyh9f0ad1d_2    conda-forge
webencodings              0.5.1                      py_1    conda-forge
wheel                     0.36.2             pyhd3deb0d_0    conda-forge
wrapt                     1.12.1           py39h3811e60_3    conda-forge
wurlitzer                 2.1.0            py39hf3d152e_0    conda-forge
x264                      1!161.3030           h7f98852_0    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libice               1.0.10               h7f98852_0    conda-forge
xorg-libsm                1.2.3             hd9c2040_1000    conda-forge
xorg-libx11               1.7.2                h7f98852_0    conda-forge
xorg-libxau               1.0.9                h7f98852_0    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h7f98852_1    conda-forge
xorg-libxt                1.2.1                h7f98852_2    conda-forge
xorg-xextproto            7.3.0             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.5                h516909a_1    conda-forge
yaml                      0.2.5                h516909a_0    conda-forge
yapf                      0.31.0             pyhd8ed1ab_0    conda-forge
zeromq                    4.3.4                h9c3ff4c_0    conda-forge
zipp                      3.5.0              pyhd8ed1ab_0    conda-forge
zlib                      1.2.11            h516909a_1010    conda-forge
zstd                      1.5.0                ha95c52a_0    conda-forge
```

adam-urbanczyk, 2021-09-06:
What are you trying to convey by this example? Which wire is changed (yes, many methods have side effects)?

gumyr, 2021-09-06:
`outside_profile` appears to be changed from a `Wire` to `Vector` when using it in an extrude operation. This behaviour doesn't seem correct in that the order of unrelated operations dramatically impacts the operation of the code.

adam-urbanczyk, 2021-09-07:
`Extrude` consumes `pendingWires` so the second offset has nothing to act upon. No `Wire` is changed to `Vector` in the process. You can use `outside_profile.toPending().offset2D(-2)` to get the desired behavior.

gumyr, 2021-09-07:
Okay thanks I'll use `.toPending()`.  Is this intended behaviour?  I would think most users would expect idempotency here.

gumyr, 2021-09-07:
... and a thanks for helping out with these questions.  After having done many, many designs in cadquery I still don't have a good mental model for what is going on under the hood so use of methods like .toPending() are generally a mystery to me. I continue to learn though - with all of your help.
Cheers, Roger

adam-urbanczyk, 2021-09-07:
It is expected - operations like extrude "consume" pending wires/edges.

closed 2021-09-07 by adam-urbanczyk