CadQuery/cadquery #884: How to extrude in a given direction (not normal)?

RemDelaporteMathurin, 2021-09-13:
Hi all, I'm trying to extrude a 2D wire in a direction that isn't normal to the workplane. Obviously something is wrong with my usage of `extrudeLinear` - that I thought could be used the same way as `extrude`.

Could someone guide me to the proper way of doing this?

Cheers

```python
import cadquery as cq

pts = [
    (0, 0),
    (1, 0),
    (1, 1),
    (0, 1),
]

w = (
    cq.Workplane()
    .polyline(pts)
    .close()
    .extrudeLinear(......)
)

```

Jojain, 2021-09-13:
You could work with occ_impl layer and do something like : 
```python
import cadquery as cq

pts = [
    (0, 0),
    (1, 0),
    (1, 1),
    (0, 1),
]

wire = (
    cq.Workplane()
    .polyline(pts)
    .close()
    .val()
)
direction = cq.Vector(1,1,1).normalized() # extrudeLinear uses the magnitude of the vector as the extrusion length
length = 10
solid = cq.Solid.extrudeLinear(wire, [], direction*length)
#wrap it in a workplane again 
workplane = cq.Workplane(obj = solid)
``` 

You could also hack your way and change the workplane normal :  
```python 
import cadquery as cq

pts = [
    (0, 0),
    (1, 0),
    (1, 1),
    (0, 1),
]

w = (
    cq.Workplane()
    .polyline(pts)
    .close()
    )
w.plane.zDir = cq.Vector(1,1,1).normalized()
solid = w.extrude(10)
```
 I would not recommand the latter however.

RemDelaporteMathurin, 2021-09-13:
Oh thank you very much that's exactly what I was looking for!

bragostin, 2021-09-13:
How about
```python
import cadquery as cq

pts = [
    (0, 0),
    (1, 0),
    (1, 1),
    (0, 1),
]

w = (
    cq.Workplane()
    .polyline(pts)
    .close()
    )

solid = w.transformed((45,45,45)).extrude(10)
```

lorenzncode, 2024-06-09:
It's easy to do now with the new free function API:

```py
from cadquery.occ_impl.shapes import extrude, face, polyline

pts = [
    (0, 0),
    (1, 0),
    (1, 1),
    (0, 1),
]

solid = extrude(face(polyline(*pts).close()), (10, 10, 10))
```