CadQuery/cadquery #888: Bool ops/ add should merge tags on the Workplane level

adam-urbanczyk, 2021-09-17:


lorenzncode, 2021-11-22:
I started to look into this.  Here is a union example:

```
a = cq.Workplane("YZ").cylinder(30,5).faces(">X").workplane().tag("xface")

b = cq.Workplane("YZ", origin=(6,0,0)).sketch().circle(8).circle(5.1, mode="s").finalize().extrude(6)

c = b.union(a.end(2))
c = c.workplaneFromTagged("xface").sketch().arc((0,0),5,45,90).close().assemble().finalize().cutBlind(-10)
```

[image]

Without union or tags:

```
a = cq.Workplane("YZ").cylinder(30,5).faces(">X").workplane()
a = a.center(4+1/2,0).sketch().rect(10,1).finalize().cutBlind(-10)

s_collar = cq.Sketch().circle(8).circle(5.1, mode="s")

a = a.faces(">X[-2]").workplane(offset=1, origin=(0,0,0))
a = a.placeSketch(s_collar).extrude(6)
```

adam-urbanczyk, 2021-11-23:
What is the point you are trying to make?

lorenzncode, 2021-11-23:
I just thought I would share progress before making a pull request - mostly to make sure I was on the same page here with what was expected.  Since the union example that requires merged tags was a bit contrived, I thought I would compare with an example that works today without the merged tags.

adam-urbanczyk, 2021-11-24:
Thanks @lorenzncode , your first example works as expected.

closed 2021-12-24 by jmwright