CadQuery/cadquery #889: Lofting to an edge?

ricardo-re-azevedo, 2021-09-20:
Hi All, 
I am trying to join 3 solids (cubes for example), all with different rotations and positions is 3D. I can connect the faces easy enough but how can I fill in the center formed the the 3 cubes? See below a minimal example (red is the area I want to fill).

I was thinking the easiest would be to take the face of one of the connectors and loft it to the edge of the 3rd cube but as far as I know this is unsupported.

[image]
```python
import cadquery as cq

# add left cube
body = cq.Workplane('XY').box(1, 1, 1)
body.faces('>X').tag('a_right')
body.faces('>Y').tag('a_front')

# add right cube
body = body.workplane().transformed(offset=(2, .25, 1)).box(1, 1, 1)
body.faces('<X[1]').tag('b_left')
body.faces('>Y').tag('b_front')

# add front cube
body = body.workplane().transformed(offset=(-.25, 2, 1)).box(1, 1, 1)
body.faces('>>Y[-3]').tag('c_back')
body.faces('>X[-4]').tag('c_left')

# connect the three cubes
# connect left cube to front cube 
body.wires(tag=f'a_front').toPending()
body.wires(tag=f'c_left').toPending()
body = body.add(body.loft())

# connect left cube to right cube 
body.wires(tag=f'a_right').toPending()
body.wires(tag=f'b_left').toPending()
body = body.add(body.loft())

# connect right cube to front cube 
body.wires(tag=f'c_back').toPending()
body.wires(tag=f'b_front').toPending()
body = body.add(body.loft())

show_object(body, name='example')
```

Jojain, 2021-09-20:
You can't loft to an edge indeed. Cq Loft only takes wires as input, but the OCCT supports lofting from and to a vertex.

If your shape is aligned with the Z axis you can maybe take the +Z wire of the hole and loft to -Z it could work.

However it seems like a bad design. If we know what you want to do we could maybe give better modelling advice

ricardo-re-azevedo, 2021-09-21:
Thanks for the input. I am trying to prototype a keyboard similar to a dactyl [https://github.com/abstracthat/dactyl-manuform] with angled key positions. I can drop the key sockets just fine:
[image]

Next, I would like to connect all of these sockets to create the keyboard face. I have seen it done before is by breaking down they sockets into vertices, building a convex hull from those vertices, getting the faces from the hull, and converting them to a solid. I think its a smart way of doing it but it is a bit limiting since cuts in the surface will be removed. Is there a cleaner way to connect them all?

jmwright, 2021-09-24:
Maybe [interpPlate](https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=interpplate#cadquery.Workplane.interpPlate) would be useful here?

adam-urbanczyk, 2021-09-25:
`interPlate` could work here, but you'll be better off if you build the bottom surface first using `loft` or `approxSurface` and then add the holes. I assume here that you know how to construct e.g. an equation describing this surface. If not, you could just used the centers of your sockets.