CadQuery/cadquery #891: Installing cadquery on macOS on Apple silicon

jchidley, 2021-09-25:
Is there a way to install the current cadquery python library (the current *master*) on on an ARM based mac (currently M1)? 

It is possible to install both the CQ-editor (binary download, I think) and Jupyter-CadQuery (using Docker) but not the vanilla library, as far as I can tell. I believe that there are no osx-arm64 versions of the packages.

Alternatively, is there a manual way to install the required files, perhaps by compiling them? A quick look suggests that the documentation on this is pretty sparse and that the process will be complex.

jmwright, 2021-09-25:
It is a non-trivial task to build CadQuery's dependencies. You are correct that the documentation on this topic is pretty sparse. If you just want to use the library by itself, I have been doing manual assembly of an environment and then using LD_LIBRARY_PATH on MacOS and Linux to make sure the libs are loaded properly. As long as the underlying deps don't change too much, you can just pull the latest from CadQuery master to stay more current.

https://github.com/7BIndustries/Semblage/blob/master/.github/workflows/tests.yml#L34

jchidley, 2021-09-27:
I understood your words @jmwright but I have failed to get to grips with the whole host of libraries and your link to Semblange tests.

In the meantime, I have been taking a different route: using Docker and jupyter-cadquery.

Very roughly, you run jupyter-cadquery's docker image. Then add the Docker extension to Visual Studio before attaching to the jupyter-cadquery container. This installs the required VS Code files.

Once attached to the container, open a terminal, run `conda init` then `conda activate cq` before starting `jcv`. It'll open a browser window for the viewer.

Open your python file and then (potentially) open another terminal to run `conda activate cq` before chainging the python interpreter to ('cq': conda).

Finally, you can run the python code and see the results.

jchidley, 2021-09-27:
What I need to do, it turns out, is make another docker image.

tbo, 2021-12-20:
@jmwright Are there any plans to support Apple Silicon natively?

jmwright, 2021-12-20:
> Are there any plans to support Apple Silicon natively?

I don't think the core team is opposed to it, but I also don't think that any of us has an M1 to develop/test on. I expect we'll run into signing requirements eventually that will require a paid developer account as well. It feels like MacOS is getting more difficult for small open source projects to support as time goes on rather than easier.

michaelgale, 2021-12-21:
The key requirement is to build a native M1/arm64 [OCP](https://github.com/CadQuery/OCP) with [pywrap](https://github.com/CadQuery/pywrap).  This is not trivial--I did build a macOS version of OCP to run on Mojave (10.14) (since the bundled CQ package is built for 10.15+).  I do recall having to jump through many hoops to get a successful build--mainly related to C++ library dependencies and versions.  I can sympathize with the CQ team that maintaining multiple build pipelines for different platforms is quite a burden.  The peculiarities of the Apple toolchain ecosystem centred on Xcode doesn't help!

jchidley, 2022-02-12:
Building the docker images for AMD64 on Apple M1 does work. It's a bit slow and drains the battery because it's an emulation, and you have docker to deal with too.

 I'd prefer a native solution but CadQuery is worth it no matter how tricky the install is.

seanjensengrey, 2022-03-07:
I am on an M1 mac running on the [VMware Fusion Tech Preview for M1](https://customerconnect.vmware.com/downloads/get-download?downloadGroup=FUS-PUBTP-2021H1) the OS is `Ubuntu 20.04.3 LTS`.  Do CICD pipeline support doing arm64 (heh) builds? 

miniforge has arm64 builds, install fails at

```
(cadquery) arm@arm:~$ conda install -c conda-forge -c cadquery cadquery=master
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.

PackagesNotFoundError: The following packages are not available from current channels:

  - cadquery=master

Current channels:

  - https://conda.anaconda.org/conda-forge/linux-aarch64
  - https://conda.anaconda.org/conda-forge/noarch
  - https://conda.anaconda.org/cadquery/linux-aarch64
  - https://conda.anaconda.org/cadquery/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
```

jchidley, 2022-03-07:
I don't think that all of the software is available for arm64 machines, even virtual Linux ones. The only way that I could get this to work on my Mac was to use an AMD64 Docker image `--platform=linux/amd64`. It might be possible to do the same with your VMWare Fusion instead of Docker.

schluete, 2022-10-07:
@jmwright since I'm in the same boat as others asking for an M1 version: can you elude a little bit what exactly needs to be done for proper M1 support? After some (admittedly very little) research it seems like the primary problem is getting the [OCCT wrapper](https://github.com/CadQuery/OCP) compiled for M1. If that's the case, is it just that the setup is complicated, or are there closed-source components that would technically prevent an M1 version from being created? If it's the former I'd like to give it a shot.

smallstepman, 2022-10-07:
I'm no expert, but I think its the former. Checkout these two links to gain deeper insight
- https://github.com/CadQuery/OCP/issues/68 
- https://dev.opencascade.org/content/testing-occt-apple-m1-arm64

It would be a dream come true for me to see this closed.

jmwright, 2022-10-07:
I don't believe that any core developer has access to an M1 Mac for testing, but I think the bigger challenge is the lack of an Mac M1 runner in Azure Pipelines.

https://github.com/microsoft/azure-pipelines-agent/issues/3184

I don't think we are ready to do a self-hosted runner at this time.

https://itnext.io/run-azure-self-hosted-macos-agents-on-apple-m1-mac-72726555d83b

szymansk, 2023-01-13:
I am currently writing a Dockerfile for compiling OCP for arm64v8/ubuntu:22.04. Compilation and linking works, so far... The OCP.so seems to be functional. next step is to build the occ conda package... then building cadquery... when everything is done I am going to clean it up and push it to dockerhub... 
[image]

jmwright, 2023-01-14:
@szymansk Please report back here when you publish that.

szymansk, 2023-01-17:
I published arm64v8 linux docker image here:
https://hub.docker.com/r/szymanski2adesso/cadquery
Thing is... there is currently no view... didn't have time to take a glance ad the visualisation. But we are changing the cq-server to visualise tesselated objects via REST.

adam-urbanczyk, 2023-03-17:
arm osx is available via conda-forge

closed 2023-03-17 by adam-urbanczyk

ssbb, 2023-03-22:
@adam-urbanczyk it will work for stable builds only right? (no way to build from master?) 

Also cq-editor not here yet right?

adam-urbanczyk, 2023-03-22:
Yes, only releases. All the deps are available, so you can technically install it, but not out of the box.

n8henrie, 2023-03-30:
I've been interested in cadquery for a while but never delved in due to being on an M1 (so I've been watching this issue with interest).

Also never been a conda user (and don't really want it modifying my dotfiles). Being new to both cq and conda, I thought I'd share a few steps that got a working M1 jupyter environment for me:

```bash
#!/bin/bash

set -Eeuf -o pipefail
set -x

trap 'pkill -fl voila' EXIT

main() {
  . "/opt/homebrew/Caskroom/miniconda/base/etc/profile.d/conda.sh"

  conda deactivate || true
  conda env remove -n cadquery || true

  conda create \
    -y \
    -n cadquery \
    -c conda-forge \
    -c cadquery \
    python=3.11 \
    cadquery pip

  conda activate cadquery

  python -m pip install jupyter-cadquery cadquery-massembly matplotlib

  jcv &
  jupyter lab
}
main "$@"
```

This opens the jcv viewer as well as jupyter. If I paste into jupyter:

```
import cadquery as cq
from jupyter_cadquery.viewer.client import show

length = 80.0
height = 60.0
thickness = 100.0

result = cq.Workplane("XY").box(length, height, thickness)
show(result)
```

I can open the jcv viewer window and see the box rendered.

If I eventually figure out a plain pip (or nix) install I'll post it here, hope that's okay.

adam-urbanczyk, 2023-03-30:
How is this better than https://github.com/CadQuery/cadquery#conda-installation ? You don't need to run `conda init` and modify any dot files. You can also use something like [micromamba](https://mamba.readthedocs.io/en/latest/user_guide/micromamba.html)

n8henrie, 2023-03-30:
Not trying to argue it's better in any way, just new territory for me.

Knowing nothing about conda, I had no idea what miniforge, conda forge, or minimamba are. I do all my python development with the built-in venv module and pip and generally shy away from poetry, pipenv, conda, etc.

It indeed looks like miniforge supports darwin-aarch64, and I'm sure would be a fine way to do things.

ssbb, 2023-03-30:
Yeah miniforge works great on m1. To not mess with my env I just wrapped miniforge init code in function and call it when I need to work with cadquery

mosri, 2023-06-27:
> How is this better than https://github.com/CadQuery/cadquery#conda-installation ? You don't need to run `conda init` and modify any dot files. You can also use something like [micromamba](https://mamba.readthedocs.io/en/latest/user_guide/micromamba.html)

So does the installation method mentioned here now work on Mac ARM? Just gotta use conda?

adam-urbanczyk, 2023-06-27:
It should be possible, at least OCP is available for this platform:

osx-arm64 v7.7.0.1
linux-64 v7.7.0.1
win-64 v7.7.0.1
osx-64 v7.7.0.1

mosri, 2023-06-28:
Anything willing to help me troubleshooting installing cadquery an OS X in Docker? Been messing with different versions of Conda and python and can't get it to install cadquery.

[image]

jchidley, 2023-06-29:
Sailing at the moment so I can’t help, sorry

** Jack Chidley ***@***.*** 07764 359921 **

[Screenshot 2023-06-28 at 2 35 22 PM]<https://user-images.githubusercontent.com/8907920/249595373-9becab26-c617-4a6e-98f4-f32dec91c9d8.png>

mosri, 2023-07-01:
> try [szymansk/cadquery_arm64](https://github.com/szymansk/cadquery_arm64)

Did not work for me. I got this error:
[image]

Please someone help me figure this out. I've been trying to get this to work for a really long time. Thank you.

n8henrie, 2024-07-09:
Thanks to the work the cadquery team has done making wheels compatible with the M1, I was able to package up cadquery + OCP + jupyter for Apple Silicon into a nix flake.

If one has nix installed (and configured to use flakes), one can run JupyterLab with cadquery directly, without having to install anything else, with:

```console
$ nix run github:n8henrie/nix-cadquery
```

I'm sure the intersection of nix + Apple Silicon + cadquery users is fairly small, but thought this might help lower the barrier for some to try it out. There are still several failing tests (having trouble figuring out casadi + ipopt -- help appreciated), but it seems to work for all of https://github.com/CadQuery/cadquery/blob/master/examples/CQ%20examples.ipynb