CadQuery/cadquery #893: How to slice a hollow cylinder using a sinusoidal profile?

safdark, 2021-09-27:
Say I have  cylinder:

```
cylinder = (
   cq.Workplane("XY")
   .circle(radius=10)
   .circle(radius=7)
   .extrude(distance=30)
)
```

I want to define a method, such as:
```
def sine_slice(obj: cq.Workplane, at_height: float, sine_radius: float, start_angle_degrees: int, ...) -> cq.Workplane():
   ...
```
... that cleanly cuts the given cylinder along the sine curve? Additional parameters can be added, of course, as needed.

Something like this:
[image]

adam-urbanczyk, 2021-09-27:
You can use `parametricSurface` and `split`.

safdark, 2021-09-29:
Thanks @adam-urbanczyk . Will give it a try.

closed 2021-09-29 by safdark