CadQuery/cadquery #914: Example/Tutorial for "Parametrize STEP"

kfina91, 2021-11-05:
In README.md it is stated:
> Ability to import/export STEP and the ability to begin with a STEP model, created in a CAD package, and then add parametric features. This is possible in OpenSCAD using STL, but STL is a lossy format.

I searched [cadquery.readthedocs.io](https://cadquery.readthedocs.io/en/latest/search.html?q=step&check_keywords=yes&area=default) for "step" but could not find any guide on this.

[https://github.com/CadQuery/cadquery/issues/664](https://github.com/CadQuery/cadquery/issues/664)
I also found another related issue, where you suggest to do the model by hand, instead. I know that design intent may be lost if you start with a step file, but I would like to test the feature myself to see if we can save time but starting on the 3dl model.

My question is, if there are any examples/tutorials on this feature? From what I tried, cq.importers.importStep() only imports a solid, which I cannot interact with.

Jojain, 2021-11-06:
Hello, it's as simple as : 

```python
import cadquery as cq
part = cq.importers.importStep(my_step_path)
#part is a Workplane object so you can continue add features as you would normally

part = part.faces(">Z").hole(1)

```

kfina91, 2021-11-08:
Thanks for the answer.

Yes I'm running CadQuery master and I also got a workplane object.
Because of the feature description I got the impression, I magically get some parameters which I can fiddle around with, which seems not to be the case.

As reference for others, I'll write shortly what I did to parametrize my step file:
I wanted to enlengthen a part of my model.
```
# load the model from a file
model = cq.importers.importStep("model.step")
LENGTH = 0.25

# Split the model:
sides = model.faces(">X").workplane(-0.5).split(keepTop=True, keepBottom=True)

# Take left side and extrude the faces at one end
left = left.faces("<X").wires().toPending().extrude(-LENGTH)
right = sides.item(1)

# Reassemble your split parts together, with an offset for the right part
result = cq.Assembly()
result.add(left)
result.add(right, loc=cq.Location(cq.Vector(-LENGTH,0,0)))
```

closed 2021-11-08 by kfina91