CadQuery/cadquery #915: vtk - import requires unnecessary dependencies

darrellenns, 2021-11-05:
Currently, when cadquery imports vtk python modules it does something like:
`from vtk import vtkPolyData`

This actually causes vtk to load all vtk modules. In turn, this tries to load all of the shared libraries for every vtk module. If any of those are not present, it will cause cadquery to fail with an ImportError exception. For example:
```
>>> import cadquery
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.9/site-packages/cadquery/__init__.py", line 3, in <module>
    from .occ_impl.shapes import (
  File "/usr/lib/python3.9/site-packages/cadquery/occ_impl/shapes.py", line 20, in <module>
    from vtk import (
  File "/usr/lib/python3.9/site-packages/vtk.py", line 30, in <module>
    all_m = importlib.import_module('vtkmodules.all')
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/usr/lib/python3.9/site-packages/vtkmodules/all.py", line 84, in <module>
    from .vtkIOODBC import *
ImportError: libodbc.so.2: cannot open shared object file: No such file or directory
```

Instead, we could use the newer "vtkmodules". For example:
`from vtkmodules.vtkCommonDataModel import vtkPolyData`

With this syntax, only the necessary vtk modules are loaded and everything succeeds even if you don't have the shared libraries for other (unused) vtk functionality.

Currently, AUR packages for Arch Linux are affected by this issue. Unless you manually install all the optional dependencies for vtk (which shouldn't be necessary), cadquery will not run.

darrellenns, 2021-11-05:
See also: https://gitlab.kitware.com/vtk/vtk/-/issues/18365#note_1079271

adam-urbanczyk, 2021-11-07:
Thanks, good catch!

PR #918 (2021-11-08): use vtkmodules import instead of vtk
fixes #915