CadQuery/cadquery #925: AttributeError: type object 'OCP.BRepGProp.BRepGProp' has no attribute 'LinearProperties_s'

mikessut, 2021-11-29:
## To Reproduce

Basic install following the documentation.

```
conda create --name cadquery
conda activate cadquery
conda install -c conda-forge -c cadquery cadquery=2

python
import cadquery
```

```python
```

## Backtrace

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\users\xxx\Anaconda3\envs\cadquery\lib\site-packages\cadquery\__init__.py", line 3, in <module>
    from .occ_impl.shapes import (
  File "C:\users\xxx\Anaconda3\envs\cadquery\lib\site-packages\cadquery\occ_impl\shapes.py", line 234, in <module>
    ta.TopAbs_EDGE: BRepGProp.LinearProperties_s,
AttributeError: type object 'OCP.BRepGProp.BRepGProp' has no attribute 'LinearProperties_s'
```

## Environment

OS:

Windows

Was CadQuery installed using Conda?: 
Output of `conda list` from your active Conda environment:
```
```

Using:

mikessut, 2021-11-30:
Reverting to version 2 appears to work.

```
conda install -c conda-forge -c cadquery cadquery=2
```

adam-urbanczyk, 2021-11-30:
Does it work with `cadquery=master` and `ocp=7.5.2`?

evenprimes, 2022-01-03:
I'm trying to play around with cadquery on Windows and have the same error. I can't install `ocp=7.5.2`.

```
(cadquery) PS C:\Users\didge> conda install ocp=7.5.2
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.

PackagesNotFoundError: The following packages are not available from current channels:

  - ocp=7.5.2

Current channels:

  - https://repo.anaconda.com/pkgs/main/win-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/win-64
  - https://repo.anaconda.com/pkgs/r/noarch
  - https://repo.anaconda.com/pkgs/msys2/win-64
  - https://repo.anaconda.com/pkgs/msys2/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
```

This is a fresh install of Miniconda.

## Results of `conda list`
```
# Name                    Version                   Build  Channel
aiocontextvars            0.2.2                      py_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2021.10.8            h5b45459_0    conda-forge
cadquery                  master                    py3.6    cadquery
certifi                   2020.6.20          pyhd3eb1b0_3
colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
contextvars               2.4                        py_0    conda-forge
curl                      7.80.0               h789b8ee_1    conda-forge
decorator                 5.1.0              pyhd8ed1ab_0    conda-forge
double-conversion         3.1.7                h0e60522_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
expat                     2.4.2                h39d44d4_0    conda-forge
ezdxf                     0.14.2                     py_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.1            h1989441_1005    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeimage                 3.18.0               hfcb8cd7_7    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
gl2ps                     1.4.2                h0597ee9_0    conda-forge
glew                      2.1.0                h39d44d4_2    conda-forge
hdf4                      4.2.15               h0e5069d_3    conda-forge
hdf5                      1.10.6          nompi_he0bbb20_101    conda-forge
ilmbase                   2.5.5                h12d4b20_0    conda-forge
immutables                0.16             py36h68aa20f_0    conda-forge
intel-openmp              2022.0.0          h57928b3_3663    conda-forge
ipython                   5.8.0                    py36_1    conda-forge
ipython_genutils          0.2.0                      py_1    conda-forge
jbig                      2.1               h8d14728_2003    conda-forge
[... 51 lines cut ...]
simplegeneric             0.8.1                      py_1    conda-forge
six                       1.16.0             pyh6c4a22f_0    conda-forge
sqlite                    3.37.0               h8ffe710_0    conda-forge
tbb                       2020.2               h2d74725_4    conda-forge
tbb-devel                 2020.2               h2d74725_4    conda-forge
tk                        8.6.11               h8ffe710_1    conda-forge
traitlets                 4.3.3              pyhd8ed1ab_2    conda-forge
typing-extensions         4.0.1                hd8ed1ab_0    conda-forge
typing_extensions         4.0.1              pyha770c72_0    conda-forge
typish                    1.9.3              pyhd8ed1ab_0    conda-forge
ucrt                      10.0.20348.0         h57928b3_0    conda-forge
utfcpp                    3.2.1                h57928b3_0    conda-forge
vc                        14.2                 hb210afc_5    conda-forge
vs2015_runtime            14.29.30037          h902a5da_5    conda-forge
vtk                       9.0.1           no_osmesa_py36h2bf8032_107    conda-forge
wcwidth                   0.2.5              pyh9f0ad1d_2    conda-forge
wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
win32_setctime            1.0.4              pyhd8ed1ab_0    conda-forge
wincertstore              0.2             py36ha15d459_1006    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
zlib                      1.2.11            h8ffe710_1013    conda-forge
zstd                      1.5.1                h6255e5f_0    conda-forge
```

Python --version == Python 3.6.15

esotericbyte, 2022-01-12:
This is also a documentation bug since the directions here are broken: https://cadquery.readthedocs.io/en/latest/installation.html

adam-urbanczyk, 2022-01-12:
> This is also a documentation bug since the directions here are broken: https://cadquery.readthedocs.io/en/latest/installation.html conda install -c cadquery -c conda-forge cadquery=master fails conda install -c cadquery -c conda-forge cadquery=master ocp=7.5.2 works as expected.

Fails how? Was your first cadquery install? Or was the package taken from the local cache maybe?

esotericbyte, 2022-01-13:
> Fails how? Was your first cadquery install? Or was the package taken from the local cache maybe?
It was the first time. 

If you don't specify python version for some reason it's diving to 3.6 and then using that build of ocp.
The following worked:
conda create -n cadquery python=3.9
conda activate cadquery
conda install -c cadquery -c conda-forge cadquery=master

One clear set of instructions that work without discussion of alternatives and including cq-editor would best serve most users.

The "master" branch has been changed in many repositories to "main". 
See https://github.com/github/renaming

adam-urbanczyk, 2022-01-13:
I cannot reproduce the story with 3.6 being installed by default. Was it already there?

esotericbyte, 2022-01-16:
To see the behavior start with a fresh environment and without the python version specified and nothing else installed. 
conda create -n cadquery 
I believe you have something installed in the environment that is hoisting the python version. 
This is a condition that triggers the error in the  op and this case could be resolved by either by raising the requirements in meta.yaml or by fixing the installation instructions which would be more clearly defined anyway.