CadQuery/cadquery #927: I want to use the pipeline function of cadquery to generate a model, but I don't know how to do it.

luoyu-123, 2021-12-02:
I want to use the pipeline function of cadquery to generate a model, but I don't know how to do it. I am a beginner. I don't know where the pipeline function is, and there is no related function in the document. Please kindly ask for help, thank you very much.

The graphics are like this
[image]

gumyr, 2021-12-02:
The functionality you are looking for is called `sweep` - here is an example:
```python
import cadquery as cq

# Create the path of the pipe from edges consolidated into a wire and convert to wire class
pipe_path = cq.Workplane("XY").hLine(10).vLine(10).consolidateWires().val()

# Create the pipe profile aligned with the beginning of the pipe path
pipe_outline = cq.Face.makeFromWires(
    cq.Wire.makeCircle(1, pipe_path.positionAt(0), pipe_path.tangentAt(0))
)

# Create the pipe by sweeping the profile along the path with "right" transtitions
pipe = cq.Solid.sweep(pipe_outline, pipe_path, transitionMode="right")

if "show_object" in locals():
    show_object(pipe_path, name="pipe_path")
    show_object(pipe_outline, name="pipe_outline")
    show_object(pipe, name="pipe")

```
This code results in the following:
[image]

luoyu-123, 2021-12-03:
Thank you very much for your enthusiastic help, but I don’t seem to be able to run your example, the error is reported as shown below

[image]

gumyr, 2021-12-03:
There have been some significant changes to cadquery master recently which I suspect are causing this problem for you. If you aren't using a very recent version of master, please reinstall cadquery and try again. You should check out the definition of Solid.sweep() on your systems (an IDE will help here) and ensure the parameters are correct. I'm guessing that on your system you'll need to replace the pipe assignment with this:
```python
pipe = cq.Solid.sweep(pipe_outline, [], pipe_path, transitionMode="right")
```
where the empty list is the `innerWires` parameter.

jmwright, 2021-12-03:
@luoyu-123 I think you need to call sweep on the profile and pass it the path.

```python
pipe = pipe_outline.sweep(pipe_path, transitionMode="right")
```

adam-urbanczyk, 2021-12-03:
There are many ways to skin this cat. On the latest master:
```python

import cadquery as cq

path = cq.Workplane("XZ").polyline([(0,0,),(0,5),(5,5)])
pipe = cq.Workplane("XY").circle(1).sweep(path)

show_object(pipe)
```
[image]```

luoyu-123, 2021-12-04:
Thank you very much for your help, cadquery is a great open source project. Thank you so much
Problem has been solved

[image]

best wishes！

closed 2021-12-10 by adam-urbanczyk