CadQuery/cadquery #929: Extrusion in non-expected direction

smalltoe, 2021-12-12:
I want to extrude many objects in normal direction:

```python
import cadquery as cq

pts=[(0, 0, 0),(0, 0, 3)]
wp='YZ'
extr=10

w = cq.Workplane(wp)
block = w.rect(1,1).extrude(extr)
result = w.pushPoints(pts).eachpoint(lambda loc: blcok.val().located(loc))
result = result.combine()

show_object(result, options={"rgba": (204, 204, 204, 0.0)})
```

With specifying workplane YZ I would expect an extrusion in direction X, however this is not the case. Likewise ZX for Y.

[image]

lorenzncode, 2021-12-12:
Note that eachpoint passes a Location object into the callback function.  Location includes both a translation and rotation part.

You could make the block with Workplane("XY"):

```
import cadquery as cq

pts=[(0, 0, 0),(0, 0, 3)]
wp='YZ'
extr=10

w = cq.Workplane(wp)
block = cq.Workplane('XY').rect(1,1).extrude(extr)
result = w.pushPoints(pts).eachpoint(lambda loc: block.val().located(loc))
result = result.combine()

show_object(result, options={"rgba": (204, 204, 204, 0.0)})
```

lorenzncode, 2021-12-12:
Another way is to extract only the translation part of the Location and not apply rotation.

```
import cadquery as cq

pts=[(0, 0, 0),(0, 0, 3)]
wp='YZ'
extr=10

w = cq.Workplane(wp)
block = w.rect(1,1).extrude(extr)
result = w.pushPoints(pts).eachpoint(lambda loc: block.val().translate(cq.Vector(loc.toTuple()[0])))
result = result.combine()

show_object(result, options={"rgba": (204, 204, 204, 0.0)})
```

adam-urbanczyk, 2021-12-12:
BTW: you might want to try assemblies (`cq.Assembly`) and not `eachpoint` in your use case.

lorenzncode, 2021-12-12:
I'm sorry if `eachpoint` was bad advice here!  I think adding to the assembly would look something like this:

```
assy = cq.Assembly()

for p in pts:
    assy.add(block, loc=cq.Location(cq.Vector(p)))

show_object(assy, options={"rgba": (204, 204, 204, 0.0)})
```

smalltoe, 2021-12-12:
Guys - you are awsome. I am very happy to learn CadQuery with your input! I have adapted my program accordingly. Many Thanks!

closed 2021-12-13 by adam-urbanczyk