CadQuery/cadquery #933: FAILED tests/test_cqgi.py::TestCQGI::test_that_invalid_syntax_in_script_fails_immediately - AssertionError: False is not true

greyltc, 2021-12-22:
Hi, I'm building in Arch Linux today and I have one test failing when building the master branch, commit a6cbf22.
```
==> Starting check()...                                                                                               
================================================ test session starts =================================================
platform linux -- Python 3.10.1, pytest-6.2.5, py-1.11.0, pluggy-0.13.1                                               
rootdir: src/cadquery                                                
plugins: anyio-3.4.0, hypothesis-6.17.0, cov-2.12.1, forked-1.3.0                                                     
collected 439 items                                                                                                   
                                                                                                                      
tests/test_assembly.py .........................................                                               [  9%] 
tests/test_cad_objects.py ...............................                                                      [ 16%] 
tests/test_cadquery.py ....................................................................................... [ 36%] 
....................................................................................                           [ 55%] 
tests/test_cqgi.py ............F...                                                                            [ 58%] 
tests/test_examples.py ......................................................................................  [ 78%] 
tests/test_exporters.py ..........                                                                             [ 80%] 
tests/test_hull.py ..                                                                                          [ 81%] 
tests/test_importers.py ....                                                                                   [ 82%] 
tests/test_jupyter.py ..                                                                                       [ 82%] 
tests/test_selectors.py ...........................................                                            [ 92%] 
tests/test_sketch.py ...............                                                                           [ 95%] 
tests/test_workplanes.py ..................                                                                    [100%] 
                                                                                                                      
====================================================== FAILURES ======================================================
___________________________ TestCQGI.test_that_invalid_syntax_in_script_fails_immediately ____________________________
                                                                                                                      
self = <tests.test_cqgi.TestCQGI testMethod=test_that_invalid_syntax_in_script_fails_immediately>                     
                                                                                                                      
    def test_that_invalid_syntax_in_script_fails_immediately(self):                                                   
        badscript = textwrap.dedent(                                                                                  
            """                                                                                                       
                this doesn't even compile                                                                             
            """                                                                                                       
        )                                                                                                             
                                                                                                                      
        with self.assertRaises(Exception) as context:                                                                 
            model = cqgi.CQModel(badscript)                                                                           
                                                                                                                      
>       self.assertTrue("invalid syntax" in context.exception.args)                                                   
E       AssertionError: False is not true                                                                             
                                                                                                                      
tests/test_cqgi.py:119: AssertionError                                                                                

================================================== warnings summary ==================================================
../../../../../../../../usr/lib/python3.10/site-packages/packaging/version.py:127: 2208 warnings
  /usr/lib/python3.10/site-packages/packaging/version.py:127: DeprecationWarning: Creating a LegacyVersion has been deprecated and will be removed in the next major release
    warnings.warn(

tests/test_cad_objects.py::TestCadObjects::testCompoundCenter
tests/test_cadquery.py::TestCadQuery::testCombineSolidsInLoop
tests/test_cadquery.py::TestCadQuery::testCubePlugin
tests/test_cadquery.py::TestCadQuery::testCylinderPlugin
tests/test_cadquery.py::TestCadQuery::testDoubleTwistedLoft
  src/cadquery/cadquery/utils.py:34: FutureWarning: combineSolids will be removed in the next release.
    warn(f"{f.__name__} will be removed in the next release.", FutureWarning)

tests/test_cadquery.py::TestCadQuery::testCutBlindUntilFace
  src/cadquery/tests/test_cadquery.py:3333: DeprecationWarning: Please use assertAlmostEqual instead.
    self.assertAlmostEquals(wp_ref_regular_cut.val().Volume(), wp.val().Volume())

tests/test_cadquery.py::TestCadQuery::testExtrudeUntilFace
  src/cadquery/tests/test_cadquery.py:3158: DeprecationWarning: Please use assertAlmostEqual instead.
    self.assertAlmostEquals(wp_ref_extrude.val().Volume(), wp.val().Volume())

tests/test_cadquery.py::TestCadQuery::testFindFace
  src/cadquery/cadquery/utils.py:34: FutureWarning: findFace will be removed in the next release.
    warn(f"{f.__name__} will be removed in the next release.", FutureWarning)

tests/test_exporters.py::TestExporters::testAMF
tests/test_exporters.py::TestExporters::testSTEP
tests/test_exporters.py::TestExporters::testSTL
tests/test_exporters.py::TestExporters::testSVG
tests/test_exporters.py::TestExporters::testSVGOptions
tests/test_exporters.py::TestExporters::testTJS
  src/cadquery/cadquery/utils.py:34: FutureWarning: exportShape will be removed in the next release.
    warn(f"{f.__name__} will be removed in the next release.", FutureWarning)

tests/test_exporters.py::TestExporters::testSTEP
tests/test_exporters.py::TestExporters::testSTL
  src/cadquery/cadquery/utils.py:34: FutureWarning: readAndDeleteFile will be removed in the next release.
    warn(f"{f.__name__} will be removed in the next release.", FutureWarning)

-- Docs: https://docs.pytest.org/en/stable/warnings.html
============================================== short test summary info ===============================================
FAILED tests/test_cqgi.py::TestCQGI::test_that_invalid_syntax_in_script_fails_immediately - AssertionError: False i...
=================================== 1 failed, 438 passed, 2224 warnings in 44.16s ====================================
==> ERROR: A failure occurred in check().
    Aborting...
```

adam-urbanczyk, 2021-12-22:
Does it fail on 3.9 too?

greyltc, 2021-12-26:
Unfortunately testing that properly by rolling back the system python version is not so easy to do. Could be that the error message syntax just changed in 3.10 or something. I've built the project 1000 times without issue on 3.9 but, never commit a6cbf22.

greyltc, 2021-12-26:
No idea why this has started to fail now.

PR #937 (2021-12-28): let test_that_invalid_syntax_in_script_fails_immediately work in py3.10
fixes #933
commit c19e5350 (2021-12-28): let test_that_invalid_syntax_in_script_fails_immediately work in python 3.10

fixes #933
closed 2021-12-28 by jmwright