CadQuery/cadquery #940: how to split a cylinder?

mosheliv, 2021-12-27:
All examples of split I could find use a face as the base for the workplane. however, if I want to split a cylinder so i get two half cylinders, not two smaller cylinders, this is awkward to describe :) this approach doesn't work. I guess i could use cut with half cubes but wanted to know if there is a prettier approach.

jmwright, 2021-12-28:
You can construct an arbitrary plane and pass it to `split()`.

https://github.com/CadQuery/cadquery/issues/751#issuecomment-830007320

adam-urbanczyk, 2021-12-28:
You can also use `transformed` to rotate your current workplane.

mosheliv, 2021-12-29:
ok, i did try before to do that, and again now.

```
rad = 20
hrad = 5
h = 70
outer = cq.Workplane('XY').circle(rad).extrude(h).shell(-1.2)
split_wp = cq.Workplane('XZ')
half = outer.split(split_wp)
show(half, axes=True, grid=True, ortho=False, axes0=True)
```

I checked that the plane is actually correct (show(split_wp.circle(50))), but the cylinder is still cut in half in the wrong middle (I get the top half of the cylinder, not half cylinder)