CadQuery/cadquery #944: Segfault on invalid constraint based sketch

helmecke, 2022-01-02:
Python segfaults on invalid contraint based sketch. It chrashes at the `assemble` function.
## To Reproduce

```python
import cadquery as cq

r = (
    cq.Sketch()
    .segment((0.0, 0), (0.0, 2.0), "s1")
    .segment((2.0, 0))
    .close()
    .constrain("s1", "Length", 10)
    .solve()
    .assemble()
)

show_object(r)
```

## Backtrace

```
(cq) $ python 007.py      
zsh: segmentation fault (core dumped)  python 007.py
```

## Environment

OS: 5.15.12-arch1-1 GNU/Linux

Was CadQuery installed using Conda?: 
Output of `conda list` from your active Conda environment:
<details>
<summary>conda list</summary>

```
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
alabaster                 0.7.12                     py_0    conda-forge
alsa-lib                  1.2.3                h516909a_0    conda-forge
aom                       3.2.0                h9c3ff4c_2    conda-forge
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
argh                      0.26.2          pyh9f0ad1d_1002    conda-forge
astroid                   2.9.1            py39hf3d152e_0    conda-forge
async_generator           1.10                       py_0    conda-forge
atomicwrites              1.4.0              pyh9f0ad1d_0    conda-forge
attrs                     21.4.0             pyhd8ed1ab_0    conda-forge
autopep8                  1.5.6              pyhd8ed1ab_0    conda-forge
babel                     2.9.1              pyh44b312d_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
black                     19.10b0                    py_0  
bleach                    4.1.0              pyhd8ed1ab_0    conda-forge
brotlipy                  0.7.0           py39h3811e60_1003    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
ca-certificates           2021.10.8            ha878542_0    conda-forge
cadquery                  2.1                       dev_0    <develop>
certifi                   2021.10.8        py39hf3d152e_1    conda-forge
cffi                      1.15.0           py39h4bc2ebd_0    conda-forge
chardet                   4.0.0            py39hf3d152e_2    conda-forge
charset-normalizer        2.0.9              pyhd8ed1ab_0    conda-forge
click                     8.0.3            py39hf3d152e_1    conda-forge
cloudpickle               2.0.0              pyhd8ed1ab_0    conda-forge
colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
cq-editor                 master                    py3.9    cadquery
cryptography              36.0.1           py39h95dcef6_0    conda-forge
curl                      7.80.0               h2574ce0_0    conda-forge
[... 241 lines cut ...]
wrapt                     1.13.3           py39h3811e60_1    conda-forge
wurlitzer                 3.0.2            py39hf3d152e_1    conda-forge
x264                      1!161.3030           h7f98852_1    conda-forge
x265                      3.5                  h4bd325d_1    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libice               1.0.10               h7f98852_0    conda-forge
xorg-libsm                1.2.3             hd9c2040_1000    conda-forge
xorg-libx11               1.7.2                h7f98852_0    conda-forge
xorg-libxau               1.0.9                h7f98852_0    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h7f98852_1    conda-forge
xorg-libxt                1.2.1                h7f98852_2    conda-forge
xorg-xextproto            7.3.0             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.5                h516909a_1    conda-forge
yaml                      0.2.5                h516909a_0    conda-forge
yapf                      0.31.0             pyhd8ed1ab_0    conda-forge
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zipp                      3.6.0              pyhd8ed1ab_0    conda-forge
zlib                      1.2.11            h36c2ea0_1013    conda-forge
zstd                      1.5.1                ha95c52a_0    conda-forge
```
</details>

Using: Python interpreter

lorenzncode, 2022-01-02:
My understanding is that to use `solve()` you must fully constrain the sketch.  There are no implicit constraints added.

Segment s1 may end up as (0, -4), (0, 6) in this case.  The other two segments are no longer coincident with s1 and assemble then fails.

Adding more constraints to ensure assemble is successful:

```
import cadquery as cq

s = (
    cq.Sketch()
    .segment((0, 0), (0, 2.0), "s1")
    .segment((0, 2.0), (2.0, 0), "s2")
    .segment((2.0, 0), (0, 0), "s3")
    .constrain("s1", "Length", 10)
    .constrain("s1", "s2", "Coincident", None)
    .constrain("s2", "s3", "Coincident", None)
    .constrain("s3", "s1", "Coincident", None)
    .solve()
    .assemble()
)

r = cq.Workplane().placeSketch(s).extrude(1)
```

helmecke, 2022-01-02:
> My understanding is that to use `solve()` you must fully constrain the sketch. There are no implicit constraints added.

That is my understanding as well but every time I miss to fully constrain my sketch `cq-editor` closes which is often :grin: .
An error message like _not fully constrained_ would help.

adam-urbanczyk, 2022-01-02:
Assemble should not segfault. That has nothing to do with solve being fully or partially constrained. Could you paste a real backtrace?

lorenzncode, 2022-01-02:
Seems to be in Face.makeFromWires

```
#0  0x00007ff6ad0a90d4 in TopoDS_Builder::Add(TopoDS_Shape&, TopoDS_Shape const&) const () from /home/lorenzn/tools/miniforge/envs/cqdev/lib/python3.9/site-packages/../../libTKBRep.so.7
#1  0x00007ff6ac90e464 in BRepLib_MakeFace::Add(TopoDS_Wire const&) () from /home/lorenzn/tools/miniforge/envs/cqdev/lib/python3.9/site-packages/../../libTKTopAlgo.so.7
#2  0x00007ff6ac955e24 in BRepBuilderAPI_MakeFace::Add(TopoDS_Wire const&) () from /home/lorenzn/tools/miniforge/envs/cqdev/lib/python3.9/site-packages/../../libTKTopAlgo.so.7
#3  0x00007ff6af32e800 in pybind11::cpp_function::initialize<pybind11::cpp_function::initialize<void, BRepBuilderAPI_MakeFace, TopoDS_Wire const&, pybind11::name, pybind11::is_method, pybind11::sibling, char [349], pybind11::arg>(void (BRepBuilderAPI_MakeFace::*)(TopoDS_Wire const&), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [349], pybind11::arg const&)::{lambda(BRepBuilderAPI_MakeFace*, TopoDS_Wire const&)#1}, void, BRepBuilderAPI_MakeFace*, TopoDS_Wire const&, pybind11::name, pybind11::is_method, pybind11::sibling, char [349], pybind11::arg>(pybind11::cpp_function::initialize<void, BRepBuilderAPI_MakeFace, TopoDS_Wire const&, pybind11::name, pybind11::is_method, pybind11::sibling, char [349], pybind11::arg>(void (BRepBuilderAPI_MakeFace::*)(TopoDS_Wire const&), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [349], pybind11::arg const&)::{lambda(BRepBuilderAPI_MakeFace*, TopoDS_Wire const&)#1}&&, void (*)(BRepBuilderAPI_MakeFace*, TopoDS_Wire const&), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [349], pybind11::arg const&)::{lambda(pybind11::detail::function_call&)#3}::_FUN(pybind11::detail::function_call&) () from /home/lorenzn/tools/miniforge/envs/cqdev/lib/python3.9/site-packages/OCP.cpython-39-x86_64-linux-gnu.so
#4  0x00007ff6aeb310e5 in pybind11::cpp_function::dispatcher(_object*, _object*, _object*) () from /home/lorenzn/tools/miniforge/envs/cqdev/lib/python3.9/site-packages/OCP.cpython-39-x86_64-linux-gnu.so
#5  0x00005617c9a58995 in cfunction_call (func=0x7ff6a4657810, args=<optimized out>, kwargs=<optimized out>) at /home/conda/feedstock_root/build_artifacts/python-split_1632942464411/work/Objects/methodobject.c:543
#6  0x00005617c9a3f3d6 in _PyObject_MakeTpCall (tstate=0x5617cb0028b0, callable=0x7ff6a4657810, args=<optimized out>, nargs=<optimized out>, keywords=<optimized out>) at /home/conda/feedstock_root/build_artifacts/python-split_1632942464411/work/Objects/call.c:191
#7  0x00005617c9a99b4a in _PyObject_VectorcallTstate (kwnames=<optimized out>, nargsf=<optimized out>, args=0x7ff61dedd7d8, callable=0x7ff6a4657810, tstate=0x5617cb0028b0) at /home/conda/feedstock_root/build_artifacts/python-split_1632942464411/work/Include/cpython/abstract.h:116
#8  _PyObject_VectorcallTstate (kwnames=<optimized out>, nargsf=<optimized out>, args=0x7ff61dedd7d8, callable=0x7ff6a4657810, tstate=0x5617cb0028b0) at /home/conda/feedstock_root/build_artifacts/python-split_1632942464411/work/Include/cpython/abstract.h:103
#9  method_vectorcall (method=<optimized out>, args=0x7ff61dedd7e0, nargsf=<optimized out>, kwnames=<optimized out>) at /home/conda/feedstock_root/build_artifacts/python-split_1632942464411/work/Objects/classobject.c:53
#10 0x00005617c9adcc6b in _PyObject_VectorcallTstate (kwnames=0x0, nargsf=<optimized out>, args=0x7ff61dedd7e0, callable=0x7ff61ded6f00, tstate=0x5617cb0028b0) at /home/conda/feedstock_root/build_artifacts/python-split_1632942464411/work/Include/cpython/abstract.h:118
#11 PyObject_Vectorcall (kwnames=0x0, nargsf=<optimized out>, args=0x7ff61dedd7e0, callable=0x7ff61ded6f00) at /home/conda/feedstock_root/build_artifacts/python-split_1632942464411/work/Include/cpython/abstract.h:127
#12 call_function (kwnames=0x0, oparg=<optimized out>, pp_stack=<synthetic pointer>, tstate=0x5617cb0028b0) at /home/conda/feedstock_root/build_artifacts/python-split_1632942464411/work/Python/ceval.c:5075
#13 _PyEval_EvalFrameDefault (tstate=<optimized out>, f=<optimized out>, throwflag=<optimized out>) at /home/conda/feedstock_root/build_artifacts/python-split_1632942464411/work/Python/ceval.c:3487
#14 0x00005617c9a31e20 in _PyEval_EvalFrame (throwflag=0, f=0x7ff61dedd610, tstate=0x5617cb0028b0) at /home/conda/feedstock_root/build_artifacts/python-split_1632942464411/work/Include/internal/pycore_ceval.h:40
#15 _PyEval_EvalCode (tstate=0x5617cb0028b0, _co=<optimized out>, globals=<optimized out>, locals=<optimized out>, args=<optimized out>, argcount=<optimized out>, kwnames=0x0, kwargs=0x7ffca7442bd8, kwcount=<optimized out>, kwstep=1, defs=0x7ff6a3135da8, defcount=1, kwdefs=0x0, closure=0x0, name=0x7ff6a3119cb0, qualname=0x7ff6a30f50d0)
    at /home/conda/feedstock_root/build_artifacts/python-split_1632942464411/work/Python/ceval.c:4327
#16 0x00005617c9a98e86 in _PyFunction_Vectorcall (func=<optimized out>, stack=0x7ffca7442bc0, nargsf=<optimized out>, kwnames=<optimized out>) at /home/conda/feedstock_root/build_artifacts/python-split_1632942464411/work/Objects/call.c:396
#17 0x00005617c9a996f3 in _PyObject_VectorcallTstate (kwnames=<optimized out>, nargsf=<optimized out>, args=<optimized out>, callable=<optimized out>, tstate=<optimized out>) at /home/conda/feedstock_root/build_artifacts/python-split_1632942464411/work/Include/cpython/abstract.h:118
[... cut ...]
#51 0x00005617c9b8b329 in Py_BytesMain (argc=<optimized out>, argv=<optimized out>) at /home/conda/feedstock_root/build_artifacts/python-split_1632942464411/work/Modules/main.c:1129
#52 0x00007ff6c4f58560 in __libc_start_call_main (main=main@entry=0x5617c9a17120 <main>, argc=argc@entry=2, argv=argv@entry=0x7ffca74438e8) at ../sysdeps/nptl/libc_start_call_main.h:58
#53 0x00007ff6c4f5860c in __libc_start_main_impl (main=0x5617c9a17120 <main>, argc=2, argv=0x7ffca74438e8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7ffca74438d8) at ../csu/libc-start.c:409
#54 0x00005617c9b03a51 in _start ()
```

lorenzncode, 2022-01-02:
>nothing to do with solve

Ok got it.  Can reproduce the segfault without solve:

```
import cadquery as cq

s = (
    cq.Sketch()
    .segment((0, -4.0), (0, 6.0), "s1")
    .segment((0, 0), (0, 2.0),    "s2")
    .segment((2.0, 0), (0, 0),    "s3")
    .assemble()
)
```

adam-urbanczyk, 2022-01-02:
Alright, I think that `Face.makeFromWires` needs to check if every wire is closed.

PR #946 (2022-01-06): Face.makeFromWires - check that wires are closed
This is to resolve #944.  The segfault occurs when the outer wire is open and inner wires exist. 

An exception (for now?) to the closed wire check is the outer wire need not be closed if inner wires is empty.  Existing tests rely on this (Wire.fillet2D, chamfer2D).

Example of an open outer wire that would still be allowed: 

```
s = (
    cq.Sketch()
    .segment((0, 0), (0, 1.0))
    .segment((0, 1.0), (1.0, 1.0))
    .assemble()
)

r = cq.Workplane().placeSketch(s).extrude(0.1)
```

DXF import previously could result in the following when th [... cut]
closed 2022-01-06 by jmwright