CadQuery/cadquery #947: Rotation of imported step in assembly

davidvpspace, 2022-01-04:
Hello!

I am using Cadquery 2.x with CQ.editor. I have an simple assembly created in cadquery (**assy**). I would like to add an imported STEP file (**frame**) and add it to the previous assembly. However, I would like to either rotate the step file or the assembly before putting them together.

I am trying to rotate the step file so that afterwards I can place it to the desire position with respect to the assembly using cq.Location() inside the add method. This is an example of my code:

```
frame = cq.importers.importStep("frame.stp")  ##import step file
frame.rotateAboutCenter((1,0,0), 90.0)  ##rotate about X axis 90 deg
assy.add(self.frame,  loc=cq.Location(cq.Vector(-50,50,100))) ##add step to assembly and place to desired location
```

The code gives no errors, but nothing happens. The position of the imported step is the same. I have tried changing the rotation direction and rotation angle, but it's like rotateAboutCenter doesn't do anything to the step file.

Waht would be the correct way to rotate an imported step? For my intentions, it would be also possible to rotate the assembly before adding the step file to it.

Thanks in advance!

jmwright, 2022-01-04:
Thanks for posting over here @davidvpspace since more of the community sees what's posted here, but only me on the FreeCAD workbench repo.

`rotateAboutCenter` returns a copy, and doesn't mutate the original Workplane object. So your second line of code needs to be the following to work properly.
```python
frame = frame.rotateAboutCenter((1,0,0), 90.0)
```

adam-urbanczyk, 2022-01-04:
You could skip `rotateAboutCenter` and specify a `Location` including rotation when adding to the assy (NB: locs can be multiplied to vield a composite location).

davidvpspace, 2022-01-10:
Thank you @jmwright and @adam-urbanczyk ! Both solutions worked well for my needs.

closed 2022-01-10 by davidvpspace