CadQuery/cadquery #948: How to draw 3d curves

drhlxiao, 2022-01-04:
I would like to plot some 3d  curves (I have a list of points) over a model created with cadquery. 
What would be the correct way to do that?
Thanks in advance.

jmwright, 2022-01-04:
@drhlxiao Functions like `spline` take 3D position vectors, so if you already have the 3D points you should be able to pass them in just fine.
```python
import cadquery as cq

# The points that the spline will pass through
sPnts = [
    (2.75, 1.5, 0.0),
    (2.5, 1.75, 1.0),
    (2.0, 1.5, -1.0),
    (1.5, 1.0, 0.0),
    (1.0, 1.25, 1.0),
    (0.5, 1.0, -1.0),
    (0, 1.0, 0.0),
]

result = cq.Workplane("XY").spline(sPnts, includeCurrent=True)

show_object(result)
```

drhlxiao, 2022-01-04:
@jmwright   Thanks a lot.

drhlxiao, 2022-01-04:
@jmwright  Another question: would you know how to change the thickness and the colour of a curve?

jmwright, 2022-01-04:
> would you know how to change the thickness and the colour of a curve?

CQ-editor doesn't allow you to change the displayed thickness and color of a curve, if that is what you are asking. If you want control over the resulting 3D profile of the curve, as a work-around, you could sweep a profile (i.e. circle of a certain radius) along the 3D curve to give the curve 3 dimensions. Then you can add the curve to an assembly to set the color. That's the only way to set the color of an object currently in CadQuery.

joaotcarvalho, 2023-08-28:
I have a question about this solution.

When I try running the "cq.Workplane("XY").spline(sPnts)" method, I always get the curve starting at the origin of the Workplane, which is basically (0,0,0). However, when I try to use the "includeCurrent=True" I get the following error:

**Error executing CQGI-compliant script. spline() got an unexpected keyword argument 'includeCurrent'**

jmwright, 2023-08-30:
> Error executing CQGI-compliant script. spline() got an unexpected keyword argument 'includeCurrent'

@joaotcarvalho What version of CadQuery are you running? It works fine for me. Based on the error message, you may be running the CQ workbench for FreeCAD. If so, and if you're not running the fork that supports CadQuery 2.x, you'll need to upgrade (I would suggest running CQ-editor). I'm not even sure if the CQ 2.x fork of that workbench is maintained anymore.

joaotcarvalho, 2023-08-31:
You were right! I tried it in CQ-editor and it worked, thanks!