CadQuery/cadquery #949: Union of two tori does not work.

adam-urbanczyk, 2022-01-04:
Union of two tori does not work.

## To Reproduce

```python
import cadquery as cq

result1 = (
    cq
    .Workplane()
    .moveTo(24,0)
    .circle(8)
    .revolve()
    )

result2 = (
    cq
    .Workplane()
    .moveTo(24,16)
    .circle(8)
    .revolve()
    )

result = result1.union(result2)

show_object(result)
```

Results in: `Standard_NullObject: BRep_Tool: TopoDS_Vertex hasn't gp_Png`.

Moved from: https://github.com/CadQuery/CQ-editor/issues/305

jmwright, 2022-01-04:
Using `clean=False` or `glue=True` causes the union to succeed.

jalovisko, 2022-05-18:
I have an example where union doesn't work for me
```python
thickness = 1
unit_cell_size = 10
edge_points = [
    [[0.5, 0.5],
     [0.25, 0.0],
     [0.5, - 0.5]],
    [[- 0.5, - 0.5],
     [0.0, - 0.25],
     [0.5, - 0.5]],
    [[- 0.5, - 0.5],
     [0.0, - 0.25],
     [0.5, - 0.5]],
    [[- 0.5, - 0.5],
     [- 0.25, 0.0],
     [- 0.5, 0.5]],
    [[0.5, 0.5],
     [0.0, 0.25],
     [- 0.5, 0.5]],
    [[0.5, 0.5],
     [0.0, 0.25],
     [- 0.5, 0.5]],
]
# Multiplying the edge points by the unit cell size.
edge_points = np.array(edge_points) * unit_cell_size

plane_list = ["XZ", "XY", "YZ", "XZ", "YZ", "XY"]
offset_list = [- 1, 1, 1, 1, - 1, - 1]
offset_list = np.array(offset_list) * unit_cell_size * 0.5
edge_wire = (
    cq.Workplane(plane_list[0])
    .workplane(offset = - offset_list[0])
    .moveTo(edge_points[0][0][0], edge_points[0][0][1])
    .threePointArc(tuple(edge_points[0][1]),
                tuple(edge_points[0][2]))
)
for i in range(len(edge_points) - 1):
    # Adding the spline to the wire.
    edge_wire = edge_wire.add(
        cq.Workplane(plane_list[i + 1])
        .workplane(offset = - offset_list[i + 1])
        .moveTo(edge_points[i + 1][0][0], edge_points[i + 1][0][1])
        #.spline(edge_points[i + 1])
        .threePointArc(tuple(edge_points[i + 1][1]),
                    tuple(edge_points[i + 1][2]))
    )
surface_points = [[0, 0, 0]]
plate_4p = cq.Workplane("XY").interpPlate(
    edge_wire, surface_points, thickness * 0.5)
plate_4n = cq.Workplane("XY").interpPlate(
    edge_wire, surface_points, - thickness * 0.5)
plate_4 = plate_4n.union(plate_4p, clean=False, glue=True)
```
Individually, `plate_4p` and `plate_4n` exist:
[image]

But the union is empty.