CadQuery/cadquery #953: Undefined Symbols when installing with cadquery=master

WolfieTM, 2022-01-07:
## To Reproduce

Run `conda create -n cad-dev -c conda-forge -c cadquery cadquery=master` and try to import cadquery with:

```python
 import cadquery as cq
```

## Backtrace

```
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-f17bcdf9260a> in <module>()
----> 1 import cadquery as cq
      2 from jupyter_cadquery.cadquery import (PartGroup, Part, Edges, Faces, Vertices, show)
      3 from jupyter_cadquery import set_sidecar, set_defaults
      4 
      5 import alphashape as alph

/home/wolfie/anaconda3/envs/cad-dev/lib/python3.6/site-packages/cadquery/__init__.py in <module>()
      1 # these items point to the OCC implementation
----> 2 from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
      3 from .occ_impl.shapes import (
      4     Shape,
      5     Vertex,

/home/teddy/anaconda3/envs/cad-dev/lib/python3.6/site-packages/cadquery/occ_impl/geom.py in <module>()
      3 from typing import overload, Sequence, Union, Tuple, Type, Optional
      4 
----> 5 from OCP.gp import gp_Vec, gp_Ax1, gp_Ax3, gp_Pnt, gp_Dir, gp_Trsf, gp_GTrsf, gp, gp_XYZ
      6 from OCP.Bnd import Bnd_Box
      7 from OCP.BRepBndLib import BRepBndLib

ImportError: /home/wolfie/anaconda3/envs/cad-dev/lib/python3.6/site-packages/OCP.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN18BRepOffset_Inter2d14ConnexIntByIntERK11TopoDS_FaceR17BRepOffset_OffsetR19NCollection_DataMapI12TopoDS_ShapeS6_23TopTools_ShapeMapHasherERKS8_RKN11opencascade6handleI14BRepAlgo_AsDesEEddRK18BRepOffset_AnalyseR22NCollection_IndexedMapIS6_S7_ER26NCollection_IndexedDataMapIS6_16NCollection_ListIS6_ES7_E
```

## Environment

OS: Ubuntu

Was CadQuery installed using Conda?: 
Output of `conda list` from your active Conda environment:
```
# packages in environment at /home/otacon/anaconda3/envs/cad-issue:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
aiocontextvars            0.2.2                      py_0    conda-forge
aom                       3.2.0                h9c3ff4c_2    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
ca-certificates           2021.10.8            ha878542_0    conda-forge
cadquery                  master                    py3.6    cadquery
certifi                   2016.9.26                py36_0    conda-forge
contextvars               2.4                        py_0    conda-forge
curl                      7.80.0               h2574ce0_0    conda-forge
decorator                 5.1.0              pyhd8ed1ab_0    conda-forge
double-conversion         3.1.7                h9c3ff4c_0    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
expat                     2.4.2                h9c3ff4c_0    conda-forge
ezdxf                     0.14.2                     py_0    conda-forge
ffmpeg                    4.4.1                h6987444_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
[... 97 lines cut ...]
typing-extensions         4.0.1                hd8ed1ab_0    conda-forge
typing_extensions         4.0.1              pyha770c72_0    conda-forge
typish                    1.9.3              pyhd8ed1ab_0    conda-forge
utfcpp                    3.2.1                ha770c72_0    conda-forge
vtk                       9.0.1           no_osmesa_py36h2094402_107    conda-forge
wcwidth                   0.2.5              pyh9f0ad1d_2    conda-forge
wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
x264                      1!161.3030           h7f98852_1    conda-forge
x265                      3.5                  h4bd325d_1    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libice               1.0.10               h7f98852_0    conda-forge
xorg-libsm                1.2.3             hd9c2040_1000    conda-forge
xorg-libx11               1.7.2                h7f98852_0    conda-forge
xorg-libxau               1.0.9                h7f98852_0    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h7f98852_1    conda-forge
xorg-libxt                1.2.1                h7f98852_2    conda-forge
xorg-xextproto            7.3.0             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.5                h516909a_1    conda-forge
zlib                      1.2.11            h36c2ea0_1013    conda-forge
zstd                      1.5.1                ha95c52a_0    conda-forge
```

Using: Python Interpreter, Jupyter

WolfieTM, 2022-01-07:
Managed to get it working with a dev install instead of using conda. i.e.: 

```Shell
git clone https://github.com/CadQuery/cadquery.git && cd cadquery
mamba env create -n cad-dev -f environment.yml
python setup.y develop
```

So I imagine there is some issue with the conda recipe?

adam-urbanczyk, 2022-01-07:
Looks like you had an old master package in your cache. Try:
```
conda remove cadquery
conda clean --all --force-pkgs-dirs
conda install --force-reinstall -c conda-forge cadquery=master
```

WolfieTM, 2022-01-18:
I see. That seems to have done it thanks!

closed 2022-01-18 by WolfieTM