CadQuery/cadquery #957: CadQuery at FOSDEM 2022

jmwright, 2022-01-14:
I will be giving a [presentation](https://fosdem.org/2022/schedule/event/cadquery/) at FOSDEM 2022 with a short Q&A after. The times on the schedule are GMT+1 according to the organizers.  I'll be talking about what CadQuery is, briefly highlighting the new Sketch feature, and then talking about many of the projects going on within the community. There is a chat room for the devroom the presentation is in, and it would be great if we can have a small community presence there in case there are questions (although I can't sign in right now for some reason). I'll also be pointing attendees to our Discord, Matrix, Google Groups and GitHub.

Please post here if there are community projects you want to see highlighted. I can't guarantee I can cover everything in less than 20 minutes, but I'll do my best.

adam-urbanczyk, 2022-01-14:
I'd definitely mention https://github.com/meadiode/cq_gears

jchidley, 2022-02-12:
https://cqparts.github.io/ & https://cqparts.github.io/cqparts/doc/ 
https://github.com/bernhard-42/jupyter-cadquery - obviously

jmwright, 2022-02-12:
@jchidley Thanks for the ideas.

The presentation was on Feb 5th, and I did cover jupyter-cadquery. Please also note that cqparts has not been updated for CadQuery 2.x. I created a fork a year or more ago that had the minimum set of updates to get it working, but never heard from the cq-parts core team on merging it and it's fallen out of date. That's why cqparts wasn't included in the presentation.

jmwright, 2022-03-23:
The video for this event has been posted by the FOSDEM team so you can now watch/rewatch the presentation.

https://fosdem.org/2022/schedule/event/cadquery/

dcowden, 2022-03-23:
thanks for sharing! cant wait to watch it!

closed 2022-05-19 by jmwright