CadQuery/cadquery #958: Sketch arc length constraint result incorrect after solve

lorenzncode, 2022-01-15:
Length of a1 is expected to be 1.0:

https://github.com/CadQuery/cadquery/blob/4fcada66e9fb5d8cebfdf78376154b7b253df991/tests/test_sketch.py#L442

The current result is ~0.59.  I'll open a PR for this issue.

adam-urbanczyk, 2022-01-15:
Hm, does this assert not work then? https://github.com/CadQuery/cadquery/blob/4fcada66e9fb5d8cebfdf78376154b7b253df991/tests/test_sketch.py#L453

adam-urbanczyk, 2022-01-15:
Do you understand the rootcause? Did I over-constrain or is there something wrong with the cost?

lorenzncode, 2022-01-15:
Yes this one is straightforward to fix.   The length_cost should be radius*da.  Remove x[3] here:

https://github.com/CadQuery/cadquery/blob/4fcada66e9fb5d8cebfdf78376154b7b253df991/cadquery/occ_impl/sketch_solver.py#L163

closed 2022-01-21 by jmwright